zephyrproject-rtos/zephyr #90306: Heap allocation causes OS crash

spanceac, 2025-05-21:
**Describe the bug**

For a heap of size *n*, every `k_malloc` with size > *n* - 16 will result in an OS crash.
I could reproduce on both `qemu_x86` and `nrf52840dk/nrf52840` boards.

**To Reproduce**

Configure project with `CONFIG_HEAP_MEM_POOL_SIZE=2048` and in application call `k_malloc(2033)`.

**Expected behavior**

Even if there's not enough heap space for this allocation, the OS shouldn't crash.

**Impact**

**Logs and console output**

```
[00:00:00.306,854] <err> os: ***** BUS FAULT *****
[00:00:00.306,854] <err> os:   Precise data bus error
[00:00:00.306,884] <err> os:   BFAR Address: 0x2f9140b2
[00:00:00.306,915] <err> os: r0/a1:  0x20014080  r1/a2:  0x00000100  r2/a3:  0x00000008
[00:00:00.306,915] <err> os: r3/a4:  0x00000017 r12/ip:  0x0f900030 r14/lr:  0x00000003
[00:00:00.306,945] <err> os:  xpsr:  0x01000000
[00:00:00.306,945] <err> os: Faulting instruction address (r15/pc): 0x00009fc4
[00:00:00.306,976] <err> os: >>> ZEPHYR FATAL ERROR 25: Unknown error on CPU 0
[00:00:00.307,006] <err> os: Current thread: 0x20010748 (unknown)
[00:00:02.933,380] <err> os: Halting system
```
**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain: Zephyr SDK
 - Version used: 4.1.99

**Additional context**

ljd42, 2025-05-27:
Preliminary Observations: 
--------------------------------------
Run some test with native_sim to check if `k_malloc(s)` behaves as expected for s=1..65536 for the given CONFIG_HEAP_MEM_POOL_SIZE=2048. 

I confirm this crash for the following size s:
2033 = 0b11111110001
2034 = 0b11111110010
2035 = 0b11111110011

Next POOL_SIZE where this problem occurs is 8192, for the following size s:
8177 = 0b1111111110001
8178 = 0b1111111110010
8179 = 0b1111111110011

I let you guess for which other of POOL_SIZE / size s value it crashes too...

This preliminary observations suggest that there is a pattern at work. I don't have dig the allocation code yet.

peter-mitsis, 2025-05-29:
I've been peeling back the layers on this, and it is looking like the problem may lie within sys_heap_alloc()--specifically with the following code:

```
if ((bytes == 0U) || size_too_big(h, bytes)) {
    return NULL;
}
chunksz_t chunk_sz = bytes_to_chunksz(h, bytes);
chunkid_t c = alloc_chunk(h, chunk_sz);
```

The function `bytes_to_chunksz()` may calculate a larger amount of memory than the preceding `size_too_big()` check. As a result, there is a small window of values where `alloc_chunk()` may be called when it shouldn't.

PR #90969 (2025-06-06): kernel: sys_heap: Fix size_too_big() check
Updates the heap code so that size_too_big() validates the actual number of bytes that will be subsequently be requested.

Fixes #90306

---

Updates:
* Elminates size_too_big() as it does not catch all overflow cases.
* Updates bytes_to_chunksz() so we do not overflow when converting bytes to chunks.
* Clamps return value of bytes_to_chunksz()
closed 2025-06-06 by kartben