zephyrproject-rtos/zephyr #90318: drivers: spi: spi_loopback: drivers.spi.* fail

katgiadla, 2025-05-22:
**Describe the bug**
Tests:
- `tests/drivers/spi/spi_loopback/drivers.spi.nrf_pm_runtime`
- `tests/drivers/spi/spi_loopback/drivers.spi.loopback`
- `tests/drivers/spi/spi_loopback/drivers.spi.nrf54l_32mhz`
- `tests/drivers/spi/spi_loopback/drivers.spi.nrf54l_16mhz`
- `tests/drivers/spi/spi_loopback/drivers.spi.nrf54l_8mhz`

fail.

Observed for ` nrf54l15dk/nrf54l15/cpuapp`

**To Reproduce**
Steps to reproduce the behavior:
1. have `nrf54l15dk/nrf54l15/cpuapp` connected
2. connect jump wire like in overlay
3. go to your zephyr dir
4. call `./scripts/twister -T tests/subsys/debug/cpu_load -p nrf54l15dk/nrf54l15/cpuapp --device-testing --device-serial /dev/ttyACM1 -v --inline-logs --west-flash="--erase" --fixture spi_loopback`
5. See console output with error
**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
===================================================================
START - test_spi_word_size_16
E: Word sizes other than 8 bits are not supported
Spi config invalid for this controller - skip

    Assertion failed at zephyr/tests/drivers/spi/spi_loopback/src/spi.c:628: spi_loopback_test_word_size: (memcmp(compare_data, rx_buffer, buffer_size) is true)
16-bit word buffer contents are different
 FAIL - test_spi_word_size_16 in 0.029 seconds
===================================================================
START - test_spi_word_size_24
E: Word sizes other than 8 bits are not supported
Spi config invalid for this controller - skip

    Assertion failed at zephyr/tests/drivers/spi/spi_loopback/src/spi.c:628: spi_loopback_test_word_size: (memcmp(compare_data, rx_buffer, buffer_size) is true)
24-bit word buffer contents are different
 FAIL - test_spi_word_size_24 in 0.029 seconds
===================================================================
START - test_spi_word_size_32
E: Word sizes other than 8 bits are not supported
Spi config invalid for this controller - skip

    Assertion failed at zephyr/tests/drivers/spi/spi_loopback/src/spi.c:628: spi_loopback_test_word_size: (memcmp(compare_data, rx_buffer, buffer_size) is true)
32-bit word buffer contents are different
 FAIL - test_spi_word_size_32 in 0.029 seconds
===================================================================
...
===================================================================
START - test_spi_word_size_9
E: Word sizes other than 8 bits are not supported
Spi config invalid for this controller - skip

    Assertion failed at zephyr/tests/drivers/spi/spi_loopback/src/spi.c:628: spi_loopback_test_word_size: (memcmp(compare_data, rx_buffer, buffer_size) is true)
9-bit word buffer contents are different
 FAIL - test_spi_word_size_9 in 0.029 seconds
===================================================================
...
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.17.0
 - Commit SHA or Version used: f47f3fe410681c55d17b26e11e33af3fec2c472e

bjarki-andreasen, 2025-05-22:
Very unlikely, there was a PR which refactored and added new tests cases for different word sizes, so its probably just something that has always been broken :)

bjarki-andreasen, 2025-05-22:
https://github.com/zephyrproject-rtos/zephyr/pull/87838

PR #90321 (2025-05-26): tests: drivers: spi: spi_loopback: skip tests if invalid config
The `spi_loopback_transceive()` helper currently only prints a message if a configuration is invalid, continuing the test case as if it succeeded. This results in the test case using the helper trying to validate the result from a spi transaction that was skipped.

Fix this by explicitly skipping the test using the ztest framework's ztest_test_skip() function, which skips the entire test case.

Before this patch:

```
===================================================================
START - test_spi_word_size_9
E: Word sizes oth [... cut]
closed 2025-05-26 by kartben