zephyrproject-rtos/zephyr #90325: SPI transaction fails waiting for lpspi_wait_tx_fifo_empty() on some Flexcomm but not others on MCXN947.

shortst, 2025-05-22:
## Overview
Calls to spi_write(), spi_read() and other similar functions block when using Flexcomm 5 and 8 but succeed on 1 and 7.

## Context 
Soc is MCXN947.
Zephyr revision: v4.1.0
SDK version: zephyr-sdk-0.17.1-rc1
Host OS: Fedora 41

This was initially noticed on a custom board with a custom driver. However the same behaviour is present on the frdm_mcxn947 dev board and with other drivers: lsm6dso max6675, st7789v.

When using FC1 or FC7 the call completes as expected. On FC5 or FC8, execution is spinning, waiting for the tx fifo:
```
^C
Program received signal SIGTRAP, Trace/breakpoint trap.
lpspi_wait_tx_fifo_empty (dev=0x10007564 <__device_dts_ord_145>) at /fast/db1-bringup/zephyr/drivers/spi/spi_nxp_lpspi/spi_nxp_lpspi.c:23
23		while (LPSPI_GetTxFifoCount(base) != 0) {
(gdb) 
```
Seen at INIT_LEVEL_PRE_KERNEL_2 (if the driver uses SPI during init), INIT_LEVEL_POST_KERNEL and possibly INIT_LEVEL_APPLICATION (I wasn't looking then).

 I have stepped though nxp_lp_flexcomm_init() for success and failure cases, it appears the same and at first pass, it seems that the clocks and pins are being set-up the same.

## Steps to reproduce:
I picked the max6675 sample as a simple test case because it only calls spi_read(). Hence, it can run on a dev board without any extra hardware. We are not actually using the max6675.

Sample overlays here:
https://github.com/shortst/flexcomm-bug/tree/main

### Success case
```
export ZEPHYR_SDK_INSTALL_DIR=/opt/zephyr-sdk/
cd zephyr/samples/sensor/max6675
west build -b frdm_mcxn947/mcxn947/cpu0 --  -DDTC_OVERLAY_FILE=flexcomm7.overlay
west flash --runner=jlink
# Observe serial log - repeat readings printed, 0 because no sensor attached

```
### Failure case:
```
export ZEPHYR_SDK_INSTALL_DIR=/opt/zephyr-sdk/
cd zephyr/samples/sensor/max6675
west build -b frdm_mcxn947/mcxn947/cpu0 --  -DDTC_OVERLAY_FILE=flexcomm5.overlay 
west flash --runner=jlink
# Sample does not run, so observe in debugger:
west debug --runner=jlink
# run with c, halt with ctrl-c and look at trace

```
## Impact
(As the bug crib sheet asked..)
The custom board is for a commercial product, we are using 9 Flexcomms, of the 4 tested so far 2 are unusable. Development is halted on access to the unusable ports.

decsny, 2025-05-27:
I am assuming the issue is that the flexcomms are not clocked, the clocking is hardcoded in board and soc files, you should check there.

shortst, 2025-06-03:
Thanks for the steer.
boards/nxp/frdm_mcxn947/board.c has the below block for FC1,2,3,4,7. But omits 0,5,6,8,9.

```
#if DT_NODE_HAS_STATUS_OKAY(DT_NODELABEL(flexcomm1))
	CLOCK_SetClkDiv(kCLOCK_DivFlexcom1Clk, 1u);
	CLOCK_AttachClk(kFRO12M_to_FLEXCOMM1);
#endif
```

Although it seems quite obvious with hindsight, I took me quite a journey to find as this happens on a per-instance basis but not with the other per instance basis code (flexcomm driver). I can't be the only person to get caught by this silent inconsistency. It also kind of rubs against the idea of overlays. 

Would a PR be welcome? Either to replicate the above for the missing items or perhaps something like:
```
#if DT_NODE_HAS_STATUS_OKAY(DT_NODELABEL(flexcomm9))
    __WARN("flexcomm9 enabled in device tree but clock may not be attached")
#endif
```

(I also note a potentially similar issue with gpio5)

decsny, 2025-06-03:
You can make a PR if you want, but, I personally decided that I was going to rework how this is done completely by clocking it in the flexcomm driver, instead of platform hardcoding, but don't know when I'll get to that. And your suspicion that it's a common problem people run into is correct.

closed 2025-08-19 by kartben