zephyrproject-rtos/zephyr #90332: Display driver tests fail

yishai1999, 2025-05-22:
**Describe the bug**

Running the display driver tests fails.
I'm trying to run them for `native_sim`.

**To Reproduce**

`west build -p auto -b native_sim/native/64 zephyr/tests/drivers/display/display_read_write`
`./build/zephyr/zephyr.exe`

**Logs and console output**

```
*** Booting Zephyr OS build v4.1.0-4465-g438ed6d74aea ***
Running TESTSUITE display_read_write
===================================================================
START - test_clear
 PASS - test_clear in 0.000 seconds
===================================================================
START - test_read_does_not_clear_existing_buffer

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/display/display_read_write/src/main.c:64: verify_bytes_of_area: data not equal to disp_buffer

 FAIL - test_read_does_not_clear_existing_buffer in 0.000 seconds
===================================================================
START - test_write_to_buffer_head

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/display/display_read_write/src/main.c:64: verify_bytes_of_area: data not equal to disp_buffer

 FAIL - test_write_to_buffer_head in 0.000 seconds
===================================================================
START - test_write_to_buffer_tail

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/display/display_read_write/src/main.c:210: display_read_write_test_write_to_buffer_tail: data not equal to disp_buffer + (display_width * display_height * bpp - buf_size)

 FAIL - test_write_to_buffer_tail in 0.000 seconds
===================================================================
TESTSUITE display_read_write failed.

------ TESTSUITE SUMMARY START ------

SUITE FAIL -  25.00% [display_read_write]: pass = 1, fail = 3, skip = 0, total = 4 duration = 0.000 seconds
 - PASS - [display_read_write.test_clear] duration = 0.000 seconds
 - FAIL - [display_read_write.test_read_does_not_clear_existing_buffer] duration = 0.000 seconds
 - FAIL - [display_read_write.test_write_to_buffer_head] duration = 0.000 seconds
 - FAIL - [display_read_write.test_write_to_buffer_tail] duration = 0.000 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
PROJECT EXECUTION FAILED
```

**Environment:**

 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit: master

aescolar, 2025-07-02:
CC @soburi @jfischer-no @ VynDragon @jarmouniA @faxe1008 @Finomnis 
@soburi should we have a board overlay setting `CONFIG_SDL_DISPLAY_USE_HARDWARE_ACCELERATOR=n` ?

soburi, 2025-07-02:
> CC [@soburi](https://github.com/soburi) [@jfischer-no](https://github.com/jfischer-no) [@VynDragon](https://github.com/VynDragon) [@JarmouniA](https://github.com/JarmouniA) [...]

In the scope of the this issue description, certainly need this option.
(The option is set in testcase.yaml.)
But, #88714 problem is still remain, we need to investigate the root cause.

PR #92523 (2025-07-04): tests/drivers display_read_write: Move common kconfig option to overlay
For native_sim[//64], these tests need to be built with CONFIG_SDL_DISPLAY_USE_HARDWARE_ACCELERATOR=n

Until now we defined this option in the yaml file for each test targetting the sdl display driver.
But if somebody builds the test directly, it will fail as these yaml kconfig defined options would not be used.

Let's instead define it in a board overlay, so in that case the test will also work.

Fixes #90332
closed 2025-07-04 by danieldegrasse