zephyrproject-rtos/zephyr #90403: drivers: ieee802154: mcxw71 radio_api->get_time() in wrong resolution

SynchronicIT, 2025-05-23:
In drivers/ieee802154/ieee802154_mcxw.c the function net_time_t mcxw_get_time(const struct device *dev) {.. } returns the current time in microseconds ([us]). 

This is used throughout the driver, but also applied directly in the radio api:
```
struct ieee802154_radio_api mcxw71_radio_api ={ 
...
.get_time = mcxw_get_time, 
...
}. 

```

The spec for the get_time interface is to output in  nanoseconds [ns]. 

```
	/**
	 * @brief Get the current time in nanoseconds relative to the network
	 * subsystem's local uptime clock as represented by this network
	 * interface.
	 *
	 * See @ref net_time_t for semantic details.
	 *
	 * @note requires IEEE802154_HW_TXTIME and/or IEEE802154_HW_RXTIME
	 * capabilities. Implementations SHALL be **isr-ok** and MUST NOT
	 * **sleep**. MAY be called in any interface state once the driver is
	 * fully initialized ("ready").
	 *
	 * @param dev pointer to IEEE 802.15.4 driver device
	 *
	 * @return nanoseconds relative to the network subsystem's local clock,
	 * -1 if an error occurred or the operation is not supported
	 */
	net_time_t (*get_time)(const struct device *dev);

```

Transmitting with IEEE802154_TX_MODE_TXTIME would also need to have the [us] time in the packet and not the [ns] as specified. 

Internally, the lptmr0 is claimed for this time and not properly translated to [ns]. 

Proposal is to update all external time interfaces to [ns] conform specification. Optionally, skip the use of the lptmr0, and use  PhyTime_ReadClock(); internally, although I don't know the lifecycle of this PhyClock.

henrikbrixandersen, 2025-05-24:
Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

dleach02, 2025-05-28:
@andrei-menzopol can you please investigate this issue