zephyrproject-rtos/zephyr #90426: LTO discards NXP Kinetis Flash Config

mspeder, 2025-05-24:
**Describe the bug**
When Link Time Optimization is active, the __kinetis_flash_config section is discarded by the linker

**To Reproduce**
Compile a kinetis project with CONFIG_LTO=y and look at map file.

**Fix**
Attribute "used" must be added to to prevent unwanted deletion in zephyr/soc/nxp/kinetis
/flash_configuration.c :

`uint8_t __attribute__((used)) __kinetis_flash_config_section __kinetis_flash_config[] = {`

henrikbrixandersen, 2025-05-24:
Please submit a PR for fixing this.

jevidin, 2025-05-24:
Hi @henrikbrixandersen 

Can I take a look at this as my first issue? I know we have `Good first issue` labels, but it seems like they're mostly re-implementing C libs.

Thanks!

closed 2025-05-31 by kartben