zephyrproject-rtos/zephyr #90459: st7789 display working with spi but not the new mipi configuration with the same code

ljunquera, 2025-05-25:
I am trying to get a st7789 display working with a nrf5340dk. I have the same code running with spi and mipi but it is not working with mipi. It will build and flash but I only see "Hello World" shown on the direct spi version.

This is what is working:
```
&pinctrl {

    spi4_default: spi4_default {
        group1 {
            psels = <NRF_PSEL(SPIM_SCK, 0, 6)>,
                <NRF_PSEL(SPIM_MOSI, 0, 25)>;
        };
    };

    spi4_sleep: spi4_sleep {
        group1 {
            psels = <NRF_PSEL(SPIM_SCK, 0, 6)>,
                <NRF_PSEL(SPIM_MOSI, 0, 25)>;
            low-power-enable;
        };
    };

    pwm1_default: pwm1_default {
		group1 {
			psels = <NRF_PSEL(PWM_OUT0, 0, 7)>;
		};
	};

	pwm1_sleep: pwm1_sleep {
		group1 {
			psels = <NRF_PSEL(PWM_OUT0, 0, 7)>;
			low-power-enable;
		};
	};
};

&pwm1 {
    status = "okay";
    pinctrl-0 = <&pwm1_default>;
    pinctrl-1 = <&pwm1_sleep>;
    pinctrl-names = "default", "sleep";
};

&spi4 {
	compatible = "nordic,nrf-spim";
	status = "okay";
	pinctrl-0 = <&spi4_default>;
	pinctrl-1 = <&spi4_sleep>;
        pinctrl-names = "default", "sleep";
	cs-gpios = <&gpio0 11 GPIO_ACTIVE_LOW>;

    st7789v_st7789v_tl019fqv01: st7789v@0 {
        compatible = "sitronix,st7789v";
        spi-max-frequency = <20000000>;
        reg = <0>;
        cmd-data-gpios = < &gpio1 11 GPIO_ACTIVE_LOW>;
        reset-gpios = < &gpio1 10 GPIO_ACTIVE_LOW>;
        width = <240>;
        height = <280>;
        x-offset = <0>;
        y-offset = <20>;
        vcom = <0x19>;
        gctrl = <0x35>;
        vrhs = <0x12>;
        vdvs = <0x20>;
        mdac = <0x00>;
        gamma = <0x01>;
        colmod = <0x05>;
        lcm = <0x2c>;
        porch-param = [0c 0c 00 33 33];
        cmd2en-param = [5a 69 02 01];
        pwctrl1-param = [a4 a1];
        pvgam-param = [D0 04 0D 11 13 2B 3F 54 4C 18 0D 0B 1F 23];
        nvgam-param = [D0 04 0C 11 13 2C 3F 44 51 2F 1F 1F 20 23];
        ram-param = [00 F0];
        rgb-param = [CD 08 14];
	};
};

/ {
    chosen {
        zephyr,display = &st7789v_st7789v_tl019fqv01;
    };

    aliases {
        screenblk = &display_blk;
    };

    pwmleds {
        compatible = "pwm-leds";
	
        display_blk: pwm_led_0 {
            pwms = <&pwm1 0 PWM_USEC(15) PWM_POLARITY_NORMAL>;
            label = "Display Backlight";
        };
    };

    buttons {
		compatible = "gpio-keys";
		button1: button_1 {
			gpios = <&gpio0 24 (GPIO_PULL_UP | GPIO_ACTIVE_LOW)>;
			label = "Push button 2";
			zephyr,code = <INPUT_KEY_1>;
		};
	};

};
```
This is the mipi configuration which is not working:
```
/* Include the MIPI DBI definitions */
#include <zephyr/dt-bindings/mipi_dbi/mipi_dbi.h>

&pinctrl {

    spi4_default: spi4_default {
        group1 {
            psels = <NRF_PSEL(SPIM_SCK, 0, 6)>,
                <NRF_PSEL(SPIM_MOSI, 0, 25)>;
        };
    };

    spi4_sleep: spi4_sleep {
        group1 {
            psels = <NRF_PSEL(SPIM_SCK, 0, 6)>,
                <NRF_PSEL(SPIM_MOSI, 0, 25)>;
            low-power-enable;
        };
    };

    pwm1_default: pwm1_default {
		group1 {
			psels = <NRF_PSEL(PWM_OUT0, 0, 7)>;
		};
	};

	pwm1_sleep: pwm1_sleep {
		group1 {
			psels = <NRF_PSEL(PWM_OUT0, 0, 7)>;
			low-power-enable;
		};
	};
};

&pwm1 {
    status = "okay";
    pinctrl-0 = <&pwm1_default>;
    pinctrl-1 = <&pwm1_sleep>;
    pinctrl-names = "default", "sleep";
};

&spi4 {
	status = "okay";
	pinctrl-0 = <&spi4_default>;
	pinctrl-1 = <&spi4_sleep>;
	cs-gpios = <&gpio0 11 GPIO_ACTIVE_LOW>;
};

/ {
    chosen {
        zephyr,display = &st7789v_st7789v_tl019fqv01;
    };

    aliases {
        screenblk = &display_blk;
    };

    pwmleds {
        compatible = "pwm-leds";
        display_blk: pwm_led_0 {
             pwms = <&pwm1 0 PWM_USEC(15) PWM_POLARITY_NORMAL>;
             label = "Display Backlight";
         };
    };

    mipi_dbi {
        compatible = "zephyr,mipi-dbi-spi";
        reset-gpios = < &gpio1 10 GPIO_ACTIVE_LOW>;
        dc-gpios = < &gpio1 11 GPIO_ACTIVE_HIGH>;
            spi-dev = <&spi4>;
            #address-cells = <1>;
            #size-cells = <0>;
            write-only;
            st7789v_st7789v_tl019fqv01: st7789v@0 {
                compatible = "sitronix,st7789v";
                reg = <0>;
                mipi-max-frequency = <20000000>;
                mipi-mode = <MIPI_DBI_MODE_SPI_4WIRE>;
                status = "okay";
                width = <240>;
                height = <280>;
                x-offset = <0>;
                y-offset = <20>;
                vcom = <0x19>;
                gctrl = <0x35>;
                vrhs = <0x12>;
                vdvs = <0x20>;
                mdac = <0x00>;
                gamma = <0x01>;
                colmod = <0x05>;
                lcm = <0x2c>;
                porch-param = [0c 0c 00 33 33];
                cmd2en-param = [5a 69 02 01];
                pwctrl1-param = [a4 a1];
                pvgam-param = [D0 04 0D 11 13 2B 3F 54 4C 18 0D 0B 1F 23];
                nvgam-param = [D0 04 0C 11 13 2C 3F 44 51 2F 1F 1F 20 23];
                ram-param = [00 F0];
                rgb-param = [CD 08 14];
        };
    };

    buttons {
		compatible = "gpio-keys";
		button1: button_1 {
			gpios = <&gpio0 24 (GPIO_PULL_UP | GPIO_ACTIVE_LOW)>;
			label = "Push button 2";
			zephyr,code = <INPUT_KEY_1>;
		};
	};
};
```

ljunquera, 2025-05-27:
Done. I think that's everything.

danieldegrasse, 2025-07-03:
> In MIPI-DBI configuration, the missing pinctrl-names is already present in the SPI node, even if I add it in the MIPI-DBI node, it still doesn't show anything.

Ok- maybe I missed it in the devicetree snippets you included, I didn't see one for the SPI node when using MIPI-DBI. Could you try running with `CONFIG_DISPLAY_LOG_LEVEL_DBG=y`? Wondering if there is any output there that could give a hint

nanangcreacoll, 2025-07-07:
Zephyr v.3.6.99
```
[00:00:00.395,690] <dbg> display_st7789v: st7789v_reset_display: Resetting display
*** Booting nRF Connect SDK v2.7.0-5cb85570ca43 ***
*** Using Zephyr OS v3.6.99-100befc70c74 ***
--- 4 messages dropped ---
[00:00:00.552,307] <dbg> display_st7789v: st7789v_write: Writing 240x1 (w,h) @ 0x9 (x,y)
--- 23 messages dropped ---
[00:00:00.569,824] <dbg> display_st7789v: st7789v_write: Writing 240x1 (w,h) @ 0x31 (x,y)
--- 21 messages dropped ---
[00:00:00.588,958] <dbg> display_st7789v: st7789v_write: Writing 240x1 (w,h) @ 0x55 (x,y)
--- 22 messages dropped ---
[00:00:00.607,330] <dbg> display_st7789v: st7789v_write: Writing 240x1 (w,h) @ 0x78 (x,y)
--- 22 messages dropped ---
[00:00:00.625,671] <dbg> display_st7789v: st7789v_write: Writing 240x1 (w,h) @ 0x101 (x,y)
--- 23 messages dropped ---
[00:00:00.644,042] <dbg> display_st7789v: st7789v_write: Writing 240x1 (w,h) @ 0x124 (x,y)
--- 22 messages dropped ---
[00:00:00.663,208] <dbg> display_st7789v: st7789v_write: Writing 240x1 (w,h) @ 0x148 (x,y)
--- 22 messages dropped ---
[00:00:00.681,549] <dbg> display_st7789v: st7789v_write: Writing 240x1 (w,h) @ 0x171 (x,y)
--- 23 messages dropped ---
[00:00:00.699,920] <dbg> display_st7789v: st7789v_write: Writing 240x1 (w,h) @ 0x194 (x,y)
--- 22 messages dropped ---
[00:00:00.719,085] <dbg> display_st7789v: st7789v_write: Writing 240x1 (w,h) @ 0x218 (x,y)
--- 10 messages dropped ---
[00:00:00.723,083] <dbg> display_st7789v: st7789v_write: Writing 240x1 (w,h) @ 0x223 (x,y)
--- 1 messages dropped ---
[00:00:00.724,670] <dbg> display_st7789v: st7789v_write: Writing 240x1 (w,h) @ 0x225 (x,y)
[00:00:00.725,463] <dbg> display_st7789v: st7789v_write: Writing 240x1 (w,h) @ 0x226 (x,y)
[00:00:00.726,287] <dbg> display_st7789v: st7789v_write: Writing 240x1 (w,h) @ 0x227 (x,y)
[00:00:00.727,081] <dbg> display_st7789v: st7789v_write: Writing 240x1 (w,h) @ 0x228 (x,y)
[00:00:00.727,874] <dbg> display_st7789v: st7789v_write: Writing 240x1 (w,h) @ 0x229 (x,y)
[00:00:00.728,668] <dbg> display_st7789v: st7789v_write: Writing 240x1 (w,h) @ 0x230 (x,y)
[00:00:00.729,461] <dbg> display_st7789v: st7789v_write: Writing 240x1 (w,h) @ 0x231 (x,y)
```
Zephyr v4.1.0
```
[00:00:00.384,460] <dbg> display_st7789v: st7789v_reset_display: Resetting display
--- 4 messages dropped ---
[00:00:00.541,168] <dbg> display_st7789v: st7789v_write: Writing 240x1 (w,h) @ 0x8 (x,y)
--- 23 messages dropped ---
[00:00:00.561,279] <dbg> display_st7789v: st7789v_write: Writing 240x1 (w,h) @ 0x31 (x,y)
--- 22 messages dropped ---
[00:00:00.582,336] <dbg> display_st7789v: st7789v_write: Writing 240x1 (w,h) @ 0x55 (x,y)
--- 22 messages dropped ---
[00:00:00.601,562] <dbg> display_st7789v: st7789v_write: Writing 240x1 (w,h) @ 0x77 (x,y)
--- 21 messages dropped ---
[00:00:00.621,643] <dbg> display_st7789v: st7789v_write: Writing 240x1 (w,h) @ 0x100 (x,y)
--- 23 messages dropped ---
[00:00:00.641,662] <dbg> display_st7789v: st7789v_write: Writing 240x1 (w,h) @ 0x123 (x,y)
--- 22 messages dropped ---
[00:00:00.662,689] <dbg> display_st7789v: st7789v_write: Writing 240x1 (w,h) @ 0x147 (x,y)
--- 22 messages dropped ---
[00:00:00.682,861] <dbg> display_st7789v: st7789v_write: Writing 240x1 (w,h) @ 0x170 (x,y)
--- 23 messages dropped ---
[00:00:00.703,033] <dbg> display_st7789v: st7789v_write: Writing 240x1 (w,h) @ 0x193 (x,y)
--- 22 messages dropped ---
[00:00:00.724,060] <dbg> display_st7789v: st7789v_write: Writing 240x1 (w,h) @ 0x217 (x,y)
--- 10 messages dropped ---
[00:00:00.729,339] <dbg> display_st7789v: st7789v_write: Writing 240x1 (w,h) @ 0x223 (x,y)
--- 2 messages dropped ---
[00:00:00.731,079] <dbg> display_st7789v: st7789v_write: Writing 240x1 (w,h) @ 0x225 (x,y)
[00:00:00.731,964] <dbg> display_st7789v: st7789v_write: Writing 240x1 (w,h) @ 0x226 (x,y)
[00:00:00.732,849] <dbg> display_st7789v: st7789v_write: Writing 240x1 (w,h) @ 0x227 (x,y)
[00:00:00.733,703] <dbg> display_st7789v: st7789v_write: Writing 240x1 (w,h) @ 0x228 (x,y)
[00:00:00.734,588] <dbg> display_st7789v: st7789v_write: Writing 240x1 (w,h) @ 0x229 (x,y)
[00:00:00.735,473] <dbg> display_st7789v: st7789v_write: Writing 240x1 (w,h) @ 0x230 (x,y)
[00:00:00.736,358] <dbg> display_st7789v: st7789v_write: Writing 240x1 (w,h) @ 0x231 (x,y)
[00:00:00.737,213] <dbg> display_st7789v: st7789v_write: Writing 240x1 (w,h) @ 0x232 (x,y)
```

I don't see any difference in the two debug logs. Some log messages are also dropped.