zephyrproject-rtos/zephyr #90466: nrf52: high standby current on nrf52840 when using MCUBOOT + UART_ASYNC_API + PM_DEVICE

mjm987, 2025-05-25:
**Describe the bug**
When using CONFIG_PM_DEVICE up to Zephyr v3.7 the standby current was about 6 uA by setting pm_device_action_run(uart0, PM_DEVICE_ACTION_SUSPEND) with or without any combination of CONFIG_UART_ASYNC_API and/or CONFIG_BOOTLOADER_MCUBOOT.

But since Zephyr v4.0 the standby current after pm_device_action_run(uart0, PM_DEVICE_ACTION_SUSPEND)  will be between 0.5mA and 1mA in case of following combination:
CONFIG_PM_DEVICE + CONFIG_UART_ASYNC_API + CONFIG_BOOTLOADER_MCUBOOT

**To Reproduce**
1. Use eg. the Nordic nRF52840DK board (-b nrf52840dk/nrf52840). 
2. Use blinky sample
3. Add in prj.conf:
     CONFIG_PM_DEVICE=y
     CONFIG_UART_ASYNC_API=y
     CONFIG_SERIAL=y
     CONFIG_LOG=y
     CONFIG_BOOTLOADER_MCUBOOT=y

6. Replace in main.c::main() the printk() statement by:		 pm_device_action_run(DEVICE_DT_GET(DT_NODELABEL(uart0)), PM_DEVICE_ACTION_RESUME);		LOG_INF("LED state: %s\n", led_state ? "ON" : "OFF");
		while (log_process()) ;
		pm_device_action_run(DEVICE_DT_GET(DT_NODELABEL(uart0)), PM_DEVICE_ACTION_SUSPEND);
5. 
-->

**Expected behavior**

**Impact**

**Logs and console output**

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**

carlescufi, 2025-06-05:
@nvlsianpu and @nordicjm could MCUboot be leaving a peripheral turned on?
@nordic-krch any clues?

nordic-krch, 2025-06-05:
if it's only when MCUBOOT is present then i would check this: https://github.com/zephyrproject-rtos/zephyr/pull/75703 uart driver does not attempt to do any cleanup for the bootloader. If it is needed application need to do that before UART is initialized.

mjm987, 2025-06-05:
Unfortunately the cleanup as shown in #75703 did not help.

nordicjm, 2025-06-09:
Can you disable `CONFIG_SERIAL` in MCUboot and see if current reduces? If so then points to issue with UART cleanup, not sure that is supported in zephyr, is in ncs

bjarki-andreasen, 2025-08-17:
its almost certain to be because mcuboot has not implemented cleanup of hardware, so the UART remains active when mcuboot jumps to application. This is being worked on with the device_deinit() feature, which is being implemented for the nRF UARTE, so if it is this case, it will be fixed soon enough. For reference, the NCS downstream has cleanup added manually for nrf https://github.com/nrfconnect/sdk-mcuboot/blob/main/boot/zephyr/nrf_cleanup.c which hopefully will become redundant in the future