zephyrproject-rtos/zephyr #90481: [Coverity CID: 524767] Overflowed constant in drivers/input/input_cap12xx.c

zephyrbot, 2025-05-25:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/265cfb45a818f39cf03c3527b5d1b21027d3cac1/drivers/input/input_cap12xx.c

Category: Integer handling issues
Function: `cap12xx_set_calsens`
Component: Drivers
CID: [524767](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=524767)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/265cfb45a818f39cf03c3527b5d1b21027d3cac1/drivers/input/input_cap12xx.c#L111

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=524767

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_INTEGER_OVERFLOW). ([CWE-190](http://cwe.mitre.org/data/definitions/190.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

fabiobaltieri, 2025-05-27:
@jeremydick can you take a look? does not make too much sense to me but maybe I'm missing something

jeremydick, 2025-05-27:
I suspect coverity doesn't know `calsens[i + j]` is restricted to the values 1, 2, or 4 by the devicetree binding. If it assuming `calsens[i + j]` could be any `uint8_t` value, 16 or larger would cause an overflow.

Limited to a maximum of 4 it won't overflow:
`
regval |= (ilog2(calsens[i + j]) << (CALSENS_BITS * j)) = (2 << 6) = 128
`

At 16 though it will overflow
`
regval |= (ilog2(calsens[i + j]) << (CALSENS_BITS * j)) = (4 << 6) = 256
`

jeremydick, 2025-05-27:
Yeah, I can go ahead and open a PR. I think the explicit check for calsens value being in the valid range is cleaner instead of accepting an out of bounds input and then masking the output of ilog2 to make it fit, so I'll do that

commit 621f29e3 (2025-05-27): drivers: input: cap12xx Fix integer overflow warning

Check that calsens values are valid to prevent integer overflow

Coverity-CID: 524767
Fixes #90481
closed 2025-05-27 by kartben