zephyrproject-rtos/zephyr #90502: [Coverity CID: 520288] Copy of overlapping memory in subsys/shell/backends/shell_mqtt.c

zephyrbot, 2025-05-25:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/265cfb45a818f39cf03c3527b5d1b21027d3cac1/subsys/shell/backends/shell_mqtt.c

Category: Memory - corruptions
Function: `init`
Component: Shell
CID: [520288](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=520288)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/265cfb45a818f39cf03c3527b5d1b21027d3cac1/subsys/shell/backends/shell_mqtt.c#L661

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=520288

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_OVERLAPPING_COPY). ()

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

ljd42, 2025-06-21:
Analysis
-----------
coverity reports a possible "copy of overlapping memory" for the mentioned line above. 

The reason behind this: `snprintf()` writes up to SH_MQTT_TOPIC_MAX_SIZE bytes (=10, which is the size of  sh->sub_topic array). But the address of sh->sub_topic is at 7 bytes offset from sh->device_id, so potentially causing a copy of overlapping memory. 

This doesn't happen however, because sh->device_id is an char array of 7 bytes, which is guaranteed to be '\0' terminated, see the function `shell_mqtt_get_devid`. The 3 "additional bytes" are coming from the "_rx" part of the format string "%s_rx". The memory regions copied do not overlap. 

TLDR;  It's a false positive.

ljd42, 2025-06-21:
@jakub-uC : If you agree with the previous analysis, I would mark the findings in Coverity as a false positive. 

Thanks!

jakub-uC, 2025-06-24:
@ljd42 yes, let's mark it as a false positive.

ljd42, 2025-06-24:
Marked in Coverity. Findings will be now dismissed.

@jakub-uC  or @kartben: Ticket can be closed with the appropriate labeling. Thanks!