zephyrproject-rtos/zephyr #90505: [Coverity CID: 520285] Out-of-bounds access in subsys/shell/backends/shell_mqtt.c

zephyrbot, 2025-05-25:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/265cfb45a818f39cf03c3527b5d1b21027d3cac1/subsys/shell/backends/shell_mqtt.c

Category: Memory - corruptions
Function: `client_init`
Component: Shell
CID: [520285](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=520285)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/265cfb45a818f39cf03c3527b5d1b21027d3cac1/subsys/shell/backends/shell_mqtt.c#L195

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=520285

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_OVERRUN). ([CWE-119](http://cwe.mitre.org/data/definitions/119.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

ljd42, 2025-06-18:
I think, it's a false positive on the `memset` idiom. I would mark this as such either in the code and/or document in coverity.

`
/* coverity[access_dbuff_const] */
 memset(client, 0, sizeof(*client));
`

@jakub-uC : what do you think?

jakub-uC, 2025-06-19:
@ljd42 While this looks like a false positive at first glance, I suspect it's a real and critical issue stemming from the build system.

The problem appears to be a `struct mqtt_client` size mismatch between the` subsys/net/lib/mqtt/` and `subsys/shell/mqtt/ `modules. It seems possible that if `CONFIG_MQTT_VERSION_5_0=y` is enabled, `net/lib/mqtt` builds with the larger struct, while `shell/mqtt` does not, due to a missing `Kconfig` dependency. This would directly cause the `memset` buffer overrun flagged by `Coverity`.

This points to a potential flaw in Zephyr's `Kconfig` dependency graph between these modules.

@carlescufi @cfriedt, could you or another build system expert please help verify this analysis? We need to determine the right way to fix this dependency.

So, to put the core question simply: Is it possible in a single Zephyr build for two modules to have a conflicting, compile-time size for the same struct due to their `Kconfig` dependencies? This appears to be exactly what's happening here.

If not me can mark it as a false positive.

ljd42, 2025-06-20:
Thanks @cfriedt  for checking. It's Chris-verified now :wink:

For the sake of testings, I've compiled the mqtt_publisher sample, with the SHELL_BACKEND_MQTT enabled for both MQTT 3.1.1 and 5.0. I have printed the size of the struct at compile time both in  `subsys/shell/backends/shell_mqtt.c` and `subsys/net/lib/mqtt/mqtt.c`, I am getting the same size. I've tested for ARM / intel / native_sim. Attached the output for native_sim. 

The size of the struct is 540 bytes on native_sim for MQTT5.0 vs 108 bytes for MQTT3.1.1. So it's not obvious where the 140 bytes struct size reported in coverity  are coming from. 

While we don't have a clear-cut proof, the evidences as of now points to a false positive. 

[mqtt_5.txt](https://github.com/user-attachments/files/20841727/mqtt_5.txt)
[mqtt3_1_1.txt](https://github.com/user-attachments/files/20841728/mqtt3_1_1.txt)

cfriedt, 2025-06-20:
@ljd42 - I was going to suggest printing, but it seemed trivial enough just by inspection.

Maybe there are some hints on the Coverity page about possible compiler flags that might have been used when the scan was run.

We can probably leave it as a false positive though - it would be smart for Coverity to check those.

ljd42, 2025-07-05:
AFAICS, Coverity seemed confused when a structure has conditional fields, the structure is used in two different compile units and the size is computed through sizeof(*ptr). This happens elsewhere, like #92613 or #92614. 

Marked as False Positive in Coverity.

closed 2025-07-08 by kartben