zephyrproject-rtos/zephyr #90516: [Coverity CID: 520271] Out-of-bounds read in subsys/bluetooth/host/classic/hfp_hf.c

zephyrbot, 2025-05-25:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/265cfb45a818f39cf03c3527b5d1b21027d3cac1/subsys/bluetooth/host/classic/hfp_hf.c

Category: Memory - illegal accesses
Function: `bt_hfp_hf_sco_accept`
Component: Bluetooth
CID: [520271](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=520271)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/265cfb45a818f39cf03c3527b5d1b21027d3cac1/subsys/bluetooth/host/classic/hfp_hf.c#L4053

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=520271

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_OVERRUN). ([CWE-125](http://cwe.mitre.org/data/definitions/125.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

lylezhu2012, 2025-05-29:
Hi @nashif , I cannot access the link https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=520271.

Could you share the analyse log with me?

Thalley, 2025-06-05:
[image]

lylezhu2012, 2025-06-06:
It has been fixed by https://github.com/zephyrproject-rtos/zephyr/pull/89062

closed 2025-06-06 by lylezhu2012

lylezhu2012, 2025-06-06:
This issue has been found several month ago. And I confirmed it has been fixed.
The following is main branch snapshot,

[image]

[image]

https://github.com/zephyrproject-rtos/zephyr/blame/b3775436289e12f2b7d9b06b3b2278a8fd72cb73/subsys/bluetooth/host/classic/hfp_ag.c#L3938-L3954

lylezhu2012, 2025-06-06:
From the description of the CID, it should be a false detection. The index of ACL should not more than the array size of `bt_hfp_hf_pool`, because the array size of `acl_conns` is same with `bt_hfp_hf_pool`.

To avoid the potential issue if the the array size of `acl_conns` is more than the array size of `bt_hfp_hf_pool`, create a issue https://github.com/zephyrproject-rtos/zephyr/issues/91172 to fix the potential issue.

Thalley, 2025-06-06:
> Oops, my mistake. I did not realise I was referring the wrong function. Let me create an issue for it.

@lylezhu2012 please don't create duplicate issues. This issue already exists; there's no reason to create another issue for the same thing.