zephyrproject-rtos/zephyr #90517: [Coverity CID: 520269] Unintentional integer overflow in drivers/sensor/bosch/bma4xx/bma4xx_emul.c

zephyrbot, 2025-05-25:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/265cfb45a818f39cf03c3527b5d1b21027d3cac1/drivers/sensor/bosch/bma4xx/bma4xx_emul.c

Category: Integer handling issues
Function: `bma4xx_emul_set_accel_data`
Component: Drivers
CID: [520269](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=520269)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/265cfb45a818f39cf03c3527b5d1b21027d3cac1/drivers/sensor/bosch/bma4xx/bma4xx_emul.c#L206

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=520269

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_OVERFLOW_BEFORE_WIDEN). ([CWE-190](http://cwe.mitre.org/data/definitions/190.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

ldomaigne, 2025-06-12:
The `data->regs[BMA4XX_REG_ACCEL_RANGE]` is an `uint8_t`, but only the first two bits is used per BMA456 spec (chap 5, register 0x41 ACC_RANGE). This fact is correctly implemented in `bma4xx_emul_write_byte` and ` bma4xx_emul_backend_get_sample_range`.  Therefore, we can't have an integer overflow. From my understanding, it's a false positive.

We could either document it in Coverity, or change the line as suggested below with the hope that coverity can infer that we have at a shift of 3 max:
```
 int64_t accel_range = (2 << (data->regs[BMA4XX_REG_ACCEL_RANGE] & GENMASK(1,0)  ); 
```
@MaureenHelm : what is your preferred approach?

ljd42, 2025-06-17:
@kartben: good first issue! 

No overflow is current happening in code, improvement suggested by Coverity is straightforward to implement. It's a good first issue to get familiar with Zephyr's contribution workflow.

---
Here's the improvement suggested by Coverity:

https://github.com/zephyrproject-rtos/zephyr/blob/265cfb45a818f39cf03c3527b5d1b21027d3cac1/drivers/sensor/bosch/bma4xx/bma4xx_emul.c#L206

CID 520269: (#1 of 1): Unintentional integer overflow (OVERFLOW_BEFORE_WIDEN)
overflow_before_widen: Potentially overflowing expression 2 << data->regs[65] with type int (32 bits, signed) is evaluated 
using 32-bit arithmetic, and then used in a context that expects an expression of type int64_t (64 bits, signed).  

To avoid overflow, cast 2 to type int64_t.

PR #92496 (2025-07-04): drivers: sensor: bma4xx: Avoid potential overflow
Coverity reports a potential integer overflow in the accel_range computation due to the use of a left shift on an int type.

CID 520269: Unintentional integer overflow (OVERFLOW_BEFORE_WIDEN)

Even though the register value is constrained to 0–3 by the BMA456 spec, and no real overflow occurs, an explicit cast to int64_t prevents false positives and aligns with safe coding practices.

Fixes: #90517
commit 9207749e (2025-07-04): drivers: sensor: bma4xx: Avoid potential overflow

Coverity reports a potential integer overflow in the accel_range
computation due to the use of a left shift on an int type.

CID 520269: Unintentional integer overflow (OVERFLOW_BEFORE_WIDEN)

Even though the register value is constrained to 0–3 by the BMA456 spec,
and no real overflow occurs, an explicit cast to int64_t prevents false
positives and aligns with safe coding practices.

Fixes: #90517
closed 2025-07-04 by danieldegrasse