zephyrproject-rtos/zephyr #90518: [Coverity CID: 520267] Overflowed integer argument in subsys/modem/modem_cmux.c

zephyrbot, 2025-05-25:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/265cfb45a818f39cf03c3527b5d1b21027d3cac1/subsys/modem/modem_cmux.c

Category: Insecure data handling
Function: `modem_cmux_transmit_frame`
Component: Other
CID: [520267](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=520267)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/265cfb45a818f39cf03c3527b5d1b21027d3cac1/subsys/modem/modem_cmux.c#L305

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=520267

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_INTEGER_OVERFLOW). ([CWE-190](http://cwe.mitre.org/data/definitions/190.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

ljd42, 2025-06-18:
Looking at the coverity report and the code, I think it's a false positive.
Coverity reports a possible cast overflow here: 
https://github.com/zephyrproject-rtos/zephyr/blob/265cfb45a818f39cf03c3527b5d1b21027d3cac1/subsys/modem/modem_cmux.c#L276-L277

We're computing `MIN(space, frame->data_len,CONFIG_MODEM_CMUX_MTU)`. Here `space` and `frame->data_len` are uint16_t (ok), but the macro `CONFIG_MODEM_CMUX_MTU` from autoconf.h will be treated as an int. 

As per Kconfig: 16 <=  CONFIG_MODEM_CMUX_MTU <= 1500
So there's no cast overflow / overflow assign happening. 

If my analysis is correct, we could either:
a) document it in coverity, 
or 
b) use an explicit cast as follows: 
`data_len = MIN(data_len, (uint16_t) CONFIG_MODEM_CMUX_MTU);`

I would add probably add a compile-time assert on CONFIG_MODEM_CMUX_MTU. Not that I'm expecting modem with a MTU > 64Kb any time soon. But it would help to catch invalid CONFIG MTU just in case, as using the (uint16_t) cast would prevent the compiler to catch arithmetic overflow.

ljd42, 2025-06-18:
@bjarki-andreasen : See above analysis. Any feedback or preferences?