zephyrproject-rtos/zephyr #90542: [Coverity CID: 516234] Dereference after null check in subsys/bluetooth/host/classic/sdp.c

zephyrbot, 2025-05-25:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/265cfb45a818f39cf03c3527b5d1b21027d3cac1/subsys/bluetooth/host/classic/sdp.c

Category: Null pointer dereferences
Function: `sdp_client_discover`
Component: Bluetooth
CID: [516234](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=516234)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/265cfb45a818f39cf03c3527b5d1b21027d3cac1/subsys/bluetooth/host/classic/sdp.c#L1907

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=516234

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_FORWARD_NULL). ([CWE-476](http://cwe.mitre.org/data/definitions/476.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

Thalley, 2025-06-05:
[image]

lylezhu2012, 2025-06-06:
I have created issue https://github.com/zephyrproject-rtos/zephyr/issues/91156 for it.

lylezhu2012, 2025-06-06:
> You could just have referred to this issue instead :) You also didn't follow the Bug template in the new issue...

I cannot provide information to fill the template. So I cleared the empty items. Do all these blank items need to be kept? If so, I can follow it.

lylezhu2012, 2025-07-03:
It has been fixed by https://github.com/zephyrproject-rtos/zephyr/pull/91158

closed 2025-07-03 by lylezhu2012