zephyrproject-rtos/zephyr #90567: STM32N6 USB PCD_Init Failed

james-advatek, 2025-05-26:
**Describe the bug**
When attempting to run the USB shell or CDC_ACM USB samples (either using the old or new USB device stacks) on the STM32N657-DK (stm32n6570_dk) or NUCLEO-N657X0-Q (nucleo_n657x0_q) boards, the USB initialization fails and prints `PCD_Init failed, 1` to the console.

**To Reproduce**
Run `west build -b stm32n6570_dk samples/subsys/usb/cdc_acm` or ` west build -b nucleo_n657x0_q samples/subsys/usb/cdc_acm`

**Expected behavior**
The USB samples successfully run and the STM32N6 USB device properly initializes.

**Attempted workarounds**
- Tried putting the usb device into full speed mode.
- Used KConfig option `CONFIG_USB_DC_STM32_CLOCK_CHECK=y` to verify that the USB clock was set correctly.
- Tried running on both the Discovery Kit and Nucleo boards.
- Some debugging suggests that the issue is caused by `HAL_TIMEOUT` being returned [on line 1387 of stm32n6xx_ll_usb.c.](https://github.com/zephyrproject-rtos/hal_stm32/blob/5cbc642b1a79d4f373b1587f8c3027f31bf0d30c/stm32cube/stm32n6xx/drivers/src/stm32n6xx_ll_usb.c#L1387C1-L1387C26)
- Tried to add delays as mentioned [in this ST Community post](https://community.st.com/t5/stm32-mcus-embedded-software/stm32n6-getting-stuck-in-usb-corereset/m-p/790748)
- Tried running the equivalent sample for ThreadX (provided by ST) which was able to successfully enumerate the device.

**Impact**
High

**Logs and console output**
```
[00:00:00.001,000] <dbg> usb_descriptor: ascii7_to_utf16le: idx_max 31, ascii_idx_max 15, buf 0x3419b91d
[00:00:00.001,000] <dbg> usb_descriptor: ascii7_to_utf16le: char E : 45, idx 15 -> 31
[00:00:00.001,000] <dbg> usb_descriptor: ascii7_to_utf16le: char 4 : 34, idx 14 -> 29
[00:00:00.001,000] <dbg> usb_descriptor: ascii7_to_utf16le: char 2 : 32, idx 13 -> 27
[00:00:00.001,000] <dbg> usb_descriptor: ascii7_to_utf16le: char 3 : 33, idx 12 -> 25
[00:00:00.001,000] <dbg> usb_descriptor: ascii7_to_utf16le: char 6 : 36, idx 11 -> 23
[00:00:00.001,000] <dbg> usb_descriptor: ascii7_to_utf16le: char 3 : 33, idx 10 -> 21
[00:00:00.001,000] <dbg> usb_descriptor: ascii7_to_utf16le: char 0 : 30, idx 9 -> 19
[00:00:00.001,000] <dbg> usb_descriptor: ascii7_to_utf16le: char 0 : 30, idx 8 -> 17
[00:00:00.001,000] <dbg> usb_descriptor: ascii7_to_utf16le: char E : 45, idx 7 -> 15
[00:00:00.001,000] <dbg> usb_descriptor: ascii7_to_utf16le: char 0 : 30, idx 6 -> 13
[00:00:00.001,000] <dbg> usb_descriptor: ascii7_to_utf16le: char 0 : 30, idx 5 -> 11
[00:00:00.001,000] <dbg> usb_descriptor: ascii7_to_utf16le: char 5 : 35, idx 4 -> 9
[00:00:00.001,000] <dbg> usb_descriptor: ascii7_to_utf16le: char 6 : 36, idx 3 -> 7
[00:00:00.001,000] <dbg> usb_descriptor: ascii7_to_utf16le: char 3 : 33, idx 2 -> 5
[00:00:00.001,000] <dbg> usb_descriptor: ascii7_to_utf16le: char 2 : 32, idx 1 -> 3
[00:00:00.001,000] <dbg> usb_descriptor: ascii7_to_utf16le: char 4 : 34, idx 0 -> 1
[00:00:00.001,000] <dbg> usb_dc_stm32: usb_dc_set_status_callback: 
[00:00:00.001,000] <dbg> usb_dc_stm32: usb_dc_attach: 
[00:00:00.001,000] <dbg> usb_dc_stm32: usb_dc_stm32_phy_specific_clock_enable: CLOCK CHECK
[00:00:00.001,000] <dbg> usb_dc_stm32: usb_dc_stm32_init: HAL_PCD_Init
[00:00:46.571,000] <err> usb_dc_stm32: PCD_Init failed, 1
[00:00:46.571,000] <dbg> usb_device: usb_enable: unlock usb_enable_lock mutex
*** Booting Zephyr OS build v4.1.0-3816-g958ef2288e01 ***
```

**Environment:**
 - OS: Linux (via WSL)
 - Toolchain: Zephyr SDK
 - Zephyr: v4.1.0

marwaiehm-st, 2025-05-26:
Hi @james-advatek 
Could you please try testing with this [PR](https://github.com/zephyrproject-rtos/zephyr/issues/90567) ?

james-advatek, 2025-05-26:
Hi @marwaiehm-st, thanks so much! I managed to get it working by cherry picking your PR on top of my manifest-rev HEAD.

When I tried using your branch, I got the following error: 

```
...../zephyr/include/zephyr/arch/arm/asm_inline_gcc.h:24:10: fatal error: cmsis_core.h: No such file or directory
   24 | #include <cmsis_core.h>
      |          ^~~~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
```

I'm guessing this isn't related to your PR though, and is due to another upstream change.

Should I close this issue now, or wait for the PR to be merged?

mikk-leini-krakul, 2025-05-29:
> ...../zephyr/include/zephyr/arch/arm/asm_inline_gcc.h:24:10: fatal error: cmsis_core.h: No such file or directory
>    24 | #include <cmsis_core.h>
>       |          ^~~~~~~~~~~~~~
> compilation terminated.

Maybe you had same case? https://github.com/zephyrproject-rtos/zephyr/pull/90063#issuecomment-2918533631