zephyrproject-rtos/zephyr #90675: I2C RTIO stream issue/limitation with STM32

avisconti, 2025-05-27:
**Describe the bug**

**To Reproduce**

**Expected behavior**

**Impact**

**Logs and console output**

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**

avisconti, 2025-05-27:
> > > I think that's expected, as noted in the PR's description [#87731](https://github.com/zephyrproject-rtos/zephyr/pull/87731):
> > > > It works fine with the sensor_read and sensor_get_decoder functions but there is currently a limitation with the sensor_stream function (used for example in sensor/accel_polling and sensor/stream_fifo samples).
> > 
> > 
> > Yes, of course. I already discussed with committer and we together decided to open the issue anyway, just to keep track of it. Maybe, a better choice would have been "enhancement".
> 
> Okey then, but the wording of the issue is a little bit confusing, `breaking RTIO streaming capability on my STM32 board` makes it seem like streaming was well supported and that commit broke it :)

Well, actually the i2c driver was perfectly supporting rtio before this commit. I think that the previous stm32 i2c driver was previously falling in the rtio default callback.

dkalowsk, 2025-05-27:
@teburd FYI

teburd, 2025-05-27:
Is this because it requires a reload to occur in st's terms? yeah that I didn't get working and had trouble with on my nucleo 446re board from what I recall (it was awhile ago now!)

The fram/sram test could be extended to cover this pretty easily (large read or write) and I did do that when trying to develop the driver. I'd recommend that.

etienne-lms, 2025-06-16:
Hi @avisconti,
I'm slowly progressing but still progressing. The missing part is related to transfer of more than 256 bytes.
I think I have a decent fix. I'll create a P-R straight away.

etienne-lms, 2025-06-16:
For info, the STM32 RTIO drivers still lacks target mode support as well as DMA and some SMBUS support that I've not looked at yet.

avisconti, 2025-06-16:
> Hi [@avisconti](https://github.com/avisconti), I'm slowly progressing but still progressing. The missing part is related to transfer of more than 256 bytes. I think I have a decent fix. I'll create a P-R straight away.
> 
> (edited: see [#91664](https://github.com/zephyrproject-rtos/zephyr/pull/91664))

Thanks Etienne. Sorry for pushing. The fact is that every time I rebase my code it does not work for this issue, and I spend half day debugging to finally remember of this issue. I appreciate your effort! :D

etienne-lms, 2025-06-27:
Can be closed since https://github.com/zephyrproject-rtos/zephyr/pull/91664 and https://github.com/zephyrproject-rtos/zephyr/pull/91926 have been merged?

erwango, 2025-06-27:
> Can be closed since [#91664](https://github.com/zephyrproject-rtos/zephyr/pull/91664) and [#91926](https://github.com/zephyrproject-rtos/zephyr/pull/91926) have been merged?

Sure.

closed 2025-06-27 by erwango