zephyrproject-rtos/zephyr #90714: samples: counter: sample.drivers.counter.alarm : sample missing dts dependency

hakehuang, 2025-05-28:
### Describe the bug

- Platforms
all NXP platform has such issue
- git bisect found below commit introduce problem
```
022962c90d75c871bc0aa4ba100249ce34c25798
samples: counter: alarm: remove platform_allow filterring

Because this is a common testing scenario, almost any counter instance
should be able to run. Some counter instances cannot run this sample
should be filtered by platform_exclude or filter keywords.

```

this sample need dts be defined for different platform, which need get aligned in dts level, however this code sets to be platform dependent. 

### Regression

- [x] This is a regression.

### Steps to reproduce

```
west scripts/twister -p frdm_k82f -j 12 -T samples/drivers/counter/alarm --build-only
```

### Relevant log output

```shell
/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/include/zephyr/device.h:96:41: note: each undeclared identifier is reported only once for each function it appears in 96 | #define DEVICE_NAME_GET(dev_id) _CONCAT(__device_, dev_id) | ^~~~~~~~~/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/include/zephyr/toolchain/common.h:168:26: note: in definition of macro '_DO_CONCAT' 168 | #define _DO_CONCAT(x, y) x ## y | ^/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/include/zephyr/device.h:96:33: note: in expansion of macro '_CONCAT' 96 | #define DEVICE_NAME_GET(dev_id) _CONCAT(__device_, dev_id) | ^~~~~~~/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/include/zephyr/device.h:293:37: note: in expansion of macro 'DEVICE_NAME_GET' 293 | #define DEVICE_DT_NAME_GET(node_id) DEVICE_NAME_GET(Z_DEVICE_DT_DEV_ID(node_id)) | ^~~~~~~~~~~~~~~/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/include/zephyr/device.h:310:34: note: in expansion of macro 'DEVICE_DT_NAME_GET' 310 | #define DEVICE_DT_GET(node_id) (&DEVICE_DT_NAME_GET(node_id)) | ^~~~~~~~~~~~~~~~~~/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/samples/drivers/counter/alarm/src/main.c:117:50: note: in expansion of macro 'DEVICE_DT_GET' 117 | const struct device *const counter_dev = DEVICE_DT_GET(TIMER); | ^~~~~~~~~~~~~ninja: build stopped: subcommand failed.
```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

### Environment

- Sha: v4.1.0-4825-g3e8ec3aaf299

kartben, 2025-05-28:
@hakehuang can you please make sure that the bugs you open have environment information that correspond to Zephyr upstream? g3e8ec3aaf299 is NOT a Zephyr commit, and every time this is causing a lot of churn during the triaging / qualification phase.

hakehuang, 2025-05-29:
> [@hakehuang](https://github.com/hakehuang) can you please make sure that the bugs you open have environment information that correspond to Zephyr upstream? g3e8ec3aaf299 is NOT a Zephyr commit, and every time this is causing a lot of churn during the triaging / qualification phase.

@kartben, strange I can see the commit https://github.com/zephyrproject-rtos/zephyr/commit/3e8ec3aaf299ce1a59217a1321d8f966effa5da6
the version is get from 'git describe' and the format is <latest tag>-g<11 number of sha>.

kartben, 2025-05-29:
> > [@hakehuang](https://github.com/hakehuang) can you please make sure that the bugs you open have environment information that correspond to Zephyr upstream? g3e8ec3aaf299 is NOT a Zephyr commit, and every time this is causing a lot of churn during the triaging / qualification phase.
> 
> @kartben, strange I can see the commit https://github.com/zephyrproject-rtos/zephyr/commit/3e8ec3aaf299ce1a59217a1321d8f966effa5da6
> the version is get from 'git describe' and the format is 'latest release tag'-g<11 number of sha>.

You are right, I must apologize -- copy paste issue on my side and I had included the leading "g" 🤦🏼‍♂️

hakehuang, 2025-05-29:
> You are right, I must apologize -- copy paste issue on my side and I had included the leading "g" 🤦🏼‍♂️

thanks for confirm. I made the same mistake before Anas told me.

kartben, 2025-05-29:
> > You are right, I must apologize -- copy paste issue on my side and I had included the leading "g" 🤦🏼‍♂️
> 
> thanks for confirm. I made the same mistake before Anas told me. 

Can you please open a PR reverting 022962c90d75c871bc0aa4ba100249ce34c25798
It probably shouldn't have gone in in the first place. 

Thanks!

hakehuang, 2025-05-29:
> Can you please open a PR reverting [022962c](https://github.com/zephyrproject-rtos/zephyr/commit/022962c90d75c871bc0aa4ba100249ce34c25798)
> It probably shouldn't have gone in in the first place.

sure. https://github.com/zephyrproject-rtos/zephyr/pull/90798 @kartben

nashif, 2025-05-29:
> You are right, I must apologize -- copy paste issue on my side and I had included the leading "g"

you can also do `git show v4.1.0-4825-g3e8ec3aaf299`, this is valid as well :)

commit 11c543e2 (2025-05-31): Revert "samples: counter: alarm: remove platform_allow filterring"

this sample needs add dts defines for each platform, so should not be
generally enabled.

fixes: #90714

This reverts commit 022962c90d75c871bc0aa4ba100249ce34c25798.