zephyrproject-rtos/zephyr #90750: Scaling of board overlay mechanism

PerMac, 2025-05-28:
**Is your enhancement proposal related to a problem? Please describe.**
There are thousands of overlay files located under sample/* and tests/* ...  */boards/. E.g. https://github.com/zephyrproject-rtos/zephyr/tree/main/tests/drivers/uart/uart_async_api/boards (142 files)
https://github.com/zephyrproject-rtos/zephyr/tree/main/tests/drivers/dma/chan_blen_transfer/boards (173 files).
Every time someone wants to add a support of such tests on another board they have to add extra board specific overlays. In many cases such overlays are yet another copy of the same file, just with a different name.
Such mechanism doesn't scale.

**Additional context**
We discussed the toping during testing wg meetings and we don't know how it can be changed. Maybe there won't be a single solution, but a combination of several, including rework of some tests. One approach is proposed by @hakehuang which allows for reuse  of the same overlay files by multiple boards, at the cost of adding entries in test yaml responsible for mapping such boards with more general overlays https://github.com/zephyrproject-rtos/zephyr/blob/main/tests/drivers/uart/uart_async_api/testcase.yaml#L87.  However, it can significantly increase complexity of test yaml resulting with tests with hundreds of namespaced configs.
Other idea is to moved  some implementation of overlays  to a board level, as a kind of an extension, that could be added whenever such overlay is requested by a test?

As a testing working group we wanted to flag the issue of "exploding" overlays and ask other maintainers for their take on it and possible solutions.

PerMac, 2025-05-28:
cc @nashif @hakehuang

henrikbrixandersen, 2025-06-03:
> > [@hakehuang](https://github.com/hakehuang) Could you please elaborate a bit on this suggestion?
> 
> 1. the pin mux change to a give test case usually is based on board design, so such chagnes shall bind with board not soc.
> 2. clock settings are both board and soc related, so such changes also bind with board and soc.
> 3. peripheral IP settings are soc based. but its system features such as interrupt number and dma channel are usually soc bind.
>    So we can't move all application/test related settings to soc.
>    My suggestion would be split them to 3 parts boad,soc,ip, and create an overlay sequences board over soc over ip.

How will this reduce the number of overlay files needed?

hakehuang, 2025-06-03:
> How will this reduce the number of overlay files needed?

take this commit as example, 
https://github.com/zephyrproject-rtos/zephyr/commit/513e45f60d38444b04c11b5154a3973a0db491c7
this is just partial down, original <board>.overlay is change to per-IP combiantion. and with this, we clearly see what IP and which instance we are testing.

ajf58, 2025-06-04:
> > We might also need to rethink of how overlays are applied, possibly using SoC overlays
> 
> This is a good solution for most STM32 boards.

This would also work for RP2040, and RP235xx SoCs.

It's also been proposed before, e.g. use some heirachy of directories, e.g.:
```
soc_series/foo.overlay
soc/foo12A.overlay
```
At the moment I misuse(?) the pre-processor, and #include one overlay from the others.

nordic-segl, 2025-10-07:
@nashif wrote
> Another option would be to allow board to define snippets that can be reused by many tests, so we do not need to provide an overlay for each tests/sample.

I did POC here (define snippet for test purposes) but it has several major drawbacks
https://github.com/nrfconnect/sdk-nrf/pull/24853

johnlange2, 2026-02-24:
Hi all,

Following a recent Testing WG discussion on overlay duplication in the test tree, I put together a proposal focused specifically on reducing redundant .overlay (and optionally .conf) files while keeping Twister and existing tests working.

**Motivation (why this is separate from “reduce overlays overall”)**
Even if we continue the broader effort to reduce overlay usage, the current test tree still has a large amount of pure duplication that creates maintenance cost and inconsistency risk. A quick duplication analysis found:
•	.overlay: 5,858 total, 1,918 duplicates (32.74%)
•	.conf: 4,400 total, 1,431 duplicates (32.52%)
•	References to update: ~1,400-1,500 files across testcase.yaml, a few CMakeLists, docs, and scripts 

**High-level approach**
1.	Hierarchical overlay discovery in the build system (CMake): extend overlay search so tests can inherit overlays from parent directories and shared locations, with “most specific wins” precedence. Backward compatible by keeping current local boards/ behavior highest priority. 
2.	Four-tier placement model (where overlays live):
o	Board-level (only when it’s truly board-hardware shared broadly)
o	Test-general shared overlays
o	Test-category shared overlays
o	Test-specific overlays (current per-test behavior) 
3.	Phased migration: start with a small pilot (DMA suggested in the plan), validate via Twister/CI, then expand category by category. Optional follow-on: apply similar dedupe patterns to .conf files. 

**Expected impact**
The intent is to collapse many “identical content but copied everywhere” overlays into a single source per board (or per category) without changing test behavior, and to reduce the number of tests/**/boards/ directories that exist only to hold duplicates. The plan targets a 70-90% reduction in overlay file count (from ~5.8k to ~1.5-1.8k), subject to review of what must remain test-specific. 

**Request for feedback**
•	Is the hierarchical discovery direction acceptable in principle, or would it be considered too much build-system complexity for the value?
•	Are there strong objections to any of the proposed shared locations (for example, anything under boards/<vendor>/<board>/), even as an optional later phase?
•	If this direction seems viable, I can follow up with a small proof-of-concept PR (build-system + one pilot subtree) and then turn the full plan into an RFC.

The full plan is attached. Thanks for taking a look.
John

[OVERLAY_HARMONIZATION_PLAN.md](https://github.com/user-attachments/files/25503851/OVERLAY_HARMONIZATION_PLAN.md)

aescolar, 2026-02-24:
CC @tejlmand @nordicjm

johnlange2, 2026-02-25:
Thanks for the candid feedback @nordicjm. 

For context, I created this after a recent Testing WG discussion in which @nashif discussed the large number of duplicate files. I ran the proposal by him and @hakehuang separately before posting here. 

I agree repo size isn’t the driver. The motivation was maintenance and avoiding divergence across many identical copies.

That said, your concern about added confusion is valid. I’d really appreciate learning more about @tejlmand’s suggested approach. If the preferred direction is “keep duplicates but manage them better,” I’m happy with that approach and could perhaps help with a CI/tooling check to detect drift among duplicates, or other low-risk improvements, rather than pushing hierarchical discovery.

hakehuang, 2026-03-05:
is it feasible that we create a samples/v2/ and tests/v2 folder and use this as sandbox to trial how to reduce the existing meta-data?
we can start move exiting applications into this folder, and define/demo a clear way to community.