zephyrproject-rtos/zephyr #90777: arch.shared_interrupt.lto.speed compile fail on apollo5 platform

swift-tk, 2025-05-29:
### Describe the bug

The compile fail by  west build -p always -b apollo510_evb tests/kernel/interrupt -T arch.shared_interrupt.lto.speed
The fail comes and goes in the CI/CD for some time now after the addition of apollo510 soc. CI link : https://github.com/zephyrproject-rtos/zephyr/actions/runs/15132639765/job/42545412904?pr=89074

The fail seems to related to `CONFIG_SPEED_OPTIMIZATIONS` as `arch.shared_interrupt.lto` would not fail.

### Regression

- [ ] This is a regression.

### Steps to reproduce

 west build -p always -b apollo510_evb tests/kernel/interrupt -T arch.shared_interrupt.lto.speed

### Relevant log output

```shell
The latest complete fail log from upstream commit: af46f628a0c5b21fca3ac4a78640059f3cb19442

$ west build -p always -b apollo510_evb tests/kernel/interrupt -T arch.shared_interrupt.lto.speed
-- west build: making build dir C:\work\zephyrproject\zephyr\build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: C:/work/zephyrproject/zephyr/tests/kernel/interrupt
-- CMake version: 3.31.4
-- Found Python3: C:/Users/swift.tian/AppData/Local/miniconda3/envs/zephyr/python.exe (found suitable version "3.11.0", minimum required is "3.10") found components: Interpreter
-- Cache files will be written to: C:/work/zephyrproject/zephyr/.cache
-- Zephyr version: 4.1.99 (C:/work/zephyrproject/zephyr)
-- Found west (found suitable version "1.3.0", minimum required is "0.14.0")
-- Board: apollo510_evb, qualifiers: apollo510
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.17.0 (C:/work/zephyr-sdk-0.17.0)
-- Found toolchain: zephyr 0.17.0 (C:/work/zephyr-sdk-0.17.0)
-- Found Dtc: C:/ProgramData/chocolatey/bin/dtc.exe (found suitable version "1.6.1", minimum required is "1.4.6")
-- Found BOARD.dts: C:/work/zephyrproject/zephyr/boards/ambiq/apollo510_evb/apollo510_evb.dts
-- Generated zephyr.dts: C:/work/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated pickled edt: C:/work/zephyrproject/zephyr/build/zephyr/edt.pickle
-- Generated devicetree_generated.h: C:/work/zephyrproject/zephyr/build/zephyr/include/generated/zephyr/devicetree_generated.h
Parsing C:/work/zephyrproject/zephyr/tests/kernel/interrupt/Kconfig
Loaded configuration 'C:/work/zephyrproject/zephyr/boards/ambiq/apollo510_evb/apollo510_evb_defconfig'
Merged configuration 'C:/work/zephyrproject/zephyr/tests/kernel/interrupt/prj.conf'
Merged configuration 'C:/work/zephyrproject/zephyr/build/zephyr/misc/generated/extra_kconfig_options.conf'
Configuration saved to 'C:/work/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to 'C:/work/zephyrproject/zephyr/build/zephyr/include/generated/zephyr/autoconf.h'
-- Found GnuLd: c:/work/zephyr-sdk-0.17.0/arm-zephyr-eabi/arm-zephyr-eabi/bin/ld.bfd.exe (found version "2.38")
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/work/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc.exe
-- Found gen_kobject_list: C:/work/zephyrproject/zephyr/scripts/build/gen_kobject_list.py
CMake Warning at C:/work/zephyrproject/zephyr/CMakeLists.txt:1021 (message):
  No SOURCES given to Zephyr library: modules_mctp

  Excluding target from build.

-- Configuring done (141.1s)
-- Generating done (0.9s)
-- Build files have been written to: C:/work/zephyrproject/zephyr/build
-- west build: building application
[1/175] Generating include/generated/zephyr/version.h
-- Zephyr version: 4.1.99 (C:/work/zephyrproject/zephyr), build: zephyr-v3.4.0-35853-gaf46f628a0c5
[170/175] Linking C executable zephyr\zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map C:/work/zephyrproject/zephyr/build/zephyr/zephyr_pre0.map
C:\WINDOWS\system32\cmd.exe /C "cd . && C:\work\zephyr-sdk-0.17.0\arm-zephyr-eabi\bin\arm-zephyr-eabi-gcc.exe  -gdwarf-4 -flto=auto -fno-ipa-sra -ffunction-sections -fdata-sections ze
phyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr\zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -T  zephyr/linker_zephyr_pre0.cmd
  -Wl,-Map=C:/work/zephyrproject/zephyr/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm
/core/libarch__arm__core.a  zephyr/arch/arch/arm/core/cortex_m/libarch__arm__core__cortex_m.a  zephyr/arch/arch/arm/core/cortex_m/cmse/libarch__arm__core__cortex_m__cmse.a  zephyr/arc
h/arch/arm/core/mpu/libarch__arm__core__mpu.a  zephyr/lib/libc/picolibc/liblib__libc__picolibc.a  zephyr/lib/libc/common/liblib__libc__common.a  zephyr/boards/ambiq/apollo510_evb/libb
oards__ambiq__apollo510_evb.a  zephyr/subsys/testsuite/ztest/libsubsys__testsuite__ztest.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a
 zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  modules/hal_ambiq/mcu/apollo510/lib..__modules__hal__ambiq__mcu__apollo510.a  -Wl,--no-whole-arc
hive  zephyr/kernel/libkernel.a  -LC:/work/zephyrproject/zephyr/build/zephyr  zephyr/arch/common/libisr_tables.a  -mcpu=cortex-m55  -mthumb  -mabi=aapcs  -mfp16-format=ieee  -mtp=soft
  -fuse-ld=bfd  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -
Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -Wl,-no-pie  -specs=picolibc.specs  -DPICOLIBC_LONG_LONG_PRINTF_SCANF -L"c:/work/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr
-eabi/12.2.0/thumb/v8-m.main/nofp" -lc -lgcc && C:\WINDOWS\system32\cmd.exe /C "cd /D C:\work\zephyrproject\zephyr\build\zephyr && "C:\Program Files\CMake\bin\cmake.exe" -E true""
C:\Users\SWIFT~1.TIA\AppData\Local\Temp\ccfurDVF.s: Assembler messages:
C:\Users\SWIFT~1.TIA\AppData\Local\Temp\ccfurDVF.s:40554: Error: offset out of range
C:\Users\SWIFT~1.TIA\AppData\Local\Temp\ccfurDVF.s:40560: Error: offset out of range
make: *** [C:\Users\SWIFT~1.TIA\AppData\Local\Temp\ccEzJS9d.mk:2: C:\Users\SWIFT~1.TIA\AppData\Local\Temp\ccqoxWTT.ltrans0.ltrans.o] Error 1
lto-wrapper.exe: fatal error: make returned 2 exit status
compilation terminated.
c:/work/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd.exe: error: lto-wrapper failed
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' --build 'C:\work\zephyrproject\zephyr\build'

We were able to find a workaround by changing 
am_hal_pwrctrl_temp_thresh_t dummy; ---> am_hal_pwrctrl_temp_thresh_t dummy[32]; at https://github.com/zephyrproject-rtos/zephyr/blob/af46f628a0c5b21fca3ac4a78640059f3cb19442/soc/ambiq/apollo5x/soc.c#L51 or just add more function calls to `soc_early_init_hook`.  Then the error could get ignored after compile.

-- west build: building application
[1/175] Generating include/generated/zephyr/version.h
-- Zephyr version: 4.1.99 (C:/work/zephyrproject/zephyr), build: zephyr-v3.4.0-35853-gaf46f628a0c5
[170/175] Linking C executable zephyr\zephyr_pre0.elf
make: [C:\Users\SWIFT~1.TIA\AppData\Local\Temp\ccg2MnRA.mk:3: C:\Users\SWIFT~1.TIA\AppData\Local\Temp\ccY0t2NO.ltrans0.ltrans.o] Error 1 (ignored)
make: [C:\Users\SWIFT~1.TIA\AppData\Local\Temp\ccg2MnRA.mk:6: C:\Users\SWIFT~1.TIA\AppData\Local\Temp\ccY0t2NO.ltrans1.ltrans.o] Error 1 (ignored)
make: [C:\Users\SWIFT~1.TIA\AppData\Local\Temp\ccg2MnRA.mk:9: C:\Users\SWIFT~1.TIA\AppData\Local\Temp\ccY0t2NO.ltrans2.ltrans.o] Error 1 (ignored)
[175/175] Linking C executable zephyr\zephyr.elf
make: [C:\Users\SWIFT~1.TIA\AppData\Local\Temp\cc94iGoT.mk:3: C:\Users\SWIFT~1.TIA\AppData\Local\Temp\ccWtIQUV.ltrans0.ltrans.o] Error 1 (ignored)
make: [C:\Users\SWIFT~1.TIA\AppData\Local\Temp\cc94iGoT.mk:6: C:\Users\SWIFT~1.TIA\AppData\Local\Temp\ccWtIQUV.ltrans1.ltrans.o] Error 1 (ignored)
make: [C:\Users\SWIFT~1.TIA\AppData\Local\Temp\cc94iGoT.mk:9: C:\Users\SWIFT~1.TIA\AppData\Local\Temp\ccWtIQUV.ltrans2.ltrans.o] Error 1 (ignored)
```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

### Environment

- OS Windows
- Toolchain: Zephyr SDK 0.17.0
- commit : af46f628a0c5b21fca3ac4a78640059f3cb19442

swift-tk, 2025-05-30:
The error lines at  [zephyr_pre0.elf.ltrans0.ltrans.txt](https://github.com/user-attachments/files/20519659/zephyr_pre0.elf.ltrans0.ltrans.txt)

zephyr/zephyr_pre0.elf.ltrans0.ltrans.s:40495: Error: offset out of range
zephyr/zephyr_pre0.elf.ltrans0.ltrans.s:40501: Error: offset out of range

Shows
```
@ 546 "/home/swift/zephyrproject/zephyr/arch/arm/core/cortex_m/thread.c" 1
	mov   r4,  r10
msr   PSP, r9
movs  r0,  #0
ldr   r3, =arch_irq_unlock_outlined
blx   r3
mov   r0, r4
movs  r1, #0
movs  r2, #0
movs  r3, #0
ldr   r4, =z_thread_entry
bx    r4
```
arch_irq_unlock_outlined and z_thread_entry seems to be the cultprit....

swift-tk, 2025-06-06:
I'm putting this as priority high as it has blocked all our PRs due to CI fail.

swift-tk, 2025-06-06:
I have a another workaround here https://github.com/zephyrproject-rtos/zephyr/pull/90909. It is most viable but still blocked...

stephanosio, 2025-07-03:
```
C:\Users\SWIFT~1.TIA\AppData\Local\Temp\ccfurDVF.s: Assembler messages:
C:\Users\SWIFT~1.TIA\AppData\Local\Temp\ccfurDVF.s:40554: Error: offset out of range
C:\Users\SWIFT~1.TIA\AppData\Local\Temp\ccfurDVF.s:40560: Error: offset out of range
```

This is likely because LTO moves the literal pool, which stores the addresses of `arch_irq_unlock_outlined` and `z_thread_entry`, to somewhere else out of reach by the Thumb `ldr` instruction.

stephanosio, 2025-07-03:
A fix has been implemented in https://github.com/zephyrproject-rtos/zephyr/pull/92569

PR #93126 (2025-08-11): [Backport v3.7-branch] arch: arm: cortex_m: Force literal pool placement in inline asm block
Backport https://github.com/zephyrproject-rtos/zephyr/commit/d7d3ae5c3b06617bce8d6c2ebcc99d4355ce94d9 from #92569.

Fixes #90777
closed 2025-07-07 by danieldegrasse