zephyrproject-rtos/zephyr #90779: Input: Issue with multiple inputs are enabled

AnkyXCoder, 2025-05-29:
### Describe the bug

Problem:

### Regression

- [ ] This is a regression.

### Steps to reproduce

Necessary changes added:
Updated `modules/lib/gui/lvgl/demos/keypad_encoder/lv_demo_keypad_encoder.c` to accept Touch Inputs by adding  `LV_INDEV_TYPE_POINTER` to group.
Selected `LV_Z_DEMO_KEYPAD_AND_ENCODER` in `samples/modules/lvgl/demos/prj.conf`.
Updated board specific DTS file as per requirements: [esp32s3_devkitc_procpu.dts](https://github.com/user-attachments/files/20498209/esp32s3_devkitc_procpu.txt)
Build and Flash sample application.

### Relevant log output

```shell

```

### Impact

Major – Severely degrades functionality; workaround is difficult or unavailable.

### Environment

- OS: Linux
- Zephyr Version: 4.1.0
- Zephyr SDK: 0.17.0

faxe1008, 2025-05-30:
> Observation 1:
> When using multiple input devices a.) a touch controller - LVGL Pointer and b.) an encoder - LVGL Encoder are enabled with LVGL on ESP32-S3-DevKitC-1, device is not booting up or gets stuck on the boot-up.

Can you attach a debugger and check for hard faults?

> Observation 3:
> When Input Touch Controller's Interrupt configuration (for example `CONFIG_INPUT_CST816S_INTERRUPT=y`, `CONFIG_INPUT_FT5336_INTERRUPT=y`) is enabled.
> On input to Touch Panel, an interrupt should be generated. But it is not generating and hence, the work is not getting submitted to process the event.

That is confusing to me. The input drivers usually support interrupt mode or polling. If you disable interrupt mode or not hook up the interrupt pin, of course there will not be any interrupts. From the viewpoint of LVGL it does not matter from which mode the events are emitted. You can add `CONFIG_INPUT_EVENT_DUMP` to add logging of any raised events.

I just tested the demo on the native sim and a `mimxrt1060` with a `gt911` touch driver, and it works as expected.

AnkyXCoder, 2025-05-31:
> Can you attach a debugger and check for hard faults?

I don't have a debugger but as per the logs, the device gets stuck on the boot-up only and there are no log prints or shell showing up.

> That is confusing to me. The input drivers usually support interrupt mode or polling. If you disable interrupt mode or not hook up the interrupt pin, of course there will not be any interrupts. From [...]

As mentioned in the observations and attached overlay file, all the pins for Touch Controller are properly configured and not conflicting with other peripherals. Also, to mention that, when LVGL Encoder node is disabled, then the same code and sample works with the Touch Controller Interrupt configuration enabled.

sylvioalves, 2025-06-20:
@AnkyXCoder I started to take a look at this. However, without a proper HW in place to test it sounds difficult. Anyway, I copied the info you pasted but could not find the "esp32s3_touch_lcd-pinctrl.dtsi" file. Can you provide it or a full sample code ready?

sylvioalves, 2025-06-24:
> Hi [@sylvioalves](https://github.com/sylvioalves) Kindly find the attached sample application together with the custom board files I have been working with.
> 
> [lvgl_demo.zip](https://github.com/user-attachments/files/20867957/lvgl_demo.zip) For the sample application, There isn't any implementation related to handling the Encoder related events. In the board DTS file, `encoder` node is set to `okay`. With this configuration, it is observed that the device doesn't boot up. However, when `encoder` node is set to `disabled`, the sample application works as expected. FYI, this board is just a replica of `esp32s3_touch_lcd_1_28` board with some minor changes.

@AnkyXCoder I run the sample code in a `esp32s3_devkitm` and it boots good, with or without `lvgl_encoder` set to `okay` or `disabled`. Perhaps I am missing some additional change?

sylvioalves, 2025-06-25:
@AnkyXCoder Any suggestion?

AnkyXCoder, 2025-06-26:
@sylvioalves 
Hmm, That's strange.
Am I missing anything on my side?