zephyrproject-rtos/zephyr #90790: FDC2x1x driver contains critical memory access issues

TylerBird, 2025-05-29:
### Describe the bug

I wanted to use the TI sensor fdc2x1x (onboard I have a FDC2214). The driver is also available. After integration, the controller did not start and I received a “fatal exception” error.

[image]

First, I had an ESP32 (board: M5StickC-plus, flashed via UART), then I tried it with an STM32L552 (flashed via STLink V3) controller. 

After ruling out the app code as the source of the error, I took a look at the driver. Here I discovered memory access violations. 
For example, here:
Note the variable access `cfg->ch_cfg->fin_sel` which is channel specific and therefore must be implemented as follows: `cfg->ch_cfg-[ch].fin_sel`.

[image]

I have noticed 3 of these access violations.

Furthermore, I use the chip without the `sd_gpio`.
In the code, however, the function `fdc2x1x_init(const struct device *dev)` incorrectly checks whether this has been set in the device tree.

Here is what helped me:
`if (cfg->sd_gpio.port->name) {` -> `if (cfg->sd_gpio.port != NULL) {`

I would like to open a related pull-request to solve the found issues.

### Regression

- [ ] This is a regression.

### Steps to reproduce

Execute the example from the folder `samples/sensor/fdc2x1x`.

### Relevant log output

```shell

```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

### Environment

- Linux Ubuntu
- Zephyr SDK with VSCode
- zephyrproject-rtos with revision: v4.1.0-rc3

TylerBird, 2025-05-29:
The driver appears to be correct.
The original issue was that I did not define the SD pin in the device tree.
In my case, I set the SD pin in the hardware to “always active” using a pull-down resistor. I incorrectly assumed that the SD pin was optional. Therefore, I did not specify it in the device tree.

NotAnsgar, 2025-05-29:
I checked the sensor readings, and they’re correct when an sd-gpio is provided; the documentation doesn’t list it as mandatory.

When sd-gpio isn’t specified, Zephyr raises a BUS FAULT.

While debugging the initialization I found that sd_gpio is NULL, so dereferencing its name member triggers the exception.

A straightforward fix is to replace
[if (cfg->sd_gpio.port->name) {](https://github.com/zephyrproject-rtos/zephyr/blob/4189364849b5ec6a7eb951637040dd94194a4e33/drivers/sensor/ti/fdc2x1x/fdc2x1x.c#L923)

with

`if (cfg->sd_gpio.port) {`.
 
After this change, the FDC2214 can be read both with and without an sd-gpio in the device tree and  the `fdc2x1x_init_sd_pin(dev)` will only be called if the `sd-gpio` is specified.

I’m not certain whether this will later interfere with power-management routines that use the pin to shut down the FDC, but the adjustment probably best matches the driver author’s intent.