zephyrproject-rtos/zephyr #90850: Enabling OpenThread Border Router in Zephyr

Cristib05, 2025-05-30:
# Proposal for Enabling OpenThread Border Router functionality in Zephyr

## Introduction

Zephyr has integrated OpenThread stack and supports functionality for a 15.4 node. Our proposal is to leverage what OpenThread stack supports and to enable Border Router functionality.

## Proposed changes

### Enablement of routing capability between interfaces from .conf file
* Allow IPv6 traffic to be forwarded between OpenThread and Wi-Fi/Eth interface.
* Add routing table entries and manage them accordingly.

### Notify OpenThread stack about ICMPv6 ND messages (RA, RS, ND)
* Create a raw socket using BSD socket style or register an ICMP callback through which OpenThread stack is notified.

### Implement platform APIs required by OpenThread Border Router application
* Infrastructure interface
* UDP platform
* mDNS socket
* TREL module
* Border Agent module
* DNS upstream resolver

## Concerns and Unresolved Questions

### mDNS implementation

OpenThread includes its own mDNS module, implemented in accordance with RFC 6762, and provides a platform API for sending and receiving mDNS messages. Zephyr also offers its own mDNS responder. Our preferred approach is to utilize OpenThread’s mDNS module—at least for OpenThread-specific functionalities such as advertising/discovery proxy, Border Agent, and TREL. However, we are uncertain about which mDNS module should be used by an application developed in conjunction with the Border Router functionality. We are considering using only OpenThread's mDNS implementation and disabling Zephyr's mDNS responder or an approach using a dual mDNS stack implementation.

rlubos, 2025-06-03:
Hi Cristian,

Adding OTBR capabilites to Zephyr sounds like a really nice and desirable addition. If you're willing to work on this project, contributions are more than welcome, and you can expect our support in terms of reviews and any issues with Zephyr's network stack.

Some initial feedback based on the proposal:

> Create a raw socket using BSD socket style or register an ICMP callback through which OpenThread stack is notified.

I think registering a ICMP callback with `net_icmp_init_ctx()` is the most straighforward solution for caputirng ICMP traffic. Optionally, we've recently added RAW IP sockets support to Zephyr, that could also be an option (although I still think the callback apporach would be easier).

> We are considering using only OpenThread's mDNS implementation and disabling Zephyr's mDNS responder or an approach using a dual mDNS stack implementation.

If we disable native mDNS implementation, we won't be able to use mDNS on interfaces other than OpenThread, no? Perhaps for initial support such a limitation would be acceptable, but I think we'd need to make it possible to use dual-stack approach in the future (or stick to Zephry's mDNS only, if possible).

Cristib05, 2025-06-03:
Thanks for the feedback, Robert! We are willing to implement this project and hopefully we will soon open some PRs and we could further discuss based on their content.
For the moment I suggest sticking with OpenThread mDNS implementation for the sake of having something functional, and then we can decide for an optimal solution.

piotrParf, 2025-07-01:
May I have a question about how it should work? I am trying to run this demo on nrf7002dk with nrf5340 on board. Due to Flash limitations, I am using wifi as a station there. I am running the demo. I am in the state that after the wifi station connection on shell, I run "ot ifconfig up" and "ot thread start" on Zephyr cli and I have both connections: Wifi and OpenThread to my simple OT network. But I can`t reach via "net ping" or Linux "ping" from the thread device to the Linux machine via border router. What other steps should I take? I have even the same prefix pushed to my OT device via border router and BR interface of wifi and ot have the same ipv6 prefix adresses, yet i can`t ping the wifi interface from my ot device connected to the border router via OT network. I can ping inside the OT network, and I can ping the OT border router's Wi-Fi interface from a Linux machine, but not across the border router.

Cristib05, 2025-07-14:
# Testing report and captures based on first initial code version

## Test bed:
* RW612 which acts as a Border Router
* A thread node
* RPI which acts as a node on the backbone interface

### Scenario

#### RW612 is Thread network's leader and is connected to Wi-Fi; it has the following OT IPv6 addresses:
[image]

#### A net route based on the OT slaac address is added:
[image]

####RW612 is reachable from backbone network interface:
[image]

#### A thread node is joining the network 

[image]

[image]

#### and registers a multicast address using `ipmaddr add <ip-addr>` command:

[image]

#### Thread node can be reached from RPI device:

[image]

[image]

#### RW612 device also responds to ICMPv6 Router Discovery:

[image]

[image]