zephyrproject-rtos/zephyr #90864: Build warnings with TF-M that are not treated as errors

nashif, 2025-05-30:
### Describe the bug

when building anything with TF-M, there are build warnings that are not being treated as errors, so there are 2 issues:

- The build warnings themselves
- The fact that -Werror is not being propagted to TF-M and those are not detected in CI and during development

### Regression

- [ ] This is a regression.

### Steps to reproduce

` west   build -p  -b  mps2/an521/cpu0/ns  samples/hello_world`

### Relevant log output

```shell
[48/378] Building C object bl2/ext/mcuboot/bootutil/CMakeFiles/bootutil.dir/src/bootutil_misc.o
/home/nashif/zephyrproject/bootloader/mcuboot/boot/bootutil/src/bootutil_misc.c: In function 'boot_trailer_scramble_offset':
/home/nashif/zephyrproject/bootloader/mcuboot/boot/bootutil/src/bootutil_misc.c:153:15: warning: implicit declaration of function 'flash_area_get_sector'; did you mean 'flash_area_get_sectors'? [-Wimplicit-function-declaration]
  153 |         ret = flash_area_get_sector(fa, flash_area_get_size(fa) - boot_trailer_sz(alignment),
      |               ^~~~~~~~~~~~~~~~~~~~~
      |               flash_area_get_sectors
```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

### Environment

Zephyr SDK 0.17.1 on Linux on latest main branch v4.1.0-4985-ga098f5c3a39

nashif, 2025-06-04:
Any update on this?

ceolin, 2025-06-06:
https://github.com/zephyrproject-rtos/zephyr/pull/91157

This makes TF-M build to treat warnings as errors but it is only available in 2.2.0. Since we are moving to that does not make much sense backporting this commit.

ceolin, 2025-06-06:
https://github.com/zephyrproject-rtos/zephyr/pull/91169 -> Lets start testing

nordicjm, 2025-06-09:
@ceolin @tomi-font there seems to be commits missing from the tfm branch that zephyr used for v3.7, can you backport the commits which add `flash_area_get_sector()` itself and the prototype since MCUboot fixes need it (and actually have used it from years ago but seems to have not been noticed in tfm because it does not use swap using move by default)

ceolin, 2025-06-10:
@nordicjm I didn't see any commit adding this function. Only `lash_area_get_sectors`

[image]

The commit that @tomi-font pointed seems to have it, but it is not part of any release. It is pretty much adding a candidate release to tf-m. I will work on it because it seems the best route and I hope we can point to an official MCUboot release.

nashif, 2025-07-11:
marking as a blocker for 4.2

PR #91157 (2025-07-16): tf-m: build: Treat warnings as errors
Starting in TF-M 2.2.0 there is a build option to treat warnings as errors. Lets use it.

Fixes #90864