zephyrproject-rtos/zephyr #90907: nordic: hardware pulse counting

jodzik, 2025-05-31:
**Is your feature request related to a problem? Please describe.**
I need several simple pulse counter, but signals has a high frequency(up to 1Mhz), therefore interrupt-based implementation may be too demanding on CPU.  
nRF52832(in may case) and similar hardware are capable of hardware pulse counting, implementation of this can solve the problem.

**Describe the solution you'd like**
A sensor driver like `/drivers/sensor/espressif/pcnt_esp32/`
DTS may look like:
```dts
&pulse_counter {
    compatible = "nordic,pcnt"
    status = "okay";
    gpios = <&gpio0 26 GPIO_ACTIVE_LOW>;
    timer = <&timer4>;
    mode = "...";
}
```

OR

```dts

&timer4 {
    status = "okay";
    pulse_counter0: pulse_counter0 {
        compatible = "nordic,pcnt"
        status = "okay";
        mode = "...";
        gpios = <&gpio0 26 GPIO_ACTIVE_LOW>;
    }
}
```

Maybe it's better to use pinctl node to select GPIOTE/pin.

**Describe alternatives you've considered**
Use `nrfx` directly or use external driver([e.g.](https://github.com/eivindse/zephyr_encoder_driver)).

sayooj-aerlync, 2025-06-07:
Would it not be better to add a generic implementation so that all supported platforms can make use of it?

jodzik, 2025-09-14:
> Does it make sense to have it under sensor architecture?

I think not for following reason: 
Sensing subsystem as described in doc as  "is a high level sensor framework inside the OS user space service layer", but hardware pulse counting is based on peripheral hardware (timer or dedicated unit) and can use it completely, i.e. it sounds more like as API to the peripheral and more like a **driver**.

Now there is a **counter** api, and possible solution is to add to the counters ability to select a "clock" source (just as capture api is being added in https://github.com/zephyrproject-rtos/zephyr/pull/89127).
In case of pulse counting, for STM32 it will be ETR, for NRF it will be GPIOE.
This solution is particularly suitable for NRF (because NRF counter can count everything), but not so well situated for all SoC, e.g. espressif - it has a dedicated pulse counter unit. In this case, the `counter` nodes, declared in the `timer` nodes in dts (in order to perform their basic functions) will be declared inside a peripheral block that is not capable of hardware pulse counting.

In view of all that has been said, I think the most correct way - dedicated api and drivers for pulse counter, similar to how it is done for `pwm`. In dts for stm32/nrf and similar, `pulse counter`'s nodes will be declared inside `timer`'s, for cases such as espressif - in special peripheral nodes (for espressif it is `pcnt`).

**IIUC**.

sayoojkkarun, 2025-09-17:
How do we handle Quadrature Decoders here? Should it be a part of Counters or it should be handled separately?

I would like to come up with an RFC for this.

closed 2026-09-16 by github-actions[bot]