zephyrproject-rtos/zephyr #90977: Board STM32N6570-DK Ethernet Tx not working

alister-advtek, 2025-06-02:
### Describe the bug

With an STM32N6570-DK board, samples\net\dhcpv4_client, the STM32N6 receives Ethernet frequently, but only transmits DHCP discoveries, and so it never acquires an IP address. Conclusion: STM32N6 transmit is not working.
With the same board, STM32CubeN6's Nx_WebServer example works ok.

With a NUCLEO-N657X0-Q board, samples\net\dhcpv4_client works ok.

### Regression

- [ ] This is a regression.

### Steps to reproduce

Steps to reproduce:
1. west build -b stm32n6570_dk zephyr\samples\net\dhcpv4_client
2. load and execute

### Relevant log output

```shell
[00:00:00.050,000] <inf> phy_mii: PHY (0) ID 1CC916
*** Booting Zephyr OS build v4.1.0-5122-gad3df1d0f240 ***
[00:00:00.052,000] <inf> net_dhcpv4_client_sample: Run dhcpv4 client
[00:00:00.052,000] <inf> net_dhcpv4_client_sample: Start on ethernet: index=1
[00:00:04.754,000] <inf> phy_mii: PHY (0) Link speed 1000 Mb, full duplex
uart:~$
```

### Impact

Major – Severely degrades functionality; workaround is difficult or unavailable.

### Environment

OS: Windows
Toolchain: zephyr-sdk-0.17.0
https://github.com/zephyrproject-rtos/zephyr/commit/ad3df1d0f240f29e07b1620992e2b1aa91e5cf15

james-advatek, 2025-06-10:
@marwaiehm-st is there an open PR with these modifications?

alister-advtek, 2025-06-10:
Hi @marwaiehm-st,

Is this correct? You've fixed the problem on the STM32N6570-DK, the NUCLEO-N657X0-Q will require the fix too, and you'll move the the fix's DTSI change to the dts/arm/st/n6 directory for all STM32N6 developments?

Could you merge Zephyr main to your fork's branch please?

The soc/st/stm32/stm32n6x/sections.ld doesn't appear robust to any changes of ETH_DMA_RX_CH_CNT, ETH_RXBUFNB, ETH_DMA_TX_CH_CNT and ETH_TXBUFNB.

Thanks

mathieuchopstm, 2025-06-16:
@alister-advatek please re-open issue if problems are still present after #91445 merge

PR #91445 (2025-06-13): boards: stm32: stm32n6570_dk: fix DTS errors for Ethernet
The DTS for this board contained a few mistakes:
* The PHY address was incorrect
	* This did not impair functionality because the RTL8211F treats address 0 as "broadcast" and accepts commands sent to it by default
* An incorrect PHY driver was used
	* On-board PHY (`RTL8211F`) has an existing Zephyr driver - use that instead of the generic `phy_mii`
	* Rework Ethernet MAC pinctrl configuration to allow driver to receive PHY interrupts
* The RGMII GTX CLK pinctrl must be configured at `medium-speed`
	* **Fixes #90977 - unreliable Tx issu [... cut]