zephyrproject-rtos/zephyr #91019: tests: tests/drivers/spi/spi_loopback/drivers: spi: mcux_dspi_dma test Timeout

hakehuang, 2025-06-03:
**Describe the bug**
mcux_dspi_dma test is Timeout on v4.1.0-5009-g6d8b78d57d30 on frdm_k64f/mk64f12
testcase path is tests/drivers/spi/spi_loopback

see logs for details

**To Reproduce**
1. 
```
scripts/twister --device-testing --device-serial /dev/ttyACM0 -p frdm_k64f/mk64f12  -T tests/drivers/spi/spi_loopback --sub-test frdm_k64f/mk64f12:tests/drivers/spi/spi_loopback/drivers.spi.mcux_dspi_dma.loopback
```
or
```
# cd tests/tests/drivers/spi/spi_loopback
# west build -b frdm_k64f/mk64f12
# west flash
```
2. See error

**Expected behavior**
test pass

**Impact**

**Logs and console output**

```
*** Booting Zephyr OS build v4.1.0-5009-g6d8b78d57d30 ***
SPI test on buffers TX/RX 0x20001220/0x20001200, frame size = 8, DMA enabled (without CONFIG_NOCACHE_MEMORY)
Polling...Running TESTSUITE spi_extra_api_features
===================================================================
START - test_spi_hold_on_cs
 PASS - test_spi_hold_on_cs in 0.004 seconds
===================================================================
START - test_spi_lock_release
 PASS - test_spi_lock_release in 0.002 seconds
===================================================================
TESTSUITE spi_extra_api_features succeeded
Running TESTSUITE spi_extra_api_features
===================================================================
START - test_spi_hold_on_cs
 PASS - test_spi_hold_on_cs in 0.004 seconds
===================================================================
START - test_spi_lock_release
 PASS - test_spi_lock_release in 0.002 seconds
===================================================================
TESTSUITE spi_extra_api_features succeeded
Running TESTSUITE spi_loopback
===================================================================
Testing loopback spec: SLOW
START - test_nop_nil_bufs
 PASS - test_nop_nil_bufs in 0.001 seconds
===================================================================
START - test_spi_async_call
Buffer contents are different:
 0x30,0x31,0x32,0x33,0x34,0x35,0x36,0x37,0x38,0x39,0x61,0x62,0x63,0x64,0x65,0x66,0x2d,0x00,
vs:
 0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
 FAIL - test_spi_async_call in 0.045 seconds
===================================================================
START - test_spi_complete_large_transfers

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/spi/spi_loopback/src/spi.c:530: spi_loopback_test_spi_complete_large_transfers: (memcmp(large_buffer_tx, large_buffer_rx, BUF3_SIZE) is true)
Large Buffer contents are different
 FAIL - test_spi_complete_large_transfers in 0.045 seconds
===================================================================
START - test_spi_complete_loop_mode_0
Buffer contents are different:
 0x30,0x31,0x32,0x33,0x34,0x35,0x36,0x37,0x38,0x39,0x61,0x62,0x63,0x64,0x65,0x66,0x2d,0x00,
vs:
 0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
 FAIL - test_spi_complete_loop_mode_0 in 0.021 seconds
===================================================================
START - test_spi_complete_loop_mode_1
Buffer contents are different:
 0x30,0x31,0x32,0x33,0x34,0x35,0x36,0x37,0x38,0x39,0x61,0x62,0x63,0x64,0x65,0x66,0x2d,0x00,
vs:
 0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
 FAIL - test_spi_complete_loop_mode_1 in 0.021 seconds
===================================================================
START - test_spi_complete_loop_mode_2
Buffer contents are different:
 0x30,0x31,0x32,0x33,0x34,0x35,0x36,0x37,0x38,0x39,0x61,0x62,0x63,0x64,0x65,0x66,0x2d,0x00,
vs:
 0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
 FAIL - test_spi_complete_loop_mode_2 in 0.021 seconds
===================================================================
START - test_spi_complete_loop_mode_3
Buffer contents are different:
 0x30,0x31,0x32,0x33,0x34,0x35,0x36,0x37,0x38,0x39,0x61,0x62,0x63,0x64,0x65,0x66,0x2d,0x00,
vs:
 0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
 FAIL - test_spi_complete_loop_mode_3 in 0.021 seconds
===================================================================
START - test_spi_complete_multiple
Buffer contents are different:
 0x30,0x31,0x32,0x33,0x34,0x35,0x36,0x37,0x38,0x39,0x61,0x62,0x63,0x64,0x65,0x66,0x2d,0x00,
vs:
 0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
 FAIL - test_spi_complete_multiple in 0.022 seconds
===================================================================
START - test_spi_complete_multiple_timed
Transfer took 1473 us vs theoretical minimum 864 us
Latency measurement: 609 us
Buffer contents are different:
 0x30,0x31,0x32,0x33,0x34,0x35,0x36,0x37,0x38,0x39,0x61,0x62,0x63,0x64,0x65,0x66,0x2d,0x00,
vs:
 0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
 FAIL - test_spi_complete_multiple_timed in 0.028 seconds
===================================================================
START - test_spi_concurrent_transfer_different_spec

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/spi/spi_loopback/src/spi.c:727: spi_transfer_thread: (tx_buf_ptr not equal to rx_buf_ptr)
SPI concurrent transfer data mismatch
 FAIL - test_spi_concurrent_transfer_different_spec in 0.018 seconds
===================================================================
START - test_spi_concurrent_transfer_same_spec
E: Timeout waiting for transfer complete

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/spi/spi_loopback/src/spi.c:724: spi_transfer_thread: (spi_transceive_dt(spec, &tx_bufs, &rx_bufs) not equal to 0)
SPI concurrent transfer failed
ASSERTION FAIL [!terminate] @ WEST_TOPDIR/zephyr/kernel/sched.c:442
	aborted _current back from dead
E: r0/a1:  0x00000004  r1/a2:  0x000001ba  r2/a3:  0x00000080
E: r3/a4:  0x00000004 r12/ip:  0x00002ee0 r14/lr:  0x000063ef
E:  xpsr:  0x21000000
E: Faulting instruction address (r15/pc): 0x00008088
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x20001820 (unknown)
E: Halting system

    
```
**Environment (please complete the following information):**
 - OS: (e.g. Linux )
 - Toolchain (e.g Zephyr SDK)
 - Commit SHA or Version used: v4.1.0-5009-g6d8b78d57d30

decsny, 2025-06-03:
is this a regression?

hakehuang, 2025-06-04:
> is this a regression? when did this break? there has not been any changes to the dspi driver in almost 8 months

yes, it is a regression. I forget to add git bisect, just updated.

closed 2025-07-07 by decsny