zephyrproject-rtos/zephyr #91035: esp32: Fatal exception in samples/net/wifi/apsta_mode when using the main thread

SamsonBox, 2025-06-04:
### Describe the bug

I want to build an application for the esp32 based on the samples/net/wifi/apsta_mode example. But as soon as I use the main thread to to some work, the esp throws an fatal exception. This happens when a devices tries to join the access point and the dhcp servers does his work.

### Regression

- [ ] This is a regression.

### Steps to reproduce

1. Add 
`
k_thread_priority_set (	k_current_get (), K_LOWEST_APPLICATION_THREAD_PRIO);
for(;;)
{
	k_yield();
}` 
into the main funtion after `connect_to_wifi();`
2. Build and run the application
3. Connect a device to the esp32 access point

### Relevant log output

```shell
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3ffb0000,len:14972
ho 0 tail 12 room 4
load:0x40080000,len:64064
1150 mmu set 00010000, pos 00010000
entry 0x40089354
I (48) soc_init: ESP Simple boot
I (48) soc_init: compile time Jun  4 2025 06:13:10
W (48) soc_init: Unicore bootloader
I (48) soc_init: chip revision: v3.1
I (52) flash_init: SPI Speed      : 40MHz
I (55) flash_init: SPI Mode       : DIO
I (59) flash_init: SPI Flash Size : 4MB
I (62) boot: DRAM: lma 0x00001020 vma 0x3ffb0000 len 0x3a7c   (14972)
I (69) boot: IRAM: lma 0x00004aa4 vma 0x40080000 len 0xfa40   (64064)
I (75) boot: IRAM: lma 0x000144f8 vma 0x00000000 len 0xbb00   (47872)
I (81) boot: IMAP: lma 0x00020000 vma 0x400d0000 len 0x5ae7c  (372348)
I (87) boot: IRAM: lma 0x0007ae84 vma 0x00000000 len 0x5174   (20852)
I (93) boot: DMAP: lma 0x00080000 vma 0x3f400000 len 0xd100   (53504)
I (99) boot: Image with 6 segments
I (103) boot: IROM segment: paddr=00020000h, vaddr=400d0000h, size=5AE7Ch (372348) map
I (110) boot: DROM segment: paddr=00080000h, vaddr=3f400000h, size=0D100h ( 53504) map
I (129) boot: libc heap size 69 kB.
I (129) spi_flash: detected chip: generic
I (129) spi_flash: flash io: dio
[00:00:00.166,000] <inf> wifi_init: rx ba win: 6
*** Booting Zephyr OS build v4.1.0-5166-g2c851d998caa ***
[00:00:05.168,000] <inf> MAIN: Turning on AP Mode
[00:00:05.169,000] <dbg> net_dhcpv4_server: net_dhcpv4_server_start: Started DHCPv4 server, address pool:
[00:00:05.170,000] <dbg> net_dhcpv4_server: net_dhcpv4_server_start:     0: 192.168.4.11
[00:00:05.170,000] <dbg> net_dhcpv4_server: net_dhcpv4_server_start:     1: 192.168.4.12
[00:00:05.170,000] <dbg> net_dhcpv4_server: net_dhcpv4_server_start:     2: 192.168.4.13
[00:00:05.170,000] <dbg> net_dhcpv4_server: net_dhcpv4_server_start:     3: 192.168.4.14
[00:00:05.170,000] <inf> MAIN: DHCPv4 server started...

[00:00:05.228,000] <wrn> net_if: iface 1 pkt 0x3ffcd438 send failure status -5
[00:00:05.228,000] <wrn> net_if: iface 1 pkt 0x3ffcd438 send failure status -5
[00:00:05.229,000] <wrn> net_if: iface 1 pkt 0x3ffcd438 send failure status -5
[00:00:05.229,000] <wrn> net_if: iface 1 pkt 0x3ffcd438 send failure status -5
[00:00:05.230,000] <inf> MAIN: AP Mode is enabled. Waiting for station to connect
[00:00:05.230,000] <inf> MAIN: Connecting to SSID: SSID

[00:00:05.232,000] <wrn> net_if: iface 2 pkt 0x3ffcd438 send failure status -5
[00:00:05.233,000] <wrn> net_if: iface 2 pkt 0x3ffcd438 send failure status -5
[00:00:05.233,000] <wrn> net_if: iface 2 pkt 0x3ffcd438 send failure status -5
[00:00:05.233,000] <wrn> net_if: iface 2 pkt 0x3ffcd438 send failure status -5
[00:00:06.230,000] <wrn> net_if: iface 1 pkt 0x3ffcd438 send failure status -5
[00:00:06.234,000] <wrn> net_if: iface 2 pkt 0x3ffcd438 send failure status -5
[00:00:07.231,000] <wrn> net_if: iface 1 pkt 0x3ffcd438 send failure status -5
[00:00:07.234,000] <wrn> net_if: iface 2 pkt 0x3ffcd438 send failure status -5
[00:00:08.074,000] <inf> MAIN: Disconnected from SSID
[00:00:12.119,000] <inf> MAIN: station: 11:22:33:44:55:66 joined
[00:00:12.302,000] <dbg> net_dhcpv4_server: dhcpv4_server_cb: enter dhcpv4_server_cb
[00:00:12.340,000] <dbg> net_dhcpv4_server: dhcpv4_server_cb: enter dhcpv4_server_cb
[00:00:12.343,000] <dbg> net_dhcpv4_server: dhcpv4_handle_discover: DHCPv4 processing Discover - reserved 192.168.4.11
[00:00:13.345,000] <err> os:  ** FATAL EXCEPTION
[00:00:13.345,000] <err> os:  ** CPU 0 EXCCAUSE 28 (load prohibited)
[00:00:13.345,000] <err> os:  **  PC 0x40084e6c VADDR 0xfffffff0
[00:00:13.345,000] <err> os:  **  PS 0x60e20
[00:00:13.345,000] <err> os:  **    (INTLEVEL:0 EXCM: 0 UM:1 RING:0 WOE:1 OWB:14 CALLINC:2)
[00:00:13.345,000] <err> os:  **  A0 0x40084e98  SP 0x3ffc9ef4  A2 0xfffffff0  A3 0
[00:00:13.345,000] <err> os:  **  A4 0x3ffb08cc  A5 0xffffffff  A6 0x3ffb4278  A7 0x60a20
[00:00:13.345,000] <err> os:  **  A8 0x800873d5  A9 0x3ffb4fc0 A10 0x20a19 A11 0
[00:00:13.345,000] <err> os:  ** A12 0x60a23 A13 0 A14 0x3ffc08c0 A15 0x19
[00:00:13.346,000] <err> os:  ** LBEG 0x4000c2e0 LEND 0x4000c2f6 LCOUNT 0xffffffff
[00:00:13.346,000] <err> os:  ** SAR 0x2
[00:00:13.346,000] <err> os:  **  THREADPTR 0x3ffc1fba

Backtrace:0x40084e69:0x3ffc9ef4  |<-CORRUPTED

[00:00:13.346,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:13.346,000] <err> os: Current thread: 0x3ffc07c0 (unknown)
[00:00:13.438,000] <err> os: Halting system
```

### Impact

Functional Limitation – Some features not working as expected, but system usable.

### Environment

- OS Windows
- Zephyr SDK
- commit hash 2c851d998caa613f42eadab6a38c3bd1835d2cb3

sylvioalves, 2025-06-05:
@SamsonBox, increase `SYSTEM_WORKQUEUE_STACK_SIZE` to at least 2048 (or better 4096) and you will be fine.
Perhaps we can update the sample to use this as default value. Let me know how it goes.

SamsonBox, 2025-06-06:
@sylvioalves, yes it works now! Thank you for the quick support! 

So from the PC I found, that the exception happen anywher in the extensa_restore_high_regs function which is call (I guessed) during the task switches. So I checkt the thread status using the shell kernel commands, but I didn's see any critical values. How could I have seen the issiue? Do you have a tipp?

Yes I think updateing the sample would be good. Or you put some word's to this topic in the README.

sylvioalves, 2025-06-09:
I enabled CONFIG_SHELL and run `kernel thread stack` to analyze it only. sysworkq was 100% I think. I'll close this. Re-open if needed.