zephyrproject-rtos/zephyr #91069: SPDX: Missing license/copyright info in generated SPDX files

MustafaPrimes, 2025-06-04:
### Describe the bug

When generating SPDX output with `west spdx`, several files—especially from Zephyr or external modules—show `NOASSERTION` or empty fields, even though these files include valid license and copyright comments.

### Regression

- [ ] This is a regression.

### Steps to reproduce

1. Generate SPDX using `west spdx`
2. Inspect the SPDX output (e.g. `modules-deps.spdx`, `zephyr.spdx`)
3. Notice missing copyright/license data

### Relevant log output

```shell

```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

### Environment

- Linux
- Zephyr
- west

### Additional Context

Expected Result:
Fields should be filled based on the license headers in the source code.

kartben, 2025-06-19:
> ** I have a question @kartben – do you have a rough estimate for when the copyright extraction support might be available?

@MustafaPrimes hopefully really soon - see https://github.com/zephyrproject-rtos/zephyr/pull/91855 :)

closed 2025-06-22 by kartben

pdgendt, 2025-06-23:
@kartben isn't #91855 supposed to close this?