zephyrproject-rtos/zephyr #91085: SAME70_Explained undefined reference to device_dts when using usbd_next_prj.conf in cdc_acm sample

JayToltTech, 2025-06-04:
### Describe the bug

Zephyr v4.1.0

```
west build -p -b sam_e70_xplained/same70q21b samples/subsys/usb/cdc_acm/ -- -DEXTRA_CONF_FILE="usbd_next_prj.conf"
```

```
arm-zephyr-eabi/bin/ld.bfd: zephyr/subsys/usb/device_next/libsubsys__usb__device_next.a(usbd_cdc_acm.c.obj):/Users/jay/Code/zephyrproject/zephyr/subsys/usb/device_next/class/usbd_cdc_acm.c:1327: multiple definition of `__device_dts_ord_116'; zephyr/libzephyr.a(cdc_acm.c.obj):/Users/jay/Code/zephyrproject/zephyr/subsys/usb/device/class/cdc_acm.c:1229: first defined here
arm-zephyr-eabi/bin/ld.bfd: app/libapp.a(sample_usbd_init.c.obj):(._usbd_context.static.sample_usbd_+0x18): undefined reference to `__device_dts_ord_115'
```

### Regression

- [ ] This is a regression.

### Relevant log output

```shell

```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

jfischer-no, 2025-06-04:
That is not a bug, it is because the driver has not been ported. https://github.com/zephyrproject-rtos/zephyr/issues/74663

closed 2025-06-04 by jfischer-no

JayToltTech, 2025-06-05:
Ooof.  Three year old issue.  Thanks for all you do, @jfischer-no.