zephyrproject-rtos/zephyr #91095: device_init doesn't pass on negative error codes from drivers

madone8, 2025-06-05:
### Describe the bug

Context: I have a custom device driver which is configured with the `zephyr,deferred-init` property in the devicetree.

I noticed when checking the return code of `device_init()` that although the driver's initialisation was failing and should be returning `-EIO`, this value was not being passed on to my application.

Looking through the codebase, `device_init()` ultimately calls `do_device_init()`, which converts negative error codes returned by the driver into positive codes.

Is this behaviour correct? It does seem contrary to the documentation.

This function also sets the initialised state to `true` irrespective of the return code from the driver.

```
static int do_device_init(const struct device *dev)
{
	int rc = 0;

	if (dev->ops.init != NULL) {
		rc = dev->ops.init(dev);
		/* Mark device initialized. If initialization
		 * failed, record the error condition.
		 */
		if (rc != 0) {
			if (rc < 0) {
				rc = -rc;
			}
			if (rc > UINT8_MAX) {
				rc = UINT8_MAX;
			}
			dev->state->init_res = rc;
		}
	}

	dev->state->initialized = true;

	if (rc == 0) {
		/* Run automatic device runtime enablement */
		(void)pm_device_runtime_auto_enable(dev);
	}

	return rc;
}
```

### Regression

- [ ] This is a regression.

### Relevant log output

```shell

```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

ljd42, 2025-06-05:
Hi @madone8 ,

> Is this behaviour correct? It does seem contrary to the documentation.
To which documentation part are you referring to?

The behavior is documented for the `struct device_state`  (in `device.h`):
```
struct device_state {
          /**
           * Device initialization return code (positive errno value).
           *
           * Device initialization functions return a negative errno code if they
           * fail. In Zephyr, errno values do not exceed 255, so we can store the
           * positive result value in a uint8_t type.
           */
          uint8_t init_res;
  
          /** Indicates the device initialization function has been
           * invoked.
           */
          bool initialized : 1;
  };
```
So `initalized` means that the initialization has occurred, not that the device is "ready" to be used. The `init_res` value is the corresponding "errno" value, ans is therefore positive  (0=success, >0 corresponding errno error)

We might need to improve the documentation.

ljd42, 2025-06-05:
got it, thanks! You're right, the return code for `device_init` isn't aligned with spec.

rruuaanng, 2025-06-06:
@ljd42 @ttmut  Would you choose to repair it? :)

ljd42, 2025-06-06:
@rruuaanng:  If there's no volunteer, I can take care of this issue. Equally happy if @ttmut or anyone else wants to fix it.

ljd42, 2025-06-06:
> [@ljd42](https://github.com/ljd42) [@ttmut](https://github.com/ttmut) Would you choose to repair it? :)

From the code intent, I think that `device_init` should be implement as documented. Especially if we look at:

https://github.com/zephyrproject-rtos/zephyr/blob/d9cbd4ae1d293efbbc22db210f3185e98d72af96/kernel/init.c#L384C1-L391C2

It was an oversight, I guess. Unfortunately, only the "good case" got tested in  tests/kernel/device (when 0 is returned). 

Of course, the change will potentially impact code using deferred initialization. I see 3 groups:
- group 1: does not care to check the return value of `device_init()` => no impact. But they should really start checking 😉 
- group 2: implement the error checking following the spec. Error condition is never detected => positive impact as error condition are now detected that weren't previously. 
- group 3: has implemented a workaround for the faulty error condition => they need to remove their workaround.  We should put a note in the migration guide.

I personally think we should fix it. Interested to hear the feedbacks from "Senior Zephyr dev".

rruuaanng, 2025-06-06:
@ljd42  This is actually a logic error in the code. The documentation is correct, and the description of structure `device_state` is also accurate. It seems that when handling `init_res`, the code forgot to account for the case where `rc` is negative.

I think the code should be like this:
```
static int do_device_init(const struct device *dev)
{
	int rc = 0;
	dev->state->initialized = true;

	if (dev->ops.init != NULL) {
		rc = dev->ops.init(dev);
		/* Mark device initialized. If initialization
		 * failed, record the error condition.
		 */
		if (rc == 0) {
			/* Run automatic device runtime enablement */
			(void)pm_device_runtime_auto_enable(dev);
			dev->state->init_res = rc;
		} else {
			rc = (rc > UINT8_MAX) ? UINT8_MAX : rc;
			dev->state->init_res = (rc < 0) ? -rc : rc;
		}
	}

	return rc;
}
```

ljd42, 2025-06-06:
@rruuaanng : yes, that's exactly what I meant. 👍 

I've a minor comment for the "else part". It works only if we can guarantee that the negative values are >= -255 (possibly the case by design?), otherwise we'll get an overflow.

ttmut, 2025-06-06:
> [@rruuaanng](https://github.com/rruuaanng): If there's no volunteer, I can take care of this issue. Equally happy if [@ttmut](https://github.com/ttmut) or anyone else wants to fix it.

Thanks. You are welcome to submit a fix.

ljd42, 2025-08-10:
@rruuaanng  please remove the stale label. PR is still waiting to be approved & merged.

PR #91221 (2025-09-04): kernel: fix error propagation for device deferred initialization
This fix propagates the failure code returned from the device initialization function to the application when deferred initialization is used. Previously, `device_init()` mistakenly returned `errno` instead of `-errno`.

A test has been added to test.kernel.device to verify the changes  (fail before the fix, pass after).

Fix: #91095