zephyrproject-rtos/zephyr #91118: Zephyr 4.1 Socket zsock_connect() Fails Immediately with errno 116 (ETIMEDOUT)

maxbernard3, 2025-06-05:
### Describe the bug

When attempting to establish a socket connection using `zsock_connect()`, the call fails immediately with `errno 116` (ETIMEDOUT), despite the timeout configurations being set:

```conf
CONFIG_NET_CONFIG_INIT_TIMEOUT=60
CONFIG_NET_SOCKETS_CONNECT_TIMEOUT=10000
```

This is unexpected behavior—`zsock_connect()` should either succeed or return an appropriate error, only timeout **after** a the 10s set by `CONFIG_NET_SOCKETS_CONNECT_TIMEOUT`, not immediately.

**Expected Behavior:**
`zsock_connect()` should block (or at least honor the configured timeouts) and not return ETIMEDOUT immediately unless the timeout is truly reached.

### Regression

- [ ] This is a regression.

### Steps to reproduce

A small reproducible example is available in a [GitLab project here](https://gitlab.com/MaxBERNARD1/zephyr_socket_reduced_case.git) to simplify reproducing the issue. The failure occurs at line 171 in `main.c`.

### Relevant log output

```shell
[00:00:00.001,000] <dbg> eth_stm32_hal: eth_initialize: MAC 00:80:e1:72:71:57
[00:00:00.051,000] <inf> phy_mii: PHY (0) ID 7C131
*** Booting Zephyr OS build v4.1.0 ***
[00:00:00.052,000] <inf> net_config: Initializing network
[00:00:00.052,000] <inf> net_config: Waiting interface 1 (0x24001ae8) to be up...
[00:00:03.753,000] <inf> phy_mii: PHY (0) Link speed 100 Mb, full duplex
[00:00:04.052,000] <inf> net_config: Interface 1 (0x24001ae8) coming up
[00:00:04.053,000] <inf> net_config: IPv4 address: 192.0.2.10
dbg errno = 116
```

### Impact

Functional Limitation – Some features not working as expected, but system usable.

### Environment

**Tested on:**

* Zephyr 4.1.0 with SDK 0.17.0
* Boards:

  * STM32H573I-DK
  * NUCLEO-H723ZG

**Note:**
The issue is **not reproducible** on **Zephyr 3.7.0** with the same SDK version (0.17.0). The version-specific Kconfig settings are included in the minimal example.

maxbernard3, 2025-06-06:
The issue also appears in native_sim

jukkar, 2025-06-09:
Do you have the peer setup i.e., is there a server waiting at the other end of the connection?
When trying your example app and not setting up the server side, I am seeing errno 111 (ECONNREFUSED) returned which is expected in this case.
I am using the current main 4b52921df59bc51b13f28721d8872aa097befc24 and native_sim in my test

jukkar, 2025-06-09:
Never mind my question, after looking more carefully what your sample app is doing, it setups a server socket and tries to connect to it.

jukkar, 2025-06-09:
Did not find the culprit yet, but after running your sample, I am seeing this in net-shell
```
uart:~$ net conn
     Context    Iface  Flags            Local             Remote
[ 1] 0x8087ea0  2      4ST    127.0.0.1:41649          0.0.0.0:0
[ 2] 0x8087f64  1      4ST      0.0.0.0:57742    127.0.0.1:41649
```
The localhost connection is from interface 1 and not from interface 2 which is the localhost. This will of course fail.

PR #91294 (2025-06-11): Fix immediate TCP connect failure with loopback and ACD enabled
The localhost address used in the connection was marked as `tentative` if ACD was enabled. This meant that we never tried to use it properly, which then led to situation that 127.0.0.1 address was used with non-loopback network interface, which caused connection failure when connecting to a localhost address.

Fixes #91118
closed 2025-06-10 by maxbernard3