zephyrproject-rtos/zephyr #91134: I3C MCUX Driver Transfer does busy waiting

PetervdPerk-NXP, 2025-06-05:
### Describe the bug

When using the MCUX I3C controller with both I3C or I2C the transactions are consuming a lot of CPU time.

It seems to narrow down this line.
https://github.com/zephyrproject-rtos/zephyr/blob/d8b37753061a9e0ace95204eccddee1f4b4c477b/drivers/i3c/i3c_mcux.c#L964

In the function it says the following
https://github.com/zephyrproject-rtos/zephyr/blob/d8b37753061a9e0ace95204eccddee1f4b4c477b/drivers/i3c/i3c_mcux.c#L141-L145

But this is not the case we're using microseconds instead.

Suggestion is to use semaphore and then with `TXTRIG` register level to generate interrupts to signal FIFO can be filled.

### Regression

- [ ] This is a regression.

### Relevant log output

```shell

```

### Impact

Functional Limitation – Some features not working as expected, but system usable.

### Environment

- FRDM-MCXN947

mmahadevan108, 2025-06-09:
@PetervdPerk-NXP , would you be able to submit a PR with a fix.

PetervdPerk-NXP, 2025-06-09:
> [@PetervdPerk-NXP](https://github.com/PetervdPerk-NXP) , would you be able to submit a PR with a fix.

I'm currently using a pre-release board with I3C, but operating it in I2C mode only. Given this setup, there's a significant risk of regression. At the moment, using the I3C peripheral in I2C mode results in high CPU usage, which I consider quite severe. Is there an active maintainer for the NXP MCUX I3C driver who could take a look?

mmahadevan108, 2025-06-30:
@PetervdPerk-NXP, kindly provide steps to help reproduce this issue and test the fix.