zephyrproject-rtos/zephyr #91155: Please add Zhiqiang-Hou to the collaborators list

dleach02, 2025-06-06:
### Full Name

Zhiqiang Hou

### GitHub Username

Zhiqiang-Hou

### Organization

NXP

### Supporting Documents

https://github.com/zephyrproject-rtos/zephyr/pulls?q=sort%3Aupdated-desc+is%3Apr+author%3AZhiqiang-Hou+is%3Amerged

stephanosio, 2025-06-06:
@Zhiqiang-Hou has been added as a Collaborator.

JiafeiPan, 2025-06-09:
Hi, @stephanosio , it seems Zhiqiang-Hou is still have not be collaborator, would you please help to have a check? thanks.

For example, I couldn't add him to be reviewer in the following PR:

https://github.com/zephyrproject-rtos/zephyr/pull/91093

And it also display only has read-only permission in this PR's approve list

[image]

stephanosio, 2025-06-09:
> Hi, [@stephanosio](https://github.com/stephanosio) , it seems Zhiqiang-Hou is still have not be collaborator, would you please help to have a check? thanks.

@JiafeiPan An invitation has been sent to @Zhiqiang-Hou. He will have the Collaborator level permissions only after he accepts the invitation.

closed 2025-06-09 by JiafeiPan