zephyrproject-rtos/zephyr #91182: SPDX generation does not seem to work according to the documentation

tijsmoree, 2025-06-06:
### Describe the bug

The documentation states that for SPDX generation, we have to do the following (in case of the `hello_world` project):
- `west spdx -i -d BUILD_DIR/hello_world`
- `west build -d BUILD_DIR/hello_world`
- `west spdx -d BUILD_DIR/hello_world`

With a pristine project (without the BUILD_DIR folder present), this does not work, because the third command errors without `BUILD_DIR/CMakeCache.txt`. I solved this issue by changing the second command to `west build -d BUILD_DIR --domain hello_world` instead.

Another problem I encountered, is that running `west build -p` will undo the `west spdx -i`. This means that running a pristine build negates the necessary precommand to enable SPDX file generation. I ran into this problem when I used the VSCode nRF Connect `Add Build Configuration` button and wanted to generate SPDX files afterwards by running `west spdx -i` and then rebuilding it. However, it does not rebuild, because it knows nothing changed, forcing me to build "pristinely", which also does not result in a build folder that can be used to generate SPDX files, because the `west spdx -i` was undone.

### Regression

- [ ] This is a regression.

### Steps to reproduce

Open the `hello_world` example in nRF Connect SDK v2.8.0 (and remove the build folder if it exists) and run:
- `west spdx -i -d build/hello_world`
- `west build -d build/hello_world --board nrf52840dk/nrf52840`
- `west spdx -d build/hello_world`

### Relevant log output

```shell
parsing CMake Cache file
ERROR: CONFIG_BUILD_OUTPUT_META must be enabled to generate spdx files; bailing
ERROR: SPDX walker failed; bailing
```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

### Environment

- Linux
- Zephyr SDK v3.7.99-ncs1
- nRF Connect SDK v2.8.0
- nRF Connect SDK Toolchain v2.8.0

kartben, 2025-06-11:
@tijsmoree you missed step 2 :)

https://docs.zephyrproject.org/latest/develop/west/zephyr-cmds.html#software-bill-of-materials-west-spdx

tijsmoree, 2025-06-11:
@kartben Oh, I actually did not forget adding the `CONFIG_BUILD_OUTPUT_META` step, that was not the problem here. I did forget to add the step in my reproduction description. Edited!

kartben, 2025-06-11:
@tijsmoree As mentioned, it seems you're errors are likely due to NCS. Following Zephyr instructions to the letter for hello world would work just fine.

```
west spdx --init -d BUILD_DIR                                  
west build -d BUILD_DIR samples/hello_world -b nrf52840dk/nrf52840 -- -DCONFIG_BUILD_OUTPUT_META=y
west spdx -d BUILD_DIR
```

kartben, 2025-06-11:
> The other "problem" I mentioned does still persist, I guess. Running west build --pristine ... will undo the west spdx -i ... command. I don't know whether this is desired?

That's the whole point of pristine :) If you nuke the build folder (pristine), then it kind of make sense you'd need to init the spdx stuff again, right?