zephyrproject-rtos/zephyr #91188: Merge binding properties from all listed compat string if they are on the same bus.

swift-tk, 2025-06-06:
### Problem Description

Zephyr only process the first compat string to get the binding properties for a given node.
Given a DTS like below where two compat string is involved but their properties does not entirely overlap.
The common `mspi-aps-z8` would have the device variant/part number property that would never make into `ambiq,mspi-device` that is platform specific. And `ambiq,mspi-device` would have `ambiq,timing-config` and `ambiq,timing-config-mask` properties that would never make into the common `mspi-aps-z8`; while both three properties are needed in the device tree.

```
	aps51216ba: aps_z8@0 {
		compatible = "ambiq,mspi-device", "mspi-aps-z8";
		size = <DT_SIZE_M(512)>;
		reg = <0>;
		status = "disabled";
		variant = "APS51216BA"
		mspi-max-frequency = <125000000>;
		mspi-io-mode = "MSPI_IO_MODE_HEX_8_8_16";
		mspi-data-rate = "MSPI_DATA_RATE_S_D_D";
		mspi-hardware-ce-num = <0>;
		mspi-dqs-enable;
		read-command = <0x20>;
		write-command = <0xA0>;
		command-length = "INSTR_1_BYTE";
		address-length = "ADDR_4_BYTE";
		rx-dummy = <7>;
		tx-dummy = <8>;
		xip-config = <1 0 DT_SIZE_M(64) 0>;
		ce-break-config = <1024 4>;
		ambiq,timing-config-mask = <0x62>;
		ambiq,timing-config = <8 7 1 0 0 3 10>;
		zephyr,pm-device-runtime-auto;
	};
```

### Proposed Change (Summary)

Update edtlib.py to support pulling bind properties from all listed compat string given that they are on the same bus.

### Proposed Change (Detailed)

No yet

decsny, 2025-06-10:
-1 ❌ 

DT compatible is supposed to describe what device programming model a device has, from most specific to least specific.  Utility bindings are antithesis to the DT paradigm of matching compatible drivers to devices.  This also would make DT even more unclear and confusing to people than it already is.  Not to mention the extra maintenance effort for DT in order to handle all the new edge cases and complication caused by these abilities to "merge" bindings and getting fingers pointed about who is supposed to be handling conflicts in the definitions of the bindings between one another.  Utility bindings also seem like they are just a way to encourage more SW configuration in DT, when, for example with what your case is, there's no reason why you can't code that behavior into your source code at the board level, so I don't see why this is even needed.

swift-tk, 2025-06-11:
> -1 ❌
>
> DT compatible is supposed to describe what device programming model a device has, from most specific to least specific. Utility bindings are antithesis to the DT paradigm of matching [...]

First of all, it is not a utility compatiable, the timing setting is a hardware setting and is required for MSPI to run at high speed. I think this is also true for other platforms and the timing parameters are all somewhat different.
I'm not saying we should merge all the compat without constraints. For complications you mentioned, please elaborate.
IOM, if we set the constriants carefully, there would not be any problem at all.

57300, 2025-06-11:
To be fair, there is a precedent for allowing devicetree properties to be mixed in from multiple bindings in dt-schema; see `additionalProperties: true`.

I had considered implementing something like this years ago, because as different kinds of bindings co-mingle - generic, vendor-specific, OS-specific - they can't always be arranged in a linear pattern of inclusion (most to least specific description of the device). My own motivating example was `zephyr,memory-region`, which is not really hardware description, but it's often used together with `mmio-sram` and could be paired with other memory compatibles.

I'm too invested in this at the moment, but wanted to give my two cents.

decsny, 2025-06-11:
> To be fair, there is a precedent for allowing devicetree properties to be mixed in from multiple bindings in dt-schema; see `additionalProperties: true`.
> 
> I had considered implementing something like this years ago, because as different kinds of bindings co-mingle - generic, vendor-specific, OS-specific - they can't always be arranged in a linear pattern of inclusion (most to least specific description of the device). My own motivating example was `zephyr,memory-region`, which is not really hardware description, but it's often used together with `mmio-sram` and could be paired with other memory compatibles.
> 
> I'm not too invested in this at the moment, but wanted to give my two cents.

AFAIK @mbolivar  was already taking on some work to just straight up introduce dt-schema to zephyr, not sure what's the status of that anymore though.

I actually would be fine with having some analogue to `additionalProperties`,  but I'm still a bit leery/concerned might introduce problems with all the weird inheritance based bindings rules that we have for zephyr, which have already been a headache as it is. @dottspina and @rruuaanng have been much more involved on the script maintenance than me lately, so maybe they have an opinion about that.

swift-tk, 2025-06-12:
> > > -1 ❌
> > > DT compatible is supposed to describe what device programming model a device has, from most specific to least specific. Utility bindings are antithesis to the DT paradigm of matching compatible drivers to devices. This also would make DT even more unclear and confusing to people than it already is. Not to mention the extra maintenance effort for DT in order to handle all the new edge cases and complication caused by these abilities to "merge" bindings and getting fingers pointed about who is supposed to be handling conflicts in the definitions of the bindings between one another. Utility bindings also seem like they are just a way to encourage more SW configuration in DT, when, for example with what your case is, there's no reason why you can't code that behavior into your source code at the board level, so I don't see why this is even needed.
> > 
> > 
> > First of all, it is not a utility compatiable, the timing setting is a hardware setting and is required for MSPI to run at high speed. I think this is also true for other platforms and the timing parameters are all somewhat different. I'm not saying we should merge all the compat without constraints. For complications you mentioned, please elaborate. IOM, if we set the constriants carefully, there would not be any problem at all.
> 
> When I say utility compatible, I'm referring to the fact that the compatible string does not match to a driver which can appreciate a full and valid programming model for the device. This compatible in the example you're giving not only doesn't represent a full programming model for the device being described by the node, but it doesn't even describe part of one at all. It describes a piece of a _specific usage_ (not a programming model) of a different device (the spi controller), so it doesn't make sense to exist as a compatible, it's not valid DT usage.
> 
> As for complications and unexpected situations, they always happen in programming. I think pretending like I would be able to explain to you all of the possible issues that will arise is either naive or dishonest. Any time you add a feature you expose yourself to the possibility of more bugs and edge cases and wrong or missing assumptions. So since this feature you're proposing to add is against one of the core design philosophies of DT, I don't want to add it to take on this maintenance burden and the inevitable new set of ongoing issues popping up around it.

That's so not true lol. The "ambiq,mspi-device" includes the generic `mspi-device.yaml` which is also included by "mspi-aps-z8". If it is not full, then it wouldn't be able to compile under current Zephyr architecture....

The `additionalProperties: true` could potentially resolve my case too. If I understand it correctly, it looks similar with adding an additional keyword(my version, a base keyword) to allow merging (if both compat have the same base keyword).

decsny, 2025-06-12:
> That's so not true lol. The "ambiq,mspi-device" includes the generic `mspi-device.yaml` which is also included by "mspi-aps-z8". If it is not full, then it wouldn't be able to compile under current [...]

The fact that a device can be on the bus of an ambiq mspi controller does not tell me anything about how to program to it or what it even is. That's why it is not describing a compatible programming model. End of story.

bjarki-andreasen, 2025-06-13:
-1 as this goes against the devicetree specification, and does not solve the root issue.

The root of your issue is a layering violation, you want to insert information into a node/layer where it does not belong, and the devicetree specification is helping prevent this. The devicetree specification is "inflexible" for a reason, it sets restrictions so everyone does not run in their own direction, repeating the same mistakes that are what formed the specification to begin with :)

The issue you are running into here we have already solved with clocks and pinctrl. Think of an i2c controller and a device on that bus. What specific pin mux/pull/bias/power etc. should be applied to the pins when the i2c controller wants to transfer something, what if we change the bitrate from 100kHz or 1MHz, or if the i2c controller should enter some low power mode. which clock config does the i2c controller use in these different scenarios? pinctrl and clock settings are soc and board specific, whereas the i2c controller and target device are mostly independent, do we try to merge a specific socs clock and pinctrl settings into the target device or i2c controller? no... That's not where they belong. Instead, we create soc/board specific nodes to store the information,give them a friendly name like `pinctrl_i2c0_fast` or `clock_i2c0_100` or whatever, and point to them.

Please consider looking into pinctrl for example, there is a good reason they are defined like they are, where they are https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/nordic/nrf5340dk/nrf5340_cpuapp_common-pinctrl.dtsi

swift-tk, 2025-06-16:
> -1 as this goes against the devicetree specification, and does not solve the root issue.
>
> The root of your issue is a layering violation, you want to insert information into a node/layer where it [...]

First of all, I don't think it is wrong to put timing on device level as they are heavily influenced by device states (frequency, io mode, dqs/nondqs mode, sdr/ddr and etc), so they should be managed by device itself, putting it at any other place makes it less transparent. Further, it makes the controller liable to know the device states and a generic controller which is expected to work with all kinds of device types should not. The pinctrl approach brings even more extra level of complexity to the management of the timing parameters and forcing controller to know its devices' state, it is at best a comlicated work around...

So I think either we could have a whole bunch of platform specific binds that includes each generic device binding or consider the `additionalProperties` approach.

closed 2026-09-16 by github-actions[bot]