zephyrproject-rtos/zephyr #91252: build problem with zephyr and lvgl

Senator3223, 2025-06-07:
### Describe the bug

 i am trying to build my miryoku firmware with nice-view-gem locally but i get the following error on windows 11 with a fresh install of zmk, miryoku-zmk and i have updated west after modifying west.yml as described in the readme. Because of the error i think this might be the place to report. I am a noob however so correct me if im wrong.

i run:
```
west build -b nice_nano_v2 -- -DSHIELD="corne_left nice_view_adapter nice_view_gem" -DZMK_CONFIG="C:/Users/edson/git/miryoku_zmk/config"
```

i get this stacktrace:
```
-- west build: building application
[2/532] Generating include/generated/version.h
-- Zephyr version: 3.5.0 (C:/Users/edson/git/zmk/zephyr), build: 489d3b00c49d
[98/532] Building C object zephyr/CMakeFiles/zephyr.dir/C_...view-gem/boards/shields/nice_view_gem/widgets/screen.c.obj
FAILED: zephyr/CMakeFiles/zephyr.dir/C_/Users/edson/git/zmk/nice-view-gem/boards/shields/nice_view_gem/widgets/screen.c.obj
C:\PROGRA~1\ZEPHYR~1.3\ARM-ZE~1\bin\AR1445~1.EXE -DKERNEL -DLV_CONF_INCLUDE_SIMPLE=1 -DLV_CONF_PATH=C:/Users/edson/git/zmk/zephyr/modules/lvgl/include/lv_conf.h -DNRF52840_XXAA -DPICOLIBC_INTEGER_PRINTF_SCANF -D_FORTIFY_SOURCE=1 -D_POSIX_C_SOURCE=200809 -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR__=1 -IC:/Users/edson/git/zmk/zephyr/kernel/include -IC:/Users/edson/git/zmk/zephyr/arch/arm/include -IC:/Users/edson/git/zmk/app/include -IC:/Users/edson/git/zmk/zephyr/include -IC:/Users/edson/git/zmk/app/build/zephyr/include/generated -IC:/Users/edson/git/zmk/zephyr/soc/arm/nordic_nrf/nrf52 -IC:/Users/edson/git/zmk/zephyr/soc/arm/nordic_nrf/common/. -IC:/Users/edson/git/zmk/zephyr/subsys/usb/device -IC:/Users/edson/git/zmk/zephyr/subsys/bluetooth/controller/ll_sw/nordic/hal/nrf5/nrfx_glue -IC:/Users/edson/git/zmk/zephyr/subsys/bluetooth -IC:/Users/edson/git/zmk/zephyr/subsys/settings/include -IC:/Users/edson/git/zmk/modules/hal/cmsis/CMSIS/Core/Include -IC:/Users/edson/git/zmk/zephyr/modules/cmsis/. -IC:/Users/edson/git/zmk/modules/hal/nordic/nrfx -IC:/Users/edson/git/zmk/modules/hal/nordic/nrfx/drivers/include -IC:/Users/edson/git/zmk/modules/hal/nordic/nrfx/mdk -IC:/Users/edson/git/zmk/zephyr/modules/hal_nordic/nrfx/. -IC:/Users/edson/git/zmk/modules/lib/gui/lvgl/src -IC:/Users/edson/git/zmk/zephyr/modules/lvgl/include -IC:/Users/edson/git/zmk/modules/crypto/tinycrypt/lib/include -IC:/Users/edson/git/zmk/app/module/include -IC:/Users/edson/git/zmk/app/module/drivers/sensor/battery/. -fno-strict-aliasing -Os -imacros C:/Users/edson/git/zmk/app/build/zephyr/include/generated/autoconf.h -fno-printf-return-value -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m4 -mthumb -mabi=aapcs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mfp16-format=ieee "--sysroot=C:/Program Files/zephyr-sdk-0.16.3/arm-zephyr-eabi/arm-zephyr-eabi" -imacros C:/Users/edson/git/zmk/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -ftls-model=local-exec -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=C:/Users/edson/git/zmk/app=CMAKE_SOURCE_DIR -fmacro-prefix-map=C:/Users/edson/git/zmk/zephyr=ZEPHYR_BASE -fmacro-prefix-map=C:/Users/edson/git/zmk=WEST_TOPDIR -ffunction-sections -fdata-sections --specs=picolibc.specs -std=c99 -Wfatal-errors -MD -MT zephyr/CMakeFiles/zephyr.dir/C_/Users/edson/git/zmk/nice-view-gem/boards/shields/nice_view_gem/widgets/screen.c.obj -MF zephyr\CMakeFiles\zephyr.dir\C_\Users\edson\git\zmk\nice-view-gem\boards\shields\nice_view_gem\widgets\screen.c.obj.d -o zephyr/CMakeFiles/zephyr.dir/C_/Users/edson/git/zmk/nice-view-gem/boards/shields/nice_view_gem/widgets/screen.c.obj -c C:/Users/edson/git/zmk/nice-view-gem/boards/shields/nice_view_gem/widgets/screen.c
In file included from c:\users\edson\git\zmk\modules\lib\gui\lvgl\src/misc/lv_log.h:16,
                 from c:\users\edson\git\zmk\modules\lib\gui\lvgl\lvgl.h:25,
                 from C:/Users/edson/git/zmk/modules/lib/gui/lvgl/src/lvgl.h:17,
                 from C:/Users/edson/git/zmk/nice-view-gem/boards/shields/nice_view_gem/widgets/battery.h:3,
                 from C:/Users/edson/git/zmk/nice-view-gem/boards/shields/nice_view_gem/widgets/screen.c:21:
c:\users\edson\git\zmk\modules\lib\gui\lvgl\src\lv_conf_internal.h:35:42: fatal error: c:\users\edson\git\zmk\modules\lib\gui\lvgl\src\((((0x07) << 16) | (0x06))):/Users/edson/git/zmk/zephyr/modules/lvgl/include/lv_conf.h: Invalid argument
   35 |         #include __LV_TO_STR(LV_CONF_PATH)
      |                                          ^
compilation terminated.
[115/532] Building C object zephyr/CMakeFiles/zephyr.dir/subsys/usb/device/usb_device.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' --build 'C:\Users\edson\git\zmk\app\build'
```

### Regression

- [ ] This is a regression.

### Relevant log output

```shell

```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

faxe1008, 2025-06-08:
@Senator3223 can you please provide the missing information? Also dump what `LV_CONF_PATH` is set to.

faxe1008, 2025-06-12:
@Senator3223 Zephyr sets LV_CONF_PATH here:
https://github.com/zephyrproject-rtos/zephyr/blob/489d3b00c49d1d3a0d4ec1f693fa0719db38bbf9/modules/lvgl/CMakeLists.txt#L19C1-L20C1

The mangled output from your log suggests that therein lies the issue somewhere:
```
c:\users\edson\git\zmk\modules\lib\gui\lvgl\src\lv_conf_internal.h:35:42: fatal error: c:\users\edson\git\zmk\modules\lib\gui\lvgl\src\((((0x07) << 16) | (0x06))):/Users/edson/git/zmk/zephyr/modules/lvgl/include/lv_conf.h: Invalid argument
```

Sounds a bit like this issue:
https://github.com/lvgl/lvgl/issues/7858#issuecomment-2958396200

Can you try modifying to that version?
https://github.com/zephyrproject-rtos/zephyr/blob/main/modules/lvgl/CMakeLists.txt#L19

closed 2025-08-16 by github-actions[bot]