zephyrproject-rtos/zephyr #91317: Logging subsystem not notifying backends if CONFIG_LOG_PROCESS_TRIGGER_THRESHOLD=1

LucasRibeiro-Zuhlke, 2025-06-09:
### Describe the bug

Zephyr version: 3.7.1
CONFIG_LOG_PROCESS_TRIGGER_THRESHOLD=1
CONFIG_LOG_PROCESS_THREAD_CUSTOM_PRIORITY=y
CONFIG_LOG_PROCESS_THREAD_PRIORITY=0

On the following code, inside `log_process_thread_func()`:

[image]

Having the logging thread with high priority, the variable `processed_any` never gets set to `true` because there's no more log messages in the queue and `log_process()` always returns `false`. This breaks code expecting `LOG_BACKEND_EVT_PROCESS_THREAD_DONE`.

To workaround this you need to use threshold >=2. Any preferred solution for a pull request?

### Regression

- [ ] This is a regression.

### Relevant log output

```shell

```

### Impact

Functional Limitation – Some features not working as expected, but system usable.

akamaliz, 2025-07-16:
I am interested in a solution for this as well, my use-case is I have cases where there are sporadic logs like 1 evert second and the notify does not happen.

closed 2025-09-30 by github-actions[bot]

LucasRibeiro-Zuhlke, 2025-09-30:
Please reopen the issue, I will try to fix the CI problem from the MR.