zephyrproject-rtos/zephyr #91421: net_buf API for `const` data

Thalley, 2025-06-11:
### Summary

The current `net_buf` (and `net_buf_simple`) today only works on `void *data` and stores the data in a `uint8_t *data` pointer. 

However a common use of the `net_buf` API is also when handling incoming remote data, e.g. in Bluetooth. 

For example, when reading data in GATT, there is the 

```c
typedef uint8_t (*bt_gatt_read_func_t)(struct bt_conn *conn, uint8_t err,
				    struct bt_gatt_read_params *params,
				    const void *data, uint16_t length);
```

callback, which provides a `const void *data` pointer, which makes sense as you, as the receiver, should not modify the received data. A common approach to parse the received data, as it is often structured data defined by Bluetooth services, is to put the data into a `net_buf` and then pull the individual values using the e.g. `net_buf_pull_*` or `net_buf_simple_pull_*` functions. 

However in order to put the `const` data into a net_buf, it need to be cast to `void *` from `const void *` which isn't allowed by [rule 70 of the Zephyr coding guidelines](https://gitlab.com/MISRA/MISRA-C/MISRA-C-2012/Example-Suite/-/blob/master/R_11_08.c).

### Describe the solution you'd like

Define and implement an API that can be used to parse `const void *` data. The API should either be a near-duplicate of the existing `net_buf` API, but with support for `const void *` data, or an entire new API, to help ensure that functions that will/should only pull do not modify the data in the process (either intentionally, or by accident). 

### Alternatives

Alternatively an exception could/should be made for this. 

### Additional Context

Somewhat related to https://github.com/zephyrproject-rtos/zephyr/issues/91418

Thalley, 2025-06-11:
> > [@jhedberg](https://github.com/jhedberg) not sure if I should be assigned to this :) While I did report/request it, it's doubtful that I'll manage to work on it
> 
> I can leave it unassigned as well, however in that case there's a very high chance that nothing will be done about this.

Yeah, you are probably right. It also somewhat depends on the outcome of https://github.com/zephyrproject-rtos/zephyr/issues/91418 as if the remove the coding guideline, then this API has less value.

jhedberg, 2025-06-11:
Or to put it another way: I don't think unassigned (i.e. without a commitment to implement) enhancement requests are particularly useful for an open source project. As the maintainer, the request is useful for me to see and evaluate so that I can give my support for it or to reject it (with good justification), however Zephyr is not a project where people can request stuff and expect the project to implement it for them (especially non-trivial stuff that no-one else has a need for).

Thalley, 2025-06-17:
I think this feature is mostly used by Bluetooth to pull information from incoming data. We could consider a BT-specific API and implementation (and possibly even just an internal API, although we do use `net_buf`s to provide data (which should be read-only) to upper layers in e.g. ISO and L2CAP

jhedberg, 2026-09-08:
Narrowing this to what the data supports. The need is only for `net_buf_simple` as a read-only parsing view; pooled `net_buf`s never wrap external data. In `subsys/bluetooth`, 56 of the 81 `net_buf_simple_init_with_data()` calls cast away `const`, nearly all of them in LE Audio parsing received data, and new code keeps working around the gap by parsing bytes directly (the recent HCI packet helpers take `const uint8_t *` for exactly this reason). A full parallel API is more than this needs. The small version is one initializer for const data whose contract is pull-only, with the single cast and its rationale inside the library rather than at every call site, plus a sweep of those sites. The type system still cannot enforce read-only, but the deviation becomes one audited place instead of 56.

Thalley, 2026-09-08:
> A full parallel API is more than this needs.

I'm not sure I follow how that can be. Besides needing a way to provide `const` data to `net_buf` (without explicitly casting it away, since that is a violation of our coding guidelines), functions like `net_buf_simple_pull`, `net_buf_simple_pull_mem` and `net_buf_simple_tail` shall return `const *`

> with the single cast 

If you are referring to casting away the `const` part of the data, then that is disallowed as per coding guidelines rule 70, so that is not an option

jhedberg, 2026-09-08:
You're right on the type point: an initializer alone leaves `pull_mem()` and friends handing out `void *` into const storage, so the view is only read-only if its accessors return const. That is a new type, but the read half only: init from `const void *`, `len`, and the 15 pull variants (`u8`, the six `le` and six `be` widths, `pull`, `pull_mem`); none of the current call sites use `remove_*` or `tail`. The sites are 60 casts in 22 files, all under Bluetooth. What I cannot size without a prototype is where the pulled pointers land: most stay local, but a few go into non-const fields or out-parameters (`bap_base.c`, `sdp.c`), and each of those either takes a public const change or brings the cast back. The library-side cast would be a violation too, so scrap that. It would join the 60 we ship today, with no CI check for rule 70 and no deviation mechanism in the guidelines, which is the actual gap #91418 describes. Whether the view is worth building still comes down to someone committing to it.

Thalley, 2026-09-08:
Additionally, all the functions that modify the content of the net_buf (i.e. all `net_buf_simple_add*` and `net_buf_simple_push*` should also not be allowed for the "read only" net_buf.