zephyrproject-rtos/zephyr #91433: Name internal functions with less generic names

joolzg, 2025-06-11:
### Summary

Just spent ages trying to track down  a crash in zephyr master.

Finally found the culprit and it was the idle task being overwritten by a variable i created.

Renaming my variable to idle_data and voila  the system is working again.

I got no link errors or warnings.

### Describe the solution you'd like

Name all internal tasks/stacks to something a bit more precise for example idle becomes idle_task

joolzg, 2025-06-11:
idle.c and the task idle. I added a variable "uint8_t  idle[256]" and the code crashed. After hours of debugging i renamed this and everything worked.

As i said no warnings when building.

rruuaanng, 2025-06-16:
I'm just passing by, but I'd suggest using `__z_` as a prefix for internal global variable. Would that be a better option? :)

For public variables, I suggest providing a dedicated set of macros to manage them, ensuring proper encapsulation. Like this: #86801

joolzg, 2025-06-19:
I think it would be good to either
1. Change internal task names to use a special naming convention so that there is less of  chance to have duplicate names used in the application, as mentions in by @rruuaanng adding a prefix to show this is an internal function would be good.
2. Make these functions static so they don't pollute the global name pool, idle cannot as it is referenced in a different file.

The problem arises if you have a global variable called idle, this in my case overrides the idle task and crashes zephyr.

closed 2026-09-16 by github-actions[bot]