zephyrproject-rtos/zephyr #91483: STM32U585: USB MSC : can't pass Compliance Test by using USB3CV

testo-jeff, 2025-06-12:
### Describe the bug

USB MSC test failed when using testo tool USB3CV, below is the screenshot:

[image]

Per debug, Zephyr can receive command A8 and 9E from USB3CV, and Zephyr USB protocol layer has handled the command, but in low layer(probably ST driver? not sure) can't send out the response to USB3CV, lead to test failed(100% reproduced).

### Regression

- [ ] This is a regression.

### Steps to reproduce

Configurations:
CONFIG_USB_DEVICE_STACK_NEXT=y
CONFIG_USB_MASS_STORAGE=y
CONFIG_MAIN_STACK_SIZE=2048
CONFIG_USBD_PID=0x0045
CONFIG_USBD_VID=0x128d
CONFIG_USBD_PRODUCT="xxx Test kit"
CONFIG_USBD_MANUFACTURER="xxx CO.,LTD"
CONFIG_USBD_MSC_CLASS=y
CONFIG_USBD_MSC_LUNS_PER_INSTANCE=3
CONFIG_DISK_ACCESS=y
CONFIG_DISK_DRIVERS=y
CONFIG_FILE_SYSTEM=y
CONFIG_FAT_FILESYSTEM_ELM=y
CONFIG_UDC_STM32=y
CONFIG_USB_DC_STM32=n

### Relevant log output

```shell

```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

### Environment

- OS: Windows 10/11
- USB3CV: v3.1.8.0

testo-jeff, 2025-06-12:
@JarmouniA  Updated, would you please check again? Thx.

marwaiehm-st, 2025-06-30:
Hello @testo-jeff,

The issue was not reproduced. Below is the test result screenshot: 

[image]

I shared the  MSC Test report : [MSC_Tests_Report.zip](https://github.com/user-attachments/files/20977696/MSC_Tests_Report.zip)

Environment

- OS: Windows 11
- USB3CV version 3.1.9.0
- Zephyr v4.1.0 (commit c906d4a1039b857b56556f00e4abf0eb9afc8e7e)

nandojve, 2025-06-30:
Hi @marwaiehm-st ,

I can see good results from your test.
Can we expect to have one fully compliance with Windows and Linux working for v4.2 release?
What is the current status of USB-NEXT ? Which one we should use to release a product at moment ?

marwaiehm-st, 2025-07-07:
@testo-jeff ,

I have tried to test the USB-C interface on my board following the same procedure, but I did not observe the same LED status as in your picture. In my case, the LED blinks red.

Additionally, the USB3CV app on my PC does not detect the USB port when connected via USB-C, which suggests the device is not enumerating correctly over the USB-C interface.

Could you please confirm if there are any additional device tree (DTS) modifications or other configurations required for USB-C support on this board? .

testo-jeff, 2025-07-09:
@marwaiehm-st  Not only "Command Set Test.",  the first two tests also fail on my side. So we are on the same test result now. We need to all of test passed.

testo-jeff, 2025-07-23:
@jfischer-no  Thanks for sharing your patch, we've tested again with your patch and test result is passed.

@marwaiehm-st  Would you please review jfischer-no's patch and fix it? Thanks.

marwaiehm-st, 2025-07-29:
@testo-jeff, could you please retest using the linked PR ?

FRASTM, 2025-09-22:
could this issue be closed as PR is now merged ?

nandojve, 2025-09-22:
Hi @FRASTM ,

> could this issue be closed as PR is now merged ?

The halt is still not fixed for HS, see https://github.com/zephyrproject-rtos/zephyr/pull/95857#issuecomment-3302029406

nandojve, 2025-09-22:
CC: @erwango

mathieuchopstm, 2025-10-01:
Now that #96028 is merged, everything should be functional.

@testo-jeff please report if errors are still observed or if problem is fixed. Note that you may need to build with enlarged UDC pool for tests to pass properly: see https://github.com/zephyrproject-rtos/zephyr/pull/95857#issuecomment-3281427754