zephyrproject-rtos/zephyr #91521: ci: win11: check_compliance.py is broken

butok, 2025-06-12:
### Describe the bug

Running of the scripts/ci/check_compliance.py script on Windows OS is broken.

### Regression

- [ ] This is a regression.

### Steps to reproduce

Steps:

1. Be sure that all compliance requirements are installed. Run `pip install -r scripts\requirements-compliance.txt` log:
```
Ignoring python-magic: markers 'sys_platform != "win32"' don't match your environment
Requirement already satisfied: clang-format>=15.0.0 in c:\users\nxa12205\appdata\local\programs\python\python310\lib\site-packages (from -r scripts\requirements-compliance.txt (line 5)) (20.1.0)
Requirement already satisfied: gitlint in c:\users\nxa12205\appdata\local\programs\python\python310\lib\site-packages (from -r scripts\requirements-compliance.txt (line 6)) (0.19.1)
Requirement already satisfied: junitparser>=2 in c:\users\nxa12205\appdata\local\programs\python\python310\lib\site-packages (from -r scripts\requirements-compliance.txt (line 7)) (3.2.0)
Requirement already satisfied: lxml>=5.3.0 in c:\users\nxa12205\appdata\local\programs\python\python310\lib\site-packages (from -r scripts\requirements-compliance.txt (line 8)) (5.3.1)
Requirement already satisfied: pykwalify in c:\users\nxa12205\appdata\local\programs\python\python310\lib\site-packages (from -r scripts\requirements-compliance.txt (line 9)) (1.8.0)
Requirement already satisfied: pylint>=3 in c:\users\nxa12205\appdata\local\programs\python\python310\lib\site-packages (from -r scripts\requirements-compliance.txt (line 10)) (3.3.5)
Requirement already satisfied: python-magic-bin in c:\users\nxa12205\appdata\local\programs\python\python310\lib\site-packages (from -r scripts\requirements-compliance.txt (line 11)) (0.4.14)
Requirement already satisfied: ruff==0.11.11 in c:\users\nxa12205\appdata\local\programs\python\python310\lib\site-packages (from -r scripts\requirements-compliance.txt (line 13)) (0.11.11)
Requirement already satisfied: sphinx-lint in c:\users\nxa12205\appdata\local\programs\python\python310\lib\site-packages (from -r scripts\requirements-compliance.txt (line 14)) (1.0.0)
Requirement already satisfied: unidiff in c:\users\nxa12205\appdata\local\programs\python\python310\lib\site-packages (from -r scripts\requirements-compliance.txt (line 15)) (0.7.5)
Requirement already satisfied: vermin in c:\users\nxa12205\appdata\local\programs\python\python310\lib\site-packages (from -r scripts\requirements-compliance.txt (line 16)) (1.6.0)
Requirement already satisfied: yamllint in c:\users\nxa12205\appdata\local\programs\python\python310\lib\site-packages (from -r scripts\requirements-compliance.txt (line 17)) (1.36.2)
Requirement already satisfied: gitlint-core==0.19.1 in c:\users\nxa12205\appdata\local\programs\python\python310\lib\site-packages (from gitlint-core[trusted-deps]==0.19.1->gitlint->-r scripts\requirements-compliance.txt (line 6)) (0.19.1)
Requirement already satisfied: arrow>=1 in c:\users\nxa12205\appdata\local\programs\python\python310\lib\site-packages (from gitlint-core==0.19.1->gitlint-core[trusted-deps]==0.19.1->gitlint->-r scripts\requirements-compliance.txt (line 6)) (1.2.3)
Requirement already satisfied: click>=8 in c:\users\nxa12205\appdata\local\programs\python\python310\lib\site-packages (from gitlint-core==0.19.1->gitlint-core[trusted-deps]==0.19.1->gitlint->-r scripts\requirements-compliance.txt (line 6)) (8.1.3)
Requirement already satisfied: python-dateutil>=2.7.0 in c:\users\nxa12205\appdata\local\programs\python\python310\lib\site-packages (from arrow>=1->gitlint-core==0.19.1->gitlint-core[trusted-deps]==0.19.1->gitlint->-r scripts\requirements-compliance.txt (line 6)) (2.9.0.post0)
Requirement already satisfied: colorama in c:\users\nxa12205\appdata\local\programs\python\python310\lib\site-packages (from click>=8->gitlint-core==0.19.1->gitlint-core[trusted-deps]==0.19.1->gitlint->-r scripts\requirements-compliance.txt (line 6)) (0.4.6)
Requirement already satisfied: docopt>=0.6.2 in c:\users\nxa12205\appdata\local\programs\python\python310\lib\site-packages (from pykwalify->-r scripts\requirements-compliance.txt (line 9)) (0.6.2)
Requirement already satisfied: ruamel.yaml>=0.16.0 in c:\users\nxa12205\appdata\local\programs\python\python310\lib\site-packages (from pykwalify->-r scripts\requirements-compliance.txt (line 9)) (0.18.6)
Requirement already satisfied: dill>=0.2 in c:\users\nxa12205\appdata\local\programs\python\python310\lib\site-packages (from pylint>=3->-r scripts\requirements-compliance.txt (line 10)) (0.3.9)
Requirement already satisfied: platformdirs>=2.2.0 in c:\users\nxa12205\appdata\local\programs\python\python310\lib\site-packages (from pylint>=3->-r scripts\requirements-compliance.txt (line 10)) (4.3.6)
Requirement already satisfied: astroid<=3.4.0-dev0,>=3.3.8 in c:\users\nxa12205\appdata\local\programs\python\python310\lib\site-packages (from pylint>=3->-r scripts\requirements-compliance.txt (line 10)) (3.3.9)
Requirement already satisfied: isort!=5.13.0,<7,>=4.2.5 in c:\users\nxa12205\appdata\local\programs\python\python310\lib\site-packages (from pylint>=3->-r scripts\requirements-compliance.txt (line 10)) (5.13.2)
Requirement already satisfied: mccabe<0.8,>=0.6 in c:\users\nxa12205\appdata\local\programs\python\python310\lib\site-packages (from pylint>=3->-r scripts\requirements-compliance.txt (line 10)) (0.7.0)
Requirement already satisfied: tomli>=1.1.0 in c:\users\nxa12205\appdata\local\programs\python\python310\lib\site-packages (from pylint>=3->-r scripts\requirements-compliance.txt (line 10)) (2.0.2)
Requirement already satisfied: tomlkit>=0.10.1 in c:\users\nxa12205\appdata\local\programs\python\python310\lib\site-packages (from pylint>=3->-r scripts\requirements-compliance.txt (line 10)) (0.13.2)
Requirement already satisfied: typing-extensions>=4.0.0 in c:\users\nxa12205\appdata\local\programs\python\python310\lib\site-packages (from astroid<=3.4.0-dev0,>=3.3.8->pylint>=3->-r scripts\requirements-compliance.txt (line 10)) (4.12.2)
Requirement already satisfied: polib in c:\users\nxa12205\appdata\local\programs\python\python310\lib\site-packages (from sphinx-lint->-r scripts\requirements-compliance.txt (line 14)) (1.2.0)
Requirement already satisfied: regex in c:\users\nxa12205\appdata\local\programs\python\python310\lib\site-packages (from sphinx-lint->-r scripts\requirements-compliance.txt (line 14)) (2024.11.6)
Requirement already satisfied: pathspec>=0.5.3 in c:\users\nxa12205\appdata\local\programs\python\python310\lib\site-packages (from yamllint->-r scripts\requirements-compliance.txt (line 17)) (0.12.1)
Requirement already satisfied: pyyaml in c:\users\nxa12205\appdata\local\programs\python\python310\lib\site-packages (from yamllint->-r scripts\requirements-compliance.txt (line 17)) (6.0.2)
Requirement already satisfied: six>=1.5 in c:\users\nxa12205\appdata\local\programs\python\python310\lib\site-packages (from python-dateutil>=2.7.0->arrow>=1->gitlint-core==0.19.1->gitlint-core[trusted-deps]==0.19.1->gitlint->-r scripts\requirements-compliance.txt (line 6)) (1.16.0)
Requirement already satisfied: ruamel.yaml.clib>=0.2.7 in c:\users\nxa12205\appdata\local\programs\python\python310\lib\site-packages (from ruamel.yaml>=0.16.0->pykwalify->-r scripts\requirements-compliance.txt (line 9)) (0.2.12)
```
3. Run the compliance check python scripts/ci/check_compliance.py
```
Running Nits             tests in C:\_ddm\ZEPHYR\zephyr_main\zephyr ...
Running Pylint           tests in C:\_ddm\ZEPHYR\zephyr_main\zephyr ...
Running SysbuildKconfig  tests in C:\_ddm\ZEPHYR\zephyr_main\zephyr ...
Running ModulesMaintainers tests in C:\_ddm\ZEPHYR\zephyr_main\zephyr ...
Running Identity         tests in C:\_ddm\ZEPHYR\zephyr_main\zephyr ...
Running KconfigHWMv2     tests in C:\_ddm\ZEPHYR\zephyr_main\zephyr ...
Running KconfigBasicNoModules tests in C:\_ddm\ZEPHYR\zephyr_main\zephyr ...
Running Kconfig          tests in C:\_ddm\ZEPHYR\zephyr_main\zephyr ...
Running ClangFormat      tests in C:\_ddm\ZEPHYR\zephyr_main\zephyr ...
Running PythonCompat     tests in C:\_ddm\ZEPHYR\zephyr_main\zephyr ...
Running BinaryFiles      tests in C:\_ddm\ZEPHYR\zephyr_main\zephyr ...
Running Gitlint          tests in C:\_ddm\ZEPHYR\zephyr_main\zephyr ...
Running GitDiffCheck     tests in C:\_ddm\ZEPHYR\zephyr_main\zephyr ...
Running SysbuildKconfigBasicNoModules tests in C:\_ddm\ZEPHYR\zephyr_main\zephyr ...
Running SphinxLint       tests in C:\_ddm\ZEPHYR\zephyr_main\zephyr ...
[... cut ...]
  File "C:\Users\nxa12205\AppData\Local\Programs\Python\Python310\lib\site-packages\magic\magic.py", line 89, in from_file
    return maybe_decode(magic_file(self.cookie, filename))
  File "C:\Users\nxa12205\AppData\Local\Programs\Python\Python310\lib\site-packages\magic\magic.py", line 255, in magic_file
    return _magic_file(cookie, coerce_filename(filename))
ctypes.ArgumentError: argument 2: TypeError: wrong type

```

### Relevant log output

```shell

```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

### Environment

- OS: Windows11

danieldegrasse, 2025-06-17:
Closing this out- we don't currently support checkpatch on windows, as it is intended to be Linux specifc

butok, 2025-06-18:
Hi @danieldegrasse 

The check_compliance.py script should be supported on Windows.
This is a regression issue, as it was runnable before.
If any check is OS-specific, the Python test should avoid running these checks on unsupported operating systems.

Reopening, as the issue is not solved.

pdgendt, 2025-06-18:
@butok check if #91814 fixes your issue. I do see that `python-magic-bin` is super old (last release 2017), so not sure how long we can rely on it.

butok, 2025-06-18:
> [@butok](https://github.com/butok) check if [#91814](https://github.com/zephyrproject-rtos/zephyr/pull/91814) fixes your issue. I do see that `python-magic-bin` is super old (last release 2017), so not sure how long we can rely on it.

Great! It has solved the issue.

PR #91814 (2025-06-20): scripts: ci: check_compliance: Pass path representation to magic
On windows the `Magic.from_file` method fails to convert a `PathLike` to its representation.
Pass using `os.fspath`.

Fixes #91521