zephyrproject-rtos/zephyr #91576: Bluetooth: TBS: Cannot set URI scheme list for GTBS

Thalley, 2025-06-13:
### Describe the bug

The `bt_tbs_set_uri_scheme_list` function does not properly handle setting an URI scheme list for GTBS. This means that in case of using only GTBS, it's not possible to set any URI scheme, which shouldn't be the case. 

The function should use `inst_lookup_index(bearer_index)` instead of `&svc_insts[bearer_index];` and allow setting URI schemes specifically for the GTBS. 

Additionally the function should be updated so that instead of copying all URI scheme lists into `uri_scheme_buf` (which is only used for logging purposes?), the complete list, when notifying for GTBS, should be constructed dynamically similar to e.g. `bearer_list_current_calls_changed` and `read_current_calls`. 

### Regression

- [ ] This is a regression.

### Steps to reproduce

N/A

### Relevant log output

```shell
N/A
```

### Impact

Functional Limitation – Some features not working as expected, but system usable.

### Environment

Commit SHA: f8b70ee1794803e76b7db28c7eb06e475fc61667

### Additional Context

Found in https://github.com/zephyrproject-rtos/zephyr/pull/91547

Thalley, 2025-06-16:
@nirav-agrawal my thought to implement this is to keep `gtbs_inst.uri_scheme_list` as only the values specific for GTBS, without any of the TBS URI schemes. 

Then on notification or read, we implement a function similar to `net_buf_put_current_calls` that takes all the URI schemes of GTBS and all the TBS instances and place them in the `read_buf`. That approach is optimized for memory usage, at the cost of some memory copies on read/notify.

nirav-agrawal, 2025-06-18:
Hi @Thalley, thank you for highlighting this issue. Based on your given feedback I added the fix and tested with some random combinations setting URI schemes from multiple TBS instances, and made sure it should not override the entire list and only add new scheme. Also, added dynamic buffer concept to prepare GTBS instance for all URI scheme lists (GTBS+TBS instances). 
Please review the changes in below PR, 

https://github.com/zephyrproject-rtos/zephyr/pull/91833

Regards,
Nirav

closed 2025-08-14 by kartben