zephyrproject-rtos/zephyr #91621: tests/drivers/spi/spi_loopback fails on NXP mr_canhubk3

manuargue, 2025-06-14:
### Describe the bug

`tests/drivers/spi/spi_loopback/drivers.spi.loopback` fails on `mr_canhubk3` board

### Regression

- [ ] This is a regression.

### Steps to reproduce

`west twister -p mr_canhubk3/s32k344 -s tests/drivers/spi/spi_loopback/drivers.spi.loopback`

### Relevant log output

```shell
E: In DEBUG mode, watchdog is disabled
*** Booting Zephyr OS build v4.1.0-5867-g34e47f4040c9 ***
SPI test on buffers TX/RX 0x20404160/0x20404140, frame size = 8, DMA enabled
Polling...Running TESTSUITE spi_extra_api_features
===================================================================
START - test_spi_hold_on_cs
 PASS - test_spi_hold_on_cs in 0.002 seconds
===================================================================
START - test_spi_lock_release
 PASS - test_spi_lock_release in 0.001 seconds
===================================================================
TESTSUITE spi_extra_api_features succeeded
Running TESTSUITE spi_extra_api_features
===================================================================
START - test_spi_hold_on_cs
 PASS - test_spi_hold_on_cs in 0.002 seconds
===================================================================
START - test_spi_lock_release
 PASS - test_spi_lock_release in 0.001 seconds
===================================================================
TESTSUITE spi_extra_api_features succeeded
Running TESTSUITE spi_loopback
===================================================================
Testing loopback spec: SLOW
START - test_nop_nil_bufs
 PASS - test_nop_nil_bufs in 0.001 seconds
===================================================================
START - test_spi_async_call
 PASS - test_spi_async_call in 0.126 seconds
===================================================================
START - test_spi_complete_large_transfers
 PASS - test_spi_complete_large_transfers in 0.125 seconds
===================================================================
START - test_spi_complete_loop_mode_0
Buffer contents are different:
 0x30,0x31,0x32,0x33,0x34,0x35,0x36,0x37,0x38,0x39,0x61,0x62,0x63,0x64,0x65,0x66,0x2d,0x00,
vs:
 0x30,0x31,0xb2,0x33,0xb4,0x35,0xb6,0x37,0xb8,0x39,0xe1,0xe2,0x63,0xe4,0x65,0xe6,0x2d,0x80,
 FAIL - test_spi_complete_loop_mode_0 in 0.020 seconds
===================================================================
START - test_spi_complete_loop_mode_1
 PASS - test_spi_complete_loop_mode_1 in 0.001 seconds
===================================================================
START - test_spi_complete_loop_mode_2
Buffer contents are different:
 0x30,0x31,0x32,0x33,0x34,0x35,0x36,0x37,0x38,0x39,0x61,0x62,0x63,0x64,0x65,0x66,0x2d,0x00,
vs:
 0x30,0x31,0xb2,0x33,0xb4,0x35,0xb6,0x37,0xb8,0x39,0xe1,0xe2,0x63,0xe4,0x65,0xe6,0x2d,0x80,
 FAIL - test_spi_complete_loop_mode_2 in 0.020 seconds
===================================================================
START - test_spi_complete_loop_mode_3
 PASS - test_spi_complete_loop_mode_3 in 0.001 seconds
===================================================================
START - test_spi_complete_multiple
 PASS - test_spi_complete_multiple in 0.001 seconds
[... 191 lines cut ...]
 - PASS - [spi_loopback.test_spi_rx_half_end] duration = 0.001 seconds
 - PASS - [spi_loopback.test_spi_rx_half_start] duration = 0.001 seconds
 - PASS - [spi_loopback.test_spi_same_buf_cmd] duration = 0.001 seconds
 - PASS - [spi_loopback.test_spi_word_size_16] duration = 0.001 seconds
 - FLAKY - [spi_loopback.test_spi_word_size_24] - (Failed 1 of 2 attempts) - duration = 0.022 seconds
 - PASS - [spi_loopback.test_spi_word_size_32] duration = 0.004 seconds
 - PASS - [spi_loopback.test_spi_word_size_7] duration = 0.001 seconds
 - FLAKY - [spi_loopback.test_spi_word_size_9] - (Failed 1 of 2 attempts) - duration = 0.020 seconds
 - PASS - [spi_loopback.test_spi_write_back] duration = 0.001 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
RunID: 42714ebb9fafdb094c8f126d9355ac46
PROJECT EXECUTION FAILED
```

### Impact

Functional Limitation – Some features not working as expected, but system usable.

### Environment

- Zephyr SDK 0.17.1
- 34e47f4040c

decsny, 2025-06-15:
This doesn't look like a regression to me, it looks like new tests were added which didn't exist before, so they couldn't have failed before. Is there a commit where these failing test cases (not the SPI loopback test as a whole) passed?

decsny, 2025-06-15:
In fact I can see in the log that the DMA is enabled and there are already other issues open against these exact test cases for LPSPI with DMA enabled, so I'm going to close this as duplicate

closed 2025-06-15 by decsny

decsny, 2025-06-15:
If the board can't boot because of a SPI bug, please make a separate issue with more details for this board.  I seriously doubt that any of the causes for these test failures have any relation to the cause for that, because we had it working before and those specific test case issues are not regressions from what I found, but feel free to prove that statement wrong.   Also you should not need to even use DMA based SPI to communicate to that watchdog on the S32 board, so I recommend ensuring you're not because it's a more limited driver, as you're discovering.

manuargue, 2025-06-15:
>In fact I can see in the log that the DMA is enabled and there are already other issues open against these exact test cases for LPSPI with DMA enabled. Which btw I already triaged as not regressions. So I'm going to close this as duplicate.

ok

> If the board can't boot because of a SPI bug, please make a separate issue with more details for this board. I seriously doubt that any of the causes for these test failures have any relation to the cause for that

I added the notice `Power up the board with FS26 in debug mode (depends on SPI working correctly)` in case potential bugs of LPSPI driver could affect the FS26 watchdog initialization. It was meant to be a _safe_ testing environment, not a claim that  the board can't boot because of a SPI bug.