zephyrproject-rtos/zephyr #91644: Externalizing Xen public header to module

soburi, 2025-06-15:
### Origin

Xen (MIT licensed public header-only version)
https://github.com/soburi/xen/

Xen (Upstream)
https://xenbits.xen.org/git-http/xen.git

We can use the upstream as the source and edit from there.

### Purpose

Externalizing MIT-licensed Xen public headers from  `include/zephyr/xen/public`.

### Mode of Integration

As a module: proposed name `xen` or `xen-public`

### Maintainership

@soburi

We would also like to ask @firscity, the maintainers of xen-related features, for their cooperation.

### Pull Request

https://github.com/zephyrproject-rtos/zephyr/pull/91643

### Description

Currently, the MIT-licensed Xen public headers have been incorporated
under the `include/zephyr/xen/public`.

However, the recent policy is not to include sources other than Apache-2 in the main tree.
I plan to add MIT files again this time, so I would like to make it an external module.

### Security

There is no particular problem at present.

### Dependencies

No dependencies.

### Version or SHA

b26dba6182bc2d7a14c2278af45cd243db308ad8

### License (SPDX)

MIT

soburi, 2025-06-18:
@ifyall 

> [@soburi](https://github.com/soburi) The upstream link provided in the Origin section doesn't work for me. I found this link, which I assume is correct: https://xenbits.xen.org/gitweb/?p=xen.git.

The URL you provided can be displayed in the browser, but it cannot clone it with git, so I am listing it with the current URL. (It is the URL written on the page you provided.)

firscity, 2025-06-18:
Hi @soburi, thank you for contribution and sorry for late reply!

In general, we had the same idea some time ago and wanted to implement it, I will check your changes this week and will post my comments.

Best regards.

carlescufi, 2025-06-25:
@soburi thanks for the proposal. Today we talked about this change in the TSC, and I mentioned this during the meeting: is this really worth it for a bunch of headers that are updated relatively infrequently? Sure, we would be removing some external headers from the main tree, but at a cost of yet another repo in our `west.yml` and a `west update` that becomes a bit slower.