zephyrproject-rtos/zephyr #91647: failure on bluetooth hci_uart build

m-abubakar, 2025-06-15:
### Describe the bug

Hi,
  I am trying to build bluetooth hci_uart sample for esp32 and nrf5340 development boards and haven't had much success so far. I am running these command to build: 

`west build -p always -b esp32_devkitc/esp32/procpu samples/bluetooth/hci_uart -d build`

**Error Output:**
```
warning: BT_RPA (defined at subsys/bluetooth/common/Kconfig:287) has direct dependencies (BT_HOST_CRYPTO || BT_CTLR_CRYPTO) && BT_HCI && BT with value n, but is currently being y-selected by the following symbols:
 - BT_CTLR_PRIVACY (defined at subsys/bluetooth/controller/Kconfig:677), with value y, direct dependencies BT_CTLR_PRIVACY_SUPPORT && HAS_BT_CTLR && BT_HCI && BT (value: y), and select condition BT_CTLR_PRIVACY_SUPPORT && HAS_BT_CTLR && BT_HCI && BT (value: y)

error: Aborting due to Kconfig warnings

CMake Error at cmake/modules/kconfig.cmake:373 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  cmake/modules/zephyr_default.cmake:131 (include)
  share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:5 (find_package)
```

-- Configuring incomplete, errors occurred!

**NRF 5340**
`west build -p always -b nrf5340dk/nrf5340/cpuapp samples/bluetooth/hci_uart -d build`

**Error Output:**
```
warning: BT_CTLR_DTM_HCI (defined at subsys/bluetooth/controller/Kconfig.dtm:14) was assigned the
value 'y' but got the value 'n'. Check these unsatisfied dependencies: BT_CTLR_DTM_HCI_SUPPORT (=n),
HAS_BT_CTLR (=n). See http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_BT_CTLR_DTM_HCI
and/or look up BT_CTLR_DTM_HCI in the menuconfig/guiconfig interface. The Application Development
Primer, Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual
might be helpful too.

warning: BT_CTLR_ASSERT_HANDLER (defined at subsys/bluetooth/controller/Kconfig:1181) was assigned
the value 'y' but got the value 'n'. Check these unsatisfied dependencies: HAS_BT_CTLR (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_BT_CTLR_ASSERT_HANDLER and/or look up
BT_CTLR_ASSERT_HANDLER in the menuconfig/guiconfig interface. The Application Development Primer,
Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be
helpful too.

warning: HAS_NORDIC_DRIVERS (defined at modules/hal_nordic/Kconfig:7) has direct dependencies 0 with value n, but is currently being y-selected by the following symbols:
 - SOC_SERIES_NRF53X (defined at soc/nordic/Kconfig.soc:36, soc/nordic/nrf53/Kconfig.soc:6, soc/nordic/nrf53/Kconfig:6), with value y, direct dependencies y (value: y), and select condition SOC_FAMILY_NORDIC_NRF (value: y)

warning: CMSIS_CORE_HAS_SYSTEM_CORE_CLOCK (defined at soc/nxp/imxrt/imxrt6xx/Kconfig.defconfig:8, modules/cmsis_6/Kconfig:23) has direct dependencies (SOC_SERIES_IMXRT6XX && SOC_FAMILY_NXP_IMXRT) || (HAS_CMSIS_CORE && 0) with value n, but is currently being y-selected by the following symbols:
 - SOC_FAMILY_NORDIC_NRF (defined at soc/nordic/Kconfig.soc:19, soc/nordic/Kconfig:9), with value y, direct dependencies y (value: y), and select condition ARM (value: y)

warning: HAS_NRFX (defined at modules/hal_nordic/nrfx/Kconfig:4) has direct dependencies 0 with value n, but is currently being y-selected by the following symbols:
 - SOC_SERIES_NRF53X (defined at soc/nordic/Kconfig.soc:36, soc/nordic/nrf53/Kconfig.soc:6, soc/nordic/nrf53/Kconfig:6), with value y, direct dependencies y (value: y), and select condition SOC_FAMILY_NORDIC_NRF (value: y)

warning: OPENAMP (defined at modules/Kconfig.open-amp:7) has direct dependencies ZEPHYR_OPEN_AMP_MODULE with value n, but is currently being y-selected by the following symbols:
 - IPC_SERVICE_BACKEND_RPMSG (defined at subsys/ipc/ipc_service/backends/Kconfig:10), with value y, direct dependencies MBOX && DT_HAS_ZEPHYR_IPC_OPENAMP_STATIC_VRINGS_ENABLED && IPC_SERVICE (value: y), and select condition MBOX && DT_HAS_ZEPHYR_IPC_OPENAMP_STATIC_VRINGS_ENABLED && IPC_SERVICE (value: y)

warning: NRFX_CLOCK (defined at modules/hal_nordic/nrfx/Kconfig:16) has direct dependencies HAS_NRFX && 0 with value n, but is currently being y-selected by the following symbols:
 - CLOCK_CONTROL_NRF (defined at drivers/clock_control/Kconfig.nrf:13), with value y, direct dependencies DT_HAS_NORDIC_NRF_CLOCK_ENABLED && CLOCK_CONTROL (value: y), and select condition !CLOCK_CONTROL_NRF_FORCE_ALT && DT_HAS_NORDIC_NRF_CLOCK_ENABLED && CLOCK_CONTROL (value: y)

warning: NRFX_GPPI (defined at modules/hal_nordic/nrfx/Kconfig:237) has direct dependencies HAS_NRFX && 0 with value n, but is currently being y-selected by the following symbols:
 - SOC_NRF53_RTC_PRETICK (defined at soc/nordic/nrf53/Kconfig:93), with value y, direct dependencies ((SYS_CLOCK_EXISTS && SOC_NRF5340_CPUNET) || SOC_NRF5340_CPUAPP) && SOC_SERIES_NRF53X && SOC_FAMILY_NORDIC_NRF (value: y), and select condition ((SYS_CLOCK_EXISTS && SOC_NRF5340_CPUNET) || SOC_NRF5340_CPUAPP) && SOC_SERIES_NRF53X && SOC_FAMILY_NORDIC_NRF (value: y)
 - UART_0_ENHANCED_POLL_OUT (defined at drivers/serial/Kconfig.nrfx_uart_instance:21), with value y, direct dependencies !n && HAS_HW_NRF_UARTE0 && (HAS_HW_NRF_PPI || HAS_HW_NRF_DPPIC) && (HAS_HW_NRF_UART0 || HAS_HW_NRF_UARTE0) && UART_NRFX && SERIAL (value: y), and select condition !n && HAS_HW_NRF_UARTE0 && (HAS_HW_NRF_PPI || HAS_HW_NRF_DPPIC) && (HAS_HW_NRF_UART0 || HAS_HW_NRF_UARTE0) && UART_NRFX && SERIAL (value: y)

warning: NRFX_GPIOTE0 (defined at modules/hal_nordic/nrfx/Kconfig:199) has direct dependencies HAS_NRFX && 0 with value n, but is currently being y-selected by the following symbols:
 - GPIO_NRFX (defined at drivers/gpio/Kconfig.nrfx:4), with value y, direct dependencies DT_HAS_NORDIC_NRF_GPIO_ENABLED && GPIO (value: y), and select condition HAS_HW_NRF_GPIOTE0 && DT_HAS_NORDIC_NRF_GPIO_ENABLED && GPIO (value: y)

warning: NRFX_IPC (defined at modules/hal_nordic/nrfx/Kconfig:260) has direct dependencies HAS_NRFX && 0 with value n, but is currently being y-selected by the following symbols:
 - MBOX_NRFX_IPC (defined at drivers/mbox/Kconfig.nrfx:4), with value y, direct dependencies DT_HAS_NORDIC_MBOX_NRF_IPC_ENABLED && MBOX (value: y), and select condition DT_HAS_NORDIC_MBOX_NRF_IPC_ENABLED && MBOX (value: y)

warning: NRFX_CLOCK_LFXO_TWO_STAGE_ENABLED (defined at modules/hal_nordic/nrfx/Kconfig:22) has direct dependencies NRFX_CLOCK && HAS_NRFX && 0 with value n, but is currently being y-selected by the following symbols:
 - CLOCK_CONTROL_NRF_K32SRC_XTAL (defined at drivers/clock_control/Kconfig.nrf:41), with value y, direct dependencies <choice CLOCK_CONTROL_NRF_SOURCE> (value: y), and select condition !SOC_SERIES_BSIM_NRFXX && !CLOCK_CONTROL_NRF_FORCE_ALT && <choice CLOCK_CONTROL_NRF_SOURCE> (value: y)

warning: NRFX_CLOCK_LF_SRC_XTAL (defined at modules/hal_nordic/nrfx/Kconfig:31) has direct dependencies NRFX_CLOCK && HAS_NRFX && 0 with value n, but is currently being y-selected by the following symbols:
 - CLOCK_CONTROL_NRF_K32SRC_XTAL (defined at drivers/clock_control/Kconfig.nrf:41), with value y, direct dependencies <choice CLOCK_CONTROL_NRF_SOURCE> (value: y), and select condition !CLOCK_CONTROL_NRF_FORCE_ALT && <choice CLOCK_CONTROL_NRF_SOURCE> (value: y)
Parsing /Kconfig
Loaded configuration '/boards/nordic/nrf5340dk/nrf5340dk_nrf5340_cpuapp_defconfig'
Merged configuration '/samples/bluetooth/hci_uart/prj.conf'
Merged configuration '/samples/bluetooth/hci_uart/boards/nrf5340dk_nrf5340_cpuapp.conf'

error: Aborting due to Kconfig warnings

CMake Error at /cmake/modules/kconfig.cmake:373 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  /cmake/modules/zephyr_default.cmake:131 (include)
  /share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

-- Configuring incomplete, errors occurred!
```

### Regression

- [ ] This is a regression.

### Relevant log output

```shell

```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

sayooj-aerlync, 2025-06-16:
This is the top of my tree `ea2990702f670ec05c5188c3822049b28ec0e6cf`.
@m-abubakar

sylvioalves, 2025-06-25:
@m-abubakar Any feedback?

m-abubakar, 2025-06-29:
Hi @sylvioalves I am able to build fine now with this change. 

Thank you.

henryzhang62, 2025-09-02:
In Zephyr version 4.2.99, 
west build -p always -b nrf5340dk/nrf5340/cpuapp samples/bluetooth/hci_uart -d build
build successfully by default as:
......................................................................................................
[3/197] Generating include/generated/zephyr/version.h
-- Zephyr version: 4.2.99 (/home/henry/zephyr-ble/zephyr), build: v4.2.0-2670-gaec461634fc3
[197/197] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       38024 B         1 MB      3.63%
             RAM:       24812 B       448 KB      5.41%
        IDT_LIST:          0 GB        32 KB      0.00%
Generating files from /home/henry/zephyr-ble/zephyr/build/zephyr/zephyr.elf for board: nrf5340dk

closed 2026-11-16 by nashif