zephyrproject-rtos/zephyr #91738: STM32WB55RG UART power consumption in stop2

whati001, 2025-06-17:
### Describe the bug

Hi All,

I am currently evaluating the stop 2 mode of the stm32wb55rg board (nucleo_wb55rg) and face some problems with the sample `zephyr/samples/boards/st/power_mgmt/serial_wakeup`. The sample uses the peripheral UART1 which does not support stop 2 mode (due to HSI switch), therefore the stop 2 mode is removed in the overlay file.

However, I still would like to have the stop 2 mode and simply "lock" the CPU to stop 1 until I release/suspend the peripheral. Please find below the updated sample:

main.c
```c
#include "zephyr/pm/policy.h"
#include "zephyr/shell/shell.h"
#include <zephyr/kernel.h>
#include <zephyr/device.h>
#include <zephyr/pm/device.h>
#include <zephyr/devicetree.h>
#include <zephyr/sys/printk.h>
#include <zephyr/sys/__assert.h>

#define STM32_WB55RG_STOP3_SUBSTATE (3)
const struct device *const dev = DEVICE_DT_GET(DT_CHOSEN(zephyr_console));

int main(void)
{

	if (!device_is_ready(dev)) {
		printk("Console device not ready");
		return 0;
	}

#if CONFIG_PM_DEVICE
	/* In PM_DEVICE modes, enable device as a wakeup source will prevent
	 * system to switch it off (clock off, set pins to sleep configuration, ...)
	 * It is not requested in CONFIG_PM mode only as in this case, device is not
	 * suspended before stop mode entry.
	 */

	bool ret;

	ret = pm_device_wakeup_is_capable(dev);
	if (!ret) {
		printk("Device is not wakeup capable\n");
	} else {
		printk("Device is wakeup capable\n");

		ret = pm_device_wakeup_enable(dev, true);
		if (!ret) {
			printk("Could not enable wakeup source\n");
		} else {
			printk("Wakeup source enable ok\n");
		}

		ret = pm_device_wakeup_is_enabled(dev);
		if (!ret) {
			printk("Wakeup source not enabled\n");
		} else {
			printk("Wakeup source enabled\n");
		}
	}
#endif

	// only go to stop1 until we disable the shell via `power_test uart`
	pm_policy_state_lock_get(PM_STATE_SUSPEND_TO_IDLE, STM32_WB55RG_STOP3_SUBSTATE);

	return 0;
}

static int cmd_power_test_uart(const struct shell *sh, size_t argc, char **argv)
{
	ARG_UNUSED(argc);
	ARG_UNUSED(argv);

	shell_print(sh, "Disabling UART and shell");
	pm_device_wakeup_enable(dev, false);
	shell_stop(sh);
	pm_policy_state_lock_put(PM_STATE_SUSPEND_TO_IDLE, STM32_WB55RG_STOP3_SUBSTATE);
	pm_device_action_run(dev, PM_DEVICE_ACTION_SUSPEND);

	return 0;
}

SHELL_STATIC_SUBCMD_SET_CREATE(sub_power_test,
			       SHELL_CMD(uart, NULL, "Disable uart", cmd_power_test_uart),
			       SHELL_SUBCMD_SET_END /* Array terminated. */
);

// register new command to disable the shell
SHELL_CMD_REGISTER(power_test, &sub_power_test, "Power test commands", NULL);
```

boards/nucleo_wb55rg.overlay
```dts
&cpu0{
	cpu-power-states = <&stop0 &stop1 &stop2>;
};

&usart1 {
	/* Set domain clock to HSI to allow wakeup from Stop mode */
	clocks = <&rcc STM32_CLOCK_BUS_APB2 0x00004000>,
		 <&rcc STM32_SRC_HSI USART1_SEL(2)>;

	/* Configure device as wakeup source */
	wakeup-source;

	/* Enable FIFO to avoid losing chars on device wakeup */
	fifo-enable;

	/* Configure sleep pinctrl configuration which will be used when
	 * device is not configured as wakeup source by the application.
	 * This use case is only applicable in PM_DEVICE mode.
	 */
	pinctrl-1 = <&analog_pb6 &analog_pb7>;
	pinctrl-names = "default", "sleep";
};

&clk_hsi {
	/* Make sure HSI is enabled */
	status = "okay";
};
```

prj.conf -> unchanged

As you can see, the code locks the CPU to sleep 1 until the new custom cmd command `power_test uart` is executed. The code seems to work, however, the current consumption jumps to some mA (see screenshot) after executing `	pm_device_action_run(dev, PM_DEVICE_ACTION_SUSPEND);`. 

[image]

The strange thing is, if I skip the `pm_device_action_run`, the current consumption is fine (the jump does not occur).
After some debugging, I guess the problem occurs due to this clock_off call. At least when I comment it out, the current remains the same:

[image]

There is a comment that fixed clocks are not handled yet, but I don't understand what the problem is.
In my opinion, it should be fine that only the main clock (`<&rcc STM32_CLOCK_BUS_APB2 0x00004000>`) is disabled. The domain clock (`<&rcc STM32_SRC_HSI USART1_SEL(2)>`) should always stay activated. Can be there any issue with unsend characters? I assume the shell still tries to send some data, but the UART is already disabled. But why should then a clock off leak so much current? PS. I do not receive any assert or hard fault (also used RTT as Logging Interface).

I am using the zephyr v4.1.0 and to reproduce the issue, please update the sample as explained above and measure the current.

For now, I simply skip the `pm_device_action_run` call, but I think this is still an issue and should be addressed.
Maybe you can help me find the root problem.

### Regression

- [ ] This is a regression.

### Relevant log output

```shell

```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

erwango, 2025-07-11:
I've finally had some time to check this out.
I've modified slightly your test to run the `cmd_power_test_uart` command automatically after a 1.5s timeout, I'm seeing this when disabling wakeup
[image]
I haven't done more tests but my assumption is that somehow the test set up wakes up the UART and hence the ker clock (HSI)
Can you have a check and use LSE as a ker clock in stead of HSI and see if something similar but at a lower scale happens ?
It would confirm this hypothesis.

whati001, 2025-07-14:
Thank you very much for your feedback, and I need to apologize regarding my bad explanation of the issue, still try to get into zephyr.

Firstly, I am using a custom board, which is pretty much the nucleo, but without an onboard the debugger. I have mentioned in the original ticket description the nucleo_wb55rg, because I have not created a dedicated board and use the board `nucleo_wb55rg` instead. I hope this does not make any difference.

Anyway, I have tried to use the LSE (on my custom and on the nucleo_wb55rg) and it seems that I do something wrong, because the sample stops working. Please correct me, but should not the following configuration correct?

Register: RCC_CCIPR
[image]

```dts
&cpu0{
	/* USART Wakeup requires automatic HSI16 switch on in deepsleep mode
	 * which isn't possible in Stop Mode 2.
	 * Remove Stop Mode 2 from supported modes
	 */
	cpu-power-states = <&stop0 &stop1 &stop2>;
};

&usart1 {
	/* Set domain clock to HSI to allow wakeup from Stop mode */
	clocks = <&rcc STM32_CLOCK_BUS_APB2 0x00004000>,
		 <&rcc STM32_SRC_LSE USART1_SEL(3)>;

	/* Configure device as wakeup source */
	wakeup-source;

	/* Enable FIFO to avoid losing chars on device wakeup */
	fifo-enable;

	/* Configure sleep pinctrl configuration which will be used when
	 * device is not configured as wakeup source by the application.
	 * This use case is only applicable in PM_DEVICE mode.
	 */
	pinctrl-1 = <&analog_pb6 &analog_pb7>;
	pinctrl-names = "default", "sleep";
};

&clk_lse {
	/* Make sure LSE is enabled */
	status = "okay";
};
```

The final clock overlay looks as follows (build/../zephyr.dts):

```dts
clocks {
			_hse: clk-hse {
			#clock-cells = < 0x0 >;
			compatible = "fixed-clock";
			clock-frequency = < 0x1e84800 >;
			status = "okay";
			phandle = < 0x3 >;
		};
		clk_hsi: clk-hsi {
			#clock-cells = < 0x0 >;
			compatible = "fixed-clock";
			clock-frequency = < 0xf42400 >;
			status = "disabled";
		};
		clk_hsi48: clk-hsi48 {
			#clock-cells = < 0x0 >;
			compatible = "fixed-clock";
			clock-frequency = < 0x2dc6c00 >;
			status = "okay";
		};
		clk_msi: clk-msi {
			#clock-cells = < 0x0 >;
			compatible = "st,stm32-msi-clock";
			msi-range = < 0x6 >;
			status = "disabled";
		};
		clk_lse: clk-lse {
			#clock-cells = < 0x0 >;
			compatible = "st,stm32-lse-clock";
			clock-frequency = < 0x8000 >;
			driving-capability = < 0x0 >;
			status = "okay";
		};
		clk_lsi1: clk-lsi1 {
			#clock-cells = < 0x0 >;
			compatible = "fixed-clock";
			clock-frequency = < 0x7d00 >;
			status = "okay";
		};
		clk_lsi2: clk-lsi2 {
			#clock-cells = < 0x0 >;
			compatible = "fixed-clock";
			clock-frequency = < 0x7d00 >;
			status = "disabled";
		};
		pll: pll {
			#clock-cells = < 0x0 >;
			compatible = "st,stm32wb-pll-clock";
			status = "disabled";
		};
		clk48: clk48 {
			#clock-cells = < 0x0 >;
			compatible = "st,stm32-clock-mux";
			status = "disabled";
			clocks = < &rcc 0x5 0xda0088 >;
		};
	};
```

whati001, 2025-07-16:
Yes the LSE is weird. Not sure why it breaks the sample for me.

I have updated the code and used the offical `nucleo_wb55rg` board.
```c
#include "zephyr/pm/policy.h"
#include "zephyr/shell/shell.h"
#include "zephyr/shell/shell_uart.h"
#include <stdint.h>
#include <zephyr/kernel.h>
#include <zephyr/device.h>
#include <zephyr/pm/device.h>
#include <zephyr/devicetree.h>
#include <zephyr/sys/printk.h>
#include <zephyr/sys/__assert.h>

#define STM32_WB55RG_STOP2_SUBSTATE (3)
const struct device *const dev = DEVICE_DT_GET(DT_CHOSEN(zephyr_console));

static bool shell_disabled = false;
int main(void)
{

	if (!device_is_ready(dev)) {
		printk("Console device not ready");
		return 0;
	}

#if CONFIG_PM_DEVICE
	/* In PM_DEVICE modes, enable device as a wakeup source will prevent
	 * system to switch it off (clock off, set pins to sleep configuration, ...)
	 * It is not requested in CONFIG_PM mode only as in this case, device is not
	 * suspended before stop mode entry.
	 */

	bool ret;

	ret = pm_device_wakeup_is_capable(dev);
	if (!ret) {
		printk("Device is not wakeup capable\n");
	} else {
		printk("Device is wakeup capable\n");

		ret = pm_device_wakeup_enable(dev, true);
		if (!ret) {
			printk("Could not enable wakeup source\n");
		} else {
			printk("Wakeup source enable ok\n");
		}

		ret = pm_device_wakeup_is_enabled(dev);
		if (!ret) {
			printk("Wakeup source not enabled\n");
		} else {
			printk("Wakeup source enabled\n");
		}
	}
#endif

	// only go to stop1 until we disable the shell via `power_test uart`
	pm_policy_state_lock_get(PM_STATE_SUSPEND_TO_IDLE, STM32_WB55RG_STOP2_SUBSTATE);

	uint8_t idx = 0;
	while (1) {
		printk("Sleep idx: %d\n", idx);
		if (idx++ == 5) {
			if (shell_disabled == false) {
				printk("MAIN THREAD: Disabling shell and going to stop2\n");
				shell_stop(shell_backend_uart_get_ptr());
				printk("disable wakeup: %d\n", pm_device_wakeup_enable(dev, false));
				pm_policy_state_lock_put(PM_STATE_SUSPEND_TO_IDLE,
							 STM32_WB55RG_STOP2_SUBSTATE);
				pm_device_action_run(dev, PM_DEVICE_ACTION_SUSPEND);

				shell_disabled = true;
			}
		}

		k_sleep(K_SECONDS(2));
	}
	return 0;
}

static int cmd_power_test_uart(const struct shell *sh, size_t argc, char **argv)
{
	ARG_UNUSED(argc);
	ARG_UNUSED(argv);

	printk("SHELL THREAD: Disabling UART and shell");
	printk("disable wakeup: %d\n", pm_device_wakeup_enable(dev, false));
	shell_stop(sh);
	pm_policy_state_lock_put(PM_STATE_SUSPEND_TO_IDLE, STM32_WB55RG_STOP2_SUBSTATE);
	pm_device_action_run(dev, PM_DEVICE_ACTION_SUSPEND);

	shell_disabled = true;
	return 0;
}

SHELL_STATIC_SUBCMD_SET_CREATE(sub_power_test,
			       SHELL_CMD(uart, NULL, "Disable uart", cmd_power_test_uart),
			       SHELL_SUBCMD_SET_END /* Array terminated. */
);

// register new command to disable the shell
SHELL_CMD_REGISTER(power_test, &sub_power_test, "Power test commands", NULL);
```

As you can see, we can now disable the uart via shell or via a fixed timeout. In addition, I have measured the current via a Nordic PPK2, connected as ampere meter via USB STL JP1 pin.

---

The following graphs are via the HSI as domain clock and zephyr v4.1.0.
1) If the uart is disabled by the main thread via timeout, no current jump occurs.

[image]

2) If the uart is disabled via shell command, a current jump occurs.

[image]

So I guess that I either not allowed to disable the shell and uart within the shell context or that something tries to aquire the shell/uart after disabling.

whati001, 2025-07-16:
If I use the zephyr upstream (66acb364b1907751efe40e9a2cdf22080985d893) with the same sample above, I receive a current jump after hitting the first character into the shell. If I keep the shell unused, the current jump does not occur.

[image]

erwango, 2025-07-17:
I've been able to reproduce the following:
[image]
The jump around 5.3 sec is the wakeup when typing in the shell, then the relase happens at 11.5sec and indeed, we can see that the initial consumption is not recovered. I'll try to dig into it

erwango, 2025-07-17:
I've been able to identify 2 distinct issues:

First issue, when running `cmd_power_test_uart`, just before exiting the function, the following is run:
```
#0  pm_policy_state_lock_get (state=state@entry=PM_STATE_SUSPEND_TO_IDLE, substate_id=substate_id@entry=255 '\377') at /local/home/frq07517/zephyrproject/zephyr/subsys/pm/policy/policy_state_lock.c:53
#1  0x0800a964 in uart_stm32_pm_policy_state_lock_get_unconditional () at /local/home/frq07517/zephyrproject/zephyr/drivers/serial/uart_stm32.c:108
#2  0x0800a978 in uart_stm32_pm_policy_state_lock_get (dev=<optimized out>) at /local/home/frq07517/zephyrproject/zephyr/drivers/serial/uart_stm32.c:120
#3  0x0800a99a in uart_stm32_irq_tx_enable (dev=<optimized out>) at /local/home/frq07517/zephyrproject/zephyr/drivers/serial/uart_stm32.c:973
#4  0x0800852a in z_impl_uart_irq_tx_enable (dev=<optimized out>) at /local/home/frq07517/zephyrproject/zephyr/include/zephyr/drivers/uart/uart_internal.h:312
#5  uart_irq_tx_enable (dev=<optimized out>) at /local/home/frq07517/zephyrproject/zephyr/samples/boards/st/power_mgmt/serial_wakeup/build/nucleo_wb55rg/zephyr/include/generated/zephyr/syscalls/uart.h:205
#6  0x08008610 in irq_write (sh_uart=0x20000678 <shell_transport_uart_shell_uart>, data=<optimized out>, length=<optimized out>, cnt=0x200016f4 <shell_uart_stack+1844>) at /local/home/frq07517/zephyrproject/zephyr/subsys/shell/backends/shell_uart.c:381
#7  0x08008624 in write_uart (transport=<optimized out>, data=<optimized out>, length=<optimized out>, cnt=<optimized out>) at /local/home/frq07517/zephyrproject/zephyr/subsys/shell/backends/shell_uart.c:414
#8  0x080093f4 in z_shell_write (sh=0x800c0f0 <shell_uart>, data=0x20000e10 <shell_uart_out_buffer>, length=7) at /local/home/frq07517/zephyrproject/zephyr/subsys/shell/shell_ops.c:460
#9  0x0800941a in z_shell_print_stream (user_ctx=<optimized out>, data=<optimized out>, len=<optimized out>) at /local/home/frq07517/zephyrproject/zephyr/subsys/shell/shell_ops.c:478
#10 0x08008f96 in z_shell_fprintf_buffer_flush (sh_fprintf=sh_fprintf@entry=0x800ce64 <shell_uart_fprintf>) at /local/home/frq07517/zephyrproject/zephyr/subsys/shell/shell_fprintf.c:46
#11 0x08002b06 in z_shell_fprintf_fmt (sh_fprintf=0x800ce64 <shell_uart_fprintf>, fmt=fmt@entry=0x8002b07 <z_shell_fprintf_fmt+30> "矏", args=..., args@entry=...) at /local/home/frq07517/zephyrproject/zephyr/subsys/shell/shell_fprintf.c:39
#12 0x08009194 in z_shell_raw_fprintf (ctx=<optimized out>, fmt=0x800d1f4 "\033[1;3%dm") at /local/home/frq07517/zephyrproject/zephyr/subsys/shell/shell_ops.h:24
#13 0x0800315c in z_shell_vt100_color_set (sh=sh@entry=0x800c0f0 <shell_uart>, color=<optimized out>) at /local/home/frq07517/zephyrproject/zephyr/subsys/shell/shell_ops.c:519
#14 0x0800946a in z_shell_vfprintf (sh=sh@entry=0x800c0f0 <shell_uart>, color=<optimized out>, fmt=0x800ceec "%s", fmt@entry=0x20000738 <z_idle_threads> "", args=..., args@entry=...) at /local/home/frq07517/zephyrproject/zephyr/subsys/shell/shell_ops.c:545
#15 0x08009490 in z_shell_fprintf (sh=sh@entry=0x800c0f0 <shell_uart>, color=color@entry=SHELL_VT100_COLOR_GREEN, fmt=0x800ceec "%s") at /local/home/frq07517/zephyrproject/zephyr/subsys/shell/shell_ops.c:569
#16 0x0800332c in print_prompt (sh=sh@entry=0x800c0f0 <shell_uart>) at /local/home/frq07517/zephyrproject/zephyr/subsys/shell/shell_ops.c:391
#17 0x080095c0 in z_shell_print_prompt_and_cmd (sh=sh@entry=0x800c0f0 <shell_uart>) at /local/home/frq07517/zephyrproject/zephyr/subsys/shell/shell_ops.c:423
#18 0x08001e2c in state_set (sh=0x800c0f0 <shell_uart>, state=<optimized out>) at /local/home/frq07517/zephyrproject/zephyr/subsys/shell/shell.c:112
```

Which mean that shell attempts to send a char on the uart device which ends up issuing a `pm_policy_state_lock_get(PM_ALL_SUBSTATES)`.
Though, since uart device has been suspended just before it won't be able to perform the required function and hence no end transfer IRQ will be generated to resume unlock this situation (with `pm_policy_state_lock_put(PM_ALL_SUBSTATES)`).

Hence the CPU remains locked in running state. So globally the sequence is wrong, but I don't know yet what should be fixed.

erwango, 2025-07-22:
> Second issue, when shell closure is done after timeout, there is an unbalanced `uart_stm32_pm_policy_state_lock_get_unconditional()` issued from `uart_stm32_isr()` function.
> 
> The following patch fixes the issue:
> 
> @@ -1314,11 +1314,12 @@ static void uart_stm32_isr(const struct device *dev)
>  
>  		LL_USART_ClearFlag_WKUP(usart);
>  
>  		if (!data->rx_woken) {
>  			/* Prevent SoC from entering STOP mode until RX goes IDLE */
> -			uart_stm32_pm_policy_state_lock_get_unconditional();
> +			uart_stm32_pm_policy_state_lock_get(dev);
> I've not been able to identify regression vs intended implementation in [#87267](https://github.com/zephyrproject-rtos/zephyr/pull/87267), but this would have to be investigated before issuing a PR.
> 
> Would you be able to have a check on these both points and at least see if you agree on these observations ?

@JordanYates Would you be ok with proposed patch vs what you proposed in https://github.com/zephyrproject-rtos/zephyr/pull/87267 or have I missed something ?

closed 2025-10-06 by github-actions[bot]