zephyrproject-rtos/zephyr #91777: net_if: Make uptime at last operation change available to application

clamattia, 2025-06-17:
### Summary

Some protocols like `simple network management protocol` (SNMP) need to present this value.
See for example: https://www.alvestrand.no/objectid/1.3.6.1.2.1.2.2.1.9.html
See also: https://github.com/zephyrproject-rtos/zephyr/issues/2329

### Describe the solution you'd like

An API-to get this value from an `net_if`.

### Alternatives

Caller register to net mgnt-callbacks and tracks it itself?
Alternatives?

### Additional Context

Let me know, if I am missing anything. Thanks in advance.

jukkar, 2025-06-17:
As the statistics is optional feature, I don't think it makes sense to add such information there.

clamattia, 2025-06-17:
> As the statistics is optional feature, I don't think it makes sense to add such information there.

Both is fine for me.

PR #91782 (2025-06-20): Keep track of operational state change time for net_if
Fixes #91777