zephyrproject-rtos/zephyr #91862: Linting for DTS files

yishai1999, 2025-06-18:
### Problem Description

Today the zephyr [Devicetree Style Guidelines](https://docs.zephyrproject.org/latest/contribute/style/devicetree.html) (which also references the [Linux Devicetree  Coding Style](https://docs.kernel.org/devicetree/bindings/dts-coding-style.html)) are not enforced in any way and many contributors and even reviewers don't know about them.

### Proposed Change (Summary)

Create a DTS lint script that will be run in CI

### Proposed Change (Detailed)

There should be a linting script that run for every PR that makes sure the DTS files are acceptable and match the guidelines.
The PR should be blocked until the script passes.

### Alternatives Considered

* Making this issue more known so that reviewers don't let bad DTS files be merged
* Staying with the current situation where many many DTS files don't follow the guidelines.

rruuaanng, 2025-06-20:
A while ago, #83352 was merged into the main, and it checks the following rules:

```
Naming and Valid Characters[¶](https://docs.kernel.org/devicetree/bindings/dts-coding-style.html#naming-and-valid-characters)
The Devicetree Specification allows a broad range of characters in node and property names, but this coding style narrows the range down to achieve better code readability.

Node and property names can use only the following characters:

Lowercase characters: [a-z]

Digits: [0-9]

Dash: -

Labels can use only the following characters:

Lowercase characters: [a-z]

Digits: [0-9]

Underscore: _

Unless a bus defines differently, unit addresses shall use lowercase hexadecimal digits, without leading zeros (padding).

Hex values in properties, e.g. “reg”, shall use lowercase hex. The address part can be padded with leading zeros.
```

(By the way, what you mentioned isn't even implemented in Linux)

from `arch/arm64/boot/dts/apple/t8012-j132.dts`

```
/ {
	model = "Apple T2 MacBookPro15,2 (j132)";
	compatible = "apple,j132", "apple,t8012", "apple,arm-platform";
};
```

from `arch/arm64/boot/dts/apple/t8011-j207.dts`

```
/ {
	compatible = "apple,j207", "apple,t8011", "apple,arm-platform";
	model = "Apple iPad Pro 2 (10.5-inch) (Wi-Fi)";
};
```

kylebonnici, 2025-06-23:
I have been working to improve the dts lsp formating https://github.com/kylebonnici/dts-lsp to be able to use it for linting and i have a working prototype that can perhaps be used in CI

One can look at this PR https://github.com/kylebonnici/dts-lsp/pull/65 and the documentFormating.test.ts test file so see all the rules i am applying.....

I also have a very alpha runner to be able to generate diff for files.... https://github.com/kylebonnici/dts-linter to run this 
npm i -g devicetree-language-server
npm ci
 find /zephyr/boards/ -type f \( -name "*.dts" -o -name "*.dtsi" \) | xargs ts-node src/validate-format.ts

and this will generate a file like the one attached I would  any feedback 🙂

[diff.txt](https://github.com/user-attachments/files/20870222/diff.txt)

kylebonnici, 2025-09-24:
> > and this will generate a file like the one attached I would any feedback 🙂
> 
> I like it!

You can look at https://github.com/zephyrproject-rtos/zephyr/pull/92334 ... it is at final stages fingers crossed it get accepted :)

yishai1999, 2025-11-15:
Amazing job @kylebonnici ! Thank you so much for the hard work!
Let's give a last push and merge https://github.com/zephyrproject-rtos/zephyr/pull/98158 too