zephyrproject-rtos/zephyr #91884: TBD: Bluetooth: Advertising Data missing after 15 mins

cvinayak, 2025-06-19:
### Describe the bug

Advertising Data missing after 15 mins

### Regression

- [ ] This is a regression.

### Relevant log output

```shell
[2025-06-19 04:43:43.326] uart:~$ bt adv-data 06093132333435
[2025-06-19 04:43:44.944] uart:~$ bt adv-start
[2025-06-19 04:43:46.820] Advertiser[0] 0x20002350 set started
[2025-06-19 04:43:46.824] Advertiser[0] 0x20002350 RPA expired
[2025-06-19 04:56:20.443] Advertising data for Advertiser[0] 0x20002350 updated.
[2025-06-19 04:56:20.451] Advertiser[0] 0x20002350 RPA expired
[2025-06-19 05:11:20.432] Advertising data for Advertiser[0] 0x20002350 updated.
[2025-06-19 05:11:20.440] uart:~$ bt adv-data 06093132333435
[2025-06-19 05:24:54.315] uart:~$
```

### Impact

Showstopper – Prevents release or major functionality; system unusable.

sayooj-aerlync, 2025-06-23:
Which platform is this?

danieldegrasse, 2025-06-24:
@cvinayak can you please provide more details? Going to mark as low for time being